esphome/issues #1806: Add-on config 'auto_uart' is deprecated, use 'uart'

TheZoker, 2021-02-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

Not related

**ESPHome version (latest production, beta, dev branch)**

1.16.0b8

**Affected component:**

Hassio Addon:
https://github.com/esphome/hassio

**Description of problem:**

The Home Assistant Supervisor changed some config parameters:
https://github.com/home-assistant/supervisor/pull/2429

These are the warnings from the log:
```
21-02-03 12:37:16 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome (dev)
21-02-03 12:37:16 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome
21-02-03 12:37:16 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome (beta)
```

**Problem-relevant YAML-configuration entries:**
```yaml
not related

```

**Logs (if applicable):**

```
not relateed
```

**Additional information and things you've tried:**

xHirscHx, 2021-02-05:
Ya i got a few of those yesterday.

21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of OpenZWave
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Z-Wave JS
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Terminal & SSH
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of HomeMatic CCU
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of TellStick
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of deCONZ
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of deCONZ
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Z-Wave JS to MQTT
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of SSH & Web Terminal
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of SSH & Web Terminal
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Visual Studio Code
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Z-Wave to MQTT
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Node-RED
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of Node-RED
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of ZeroTier One
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of WireGuard
21-02-05 02:55:53 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of AppDaemon 4
21-02-05 02:55:54 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Network UPS Tools
21-02-05 02:55:54 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of Network UPS Tools
21-02-05 02:55:54 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
21-02-05 02:55:54 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
21-02-05 02:55:54 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/6e66619d repository
21-02-05 02:55:54 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of OpenZWave
21-02-05 02:55:54 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Z-Wave JS
21-02-05 02:55:54 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Terminal & SSH
21-02-05 02:55:54 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of HomeMatic CCU
21-02-05 02:55:54 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of TellStick
21-02-05 02:55:54 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of deCONZ
21-02-05 02:55:54 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of deCONZ
[... cut ...]
21-02-05 02:55:55 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of AppDaemon 4
21-02-05 02:55:55 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Network UPS Tools
21-02-05 02:55:55 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of Network UPS Tools

Ascariota, 2021-02-06:
Same with me.

`21-02-06 16:21:43 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome
21-02-06 16:21:43 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome (beta)
21-02-06 16:21:43 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome (dev)
21-02-06 16:21:43 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Z-Wave to MQTT
21-02-06 16:21:43 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of SSH & Web Terminal
21-02-06 16:21:43 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome
21-02-06 16:21:43 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of AppDaemon 4
21-02-06 16:21:43 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Z-Wave JS to MQTT
21-02-06 16:21:43 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Network UPS Tools`

726mrbilky, 2021-02-07:
Me too no word yet?

supergroover, 2021-02-08:
yah me too

PascalB1960, 2021-02-08:
Same problem here

21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of HomeMatic CCU
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of OpenZWave
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Z-Wave JS
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Terminal & SSH
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of deCONZ
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of deCONZ
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of TellStick
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of WireGuard
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of SSH & Web Terminal
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of SSH & Web Terminal
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Network UPS Tools
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of Network UPS Tools
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of ZeroTier One
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of AppDaemon 4
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Z-Wave to MQTT
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Z-Wave JS to MQTT
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Node-RED
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of Node-RED
21-02-08 09:41:03 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of Visual Studio Code

Final-Hawk, 2021-02-08:
Same here as well. Waiting for a solution

cinghialino, 2021-02-08:
same here. Can't access any add-ons.

LaurentMicrotec, 2021-02-08:
someone found a solution?

closed 2021-02-08 by glmnet

726mrbilky, 2021-02-09:
Check your logs I updated yesterday and this went away with the new update

Ascariota, 2021-02-09:
Yes for me now with update it's ok

viktak, 2021-02-09:
Fixed, thanks!