esphome/issues #1807: Update to v1.16.0 fails

bouwew, 2021-02-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HA Supervised in a Proxmox VM*

ESPHome version 1.15.3

**Affected component:**

**Description of problem:**
The update to v1.16.0 fails.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
21-02-03 18:17:52 INFO (SyncWorker_4) [supervisor.docker.interface] Updating image esphome/esphome-hassio-amd64:1.15.3 to esphome/esphome-hassio-amd64:1.16.0
21-02-03 18:17:52 INFO (SyncWorker_4) [supervisor.docker.interface] Downloading docker image esphome/esphome-hassio-amd64 with tag 1.16.0.
21-02-03 18:17:54 ERROR (SyncWorker_4) [supervisor.docker.interface] Can't install esphome/esphome-hassio-amd64:1.16.0 -> 500 Server Error for http+docker://localhost/v1.40/images/create?tag=1.16.0&fromImage=esphome%2Fesphome-hassio-amd64: Internal Server Error ("unauthorized: authentication required").

```

**Additional information and things you've tried:**

jesserockz, 2021-02-04:
Can you please try again?

I just installed it fine:
```text
21-02-04 09:28:00 INFO (SyncWorker_4) [supervisor.docker.interface] Downloading docker image esphome/esphome-hassio-amd64 with tag 1.16.0.
21-02-04 09:28:54 INFO (MainThread) [supervisor.addons] Add-on 'a0d7b954_esphome' successfully installed
21-02-04 09:29:03 INFO (SyncWorker_1) [supervisor.docker.addon] Starting Docker add-on esphome/esphome-hassio-amd64 with version 1.16.0
```

bouwew, 2021-02-04:
Yes, just tried it again, now it magically works :)

closed 2021-02-04 by bouwew