esphome/issues #1811: 1.15: Wrong values when reading DHT11 sensor

simonszu, 2021-02-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Debian, installed with pip. 

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266, Wemos D1 Mini

**ESPHome version (latest production, beta, dev branch)**
esphome 1.15.3

**Affected component:**

https://esphome.io/components/sensor/dht.html

**Description of problem:**

I reflashed an older component of D1 Mini / DHT11 shield to use a different MQTT topic (basically just renamed the sensor from living room to kellerklima). Since the reflash the measured temperature and humidity values are significantly wrong. The tempereature is measured as -9 degrees celsius, the humidity is measured as 7% (Reality: +20 degrees and 35% humidity). 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: kellerklima
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 'Devicefunk'
  password: 'freewaycola'
  domain: .home.simonszu.de

mqtt:
  broker: '192.168.10.4'
  username: ''
  password: ''

# Enable logging
logger:

ota:
  password: ''

api:

sensor:
  - platform: dht
    model: DHT11
    pin: D4
    temperature:
      name: "Keller Temperatur"
    humidity:
      name: "Keller Luftfeuchtigkeit"
    update_interval: 15s
  - platform: wifi_signal
    name: "Keller Wifi"
    update_interval: 15s

```

**Logs (if applicable):**

```
[12:08:22][I][app:105]: ESPHome version 1.15.3 compiled on Jan 29 2021, 10:00:29
[12:08:22][C][wifi:443]: WiFi:
[12:08:22][C][wifi:303]:   SSID: 'Devicefunk'
[12:08:22][C][wifi:304]:   IP Address: 192.168.12.15
[12:08:22][C][wifi:306]:   BSSID: 18:E8:29:A1:97:54
[12:08:22][C][wifi:307]:   Hostname: 'kellerklima'
[12:08:22][C][wifi:311]:   Signal strength: -48 dB ▂▄▆█
[12:08:22][C][wifi:315]:   Channel: 6
[12:08:22][C][wifi:316]:   Subnet: 255.255.255.0
[12:08:22][C][wifi:317]:   Gateway: 192.168.12.1
[12:08:22][C][wifi:318]:   DNS1: 192.168.12.1
[12:08:22][C][wifi:319]:   DNS2: (IP unset)
[12:08:22][C][logger:185]: Logger:
[12:08:22][C][logger:186]:   Level: DEBUG
[12:08:22][C][logger:187]:   Log Baud Rate: 115200
[12:08:22][C][logger:188]:   Hardware UART: UART0
[12:08:22][C][dht:017]: DHT:
[12:08:22][C][dht:018]:   Pin: GPIO2 (Mode: INPUT)
[12:08:22][C][dht:022]:   Model: DHT11
[12:08:22][C][dht:027]:   Update Interval: 15.0s
[12:08:22][C][dht:029]:   Temperature 'Keller Temperatur'
[12:08:22][C][dht:029]:     Unit of Measurement: '°C'
[12:08:22][C][dht:029]:     Accuracy Decimals: 1
[12:08:22][C][dht:029]:     Icon: 'mdi:thermometer'
[12:08:22][C][dht:030]:   Humidity 'Keller Luftfeuchtigkeit'
[12:08:22][C][dht:030]:     Unit of Measurement: '%'
[12:08:22][C][dht:030]:     Accuracy Decimals: 0
[12:08:22][C][dht:030]:     Icon: 'mdi:water-percent'
[12:08:22][C][ota:029]: Over-The-Air Updates:
[12:08:22][C][ota:030]:   Address: kellerklima.home.simonszu.de:8266
[12:08:22][C][mqtt:051]: MQTT:
[12:08:22][C][mqtt:053]:   Server Address: 192.168.10.4:1883 (192.168.10.4)
[12:08:22][C][mqtt:054]:   Username: ''
[12:08:22][C][mqtt:055]:   Client ID: 'kellerklima-cc50e3471415'
[12:08:22][C][mqtt:057]:   Discovery prefix: 'homeassistant'
[12:08:22][C][mqtt:058]:   Discovery retain: YES
[12:08:22][C][mqtt:060]:   Topic Prefix: 'kellerklima'
[12:08:22][C][mqtt:062]:   Log Topic: 'kellerklima/debug'
[12:08:22][C][mqtt:065]:   Availability: 'kellerklima/status'
[12:08:22][C][api:095]: API Server:
[12:08:22][C][api:096]:   Address: kellerklima.home.simonszu.de:6053
[12:08:22][C][wifi_signal.sensor:009]: WiFi Signal 'Keller Wifi'
[12:08:22][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[12:08:22][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[12:08:22][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[12:08:22][C][mqtt.sensor:024]: MQTT Sensor 'Keller Wifi':
[12:08:22][C][mqtt.sensor:028]:   State Topic: 'kellerklima/sensor/keller_wifi/state'
[12:08:22][C][mqtt.sensor:024]: MQTT Sensor 'Keller Temperatur':
[12:08:22][C][mqtt.sensor:028]:   State Topic: 'kellerklima/sensor/keller_temperatur/state'
[12:08:22][C][mqtt.sensor:024]: MQTT Sensor 'Keller Luftfeuchtigkeit':
[12:08:22][C][mqtt.sensor:028]:   State Topic: 'kellerklima/sensor/keller_luftfeuchtigkeit/state'
[12:08:22][D][api.connection:604]: Client 'Home Assistant 2021.1.5 (192.168.10.4)' connected successfully!
[12:08:27][D][dht:048]: Got Temperature=-9.9°C Humidity=7.5%
[12:08:27][D][sensor:092]: 'Keller Temperatur': Sending state -9.90000 °C with 1 decimals of accuracy
[12:08:27][D][sensor:092]: 'Keller Luftfeuchtigkeit': Sending state 7.50000 % with 0 decimals of accuracy
[12:08:30][D][sensor:092]: 'Keller Wifi': Sending state -49.00000 dB with 0 decimals of accuracy
[12:08:42][D][dht:048]: Got Temperature=-9.9°C Humidity=7.4%
[12:08:42][D][sensor:092]: 'Keller Temperatur': Sending state -9.90000 °C with 1 decimals of accuracy
[12:08:42][D][sensor:092]: 'Keller Luftfeuchtigkeit': Sending state 7.40000 % with 0 decimals of accuracy
[12:08:45][D][sensor:092]: 'Keller Wifi': Sending state -47.00000 dB with 0 decimals of accuracy
[12:08:57][D][dht:048]: Got Temperature=-9.9°C Humidity=7.4%
[12:08:57][D][sensor:092]: 'Keller Temperatur': Sending state -9.90000 °C with 1 decimals of accuracy
[12:08:57][D][sensor:092]: 'Keller Luftfeuchtigkeit': Sending state 7.40000 % with 0 decimals of accuracy
[12:09:00][D][sensor:092]: 'Keller Wifi': Sending state -50.00000 dB with 0 decimals of accuracy
[12:09:12][D][dht:048]: Got Temperature=-9.9°C Humidity=7.4%
[12:09:12][D][sensor:092]: 'Keller Temperatur': Sending state -9.90000 °C with 1 decimals of accuracy
[12:09:12][D][sensor:092]: 'Keller Luftfeuchtigkeit': Sending state 7.40000 % with 0 decimals of accuracy

```

**Additional information and things you've tried:**

Since this is reproducible on other D1 Minis and other DHT shields i have laying around, and did not happen on the older firmware i compiled with an unknown version of esphome, i assume a firmware problem. 

I have compared the measurements with the real values over time and it seems that this is not a linear offset, but the temperature stays this low, no matter if the actual temperature is changing up to 5 degrees.

A-wels, 2021-03-29:
Did you find a solution for this? 

I'm having the same issue.

simonszu, 2021-03-30:
Unfortunately i didn't yet. Had some other projects to work on, so i decided to just wait until someone has some suggestions.

EmnaX, 2021-04-28:
Just here to add a "me too"!

ESP Home 1.16.2 running on a Raspberry Pi 4 with Home Assistant core-2021.4.0.

I have this [DHT22 module](https://www.aliexpress.com/item/32769460765.html?spm=a2g0s.9042311.0.0.27424c4daYSIuW) connected to a NodeMCU V2 (esp8266). I also have a resistor (4.7kohm) connected between the 3.3V line and the signal line (in my case, D7 on the NodeMCU).

NodeMCU Config File:
```
esphome:
  name: diy14
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "IoTNet"
  password: "********"
  domain: .example.com

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Diy14 Fallback Hotspot"
    password: "********"

captive_portal:

# Enable logging
logger:
  tx_buffer_size: 1024

# Enable Home Assistant API
api:
  password: "********"

ota:
  password: "********"
  
web_server:
  port: 80

binary_sensor:
  # Motion detector PIR
  - platform: gpio
    pin: D5
    name: "Basement Stair Motion"
    device_class: motion

sensor:
  - platform: dht
    pin: D7
    model: AUTO_DETECT
    #model: DHT22
    #model: AM2302
    temperature:
      name: "Basement Stairwell Temperature"
    humidity:
      name: "Basement Stairwell Humidity"
    update_interval: 60s
```

EmnaX, 2021-04-28:
I removed the 4.7kohm resistor between 3.3V and the signal line and everything works fine. My basement temp is reporting at 19.2C and 37 %RH. Previously it was reporting around -6C and 9%.

The AliExpress seller I purchased the "DHT22 module" from sells both a "module" and a "sensor" (with and without an attached PCB respectively). I'm guessing the "module" version already has a resistor between 3.3V and signal. Hopefully that helps you @simonszu and @A-wels .

A-wels, 2021-04-29:
I will try this the next time I restart the sensor. I left it running for 2 days and noticed that it started to report the correct values. Restarting made it return the the incorrect values which again started to become correct after a while.

Thanks for the possible solution

simonszu, 2021-04-29:
I would consider that "solution" only as a workaround. There are a lot of DHT11/DHT22 shields for the D1 Mini and assumably for other ESP form factors. They have some resistors of unknown value included, and as an end user i would expect these shields to work out of the box, without the need to desolder stuff from them.

EmnaX, 2021-04-29:
Sorry @simonszu , I should clarify. I'm not suggesting to desolder any resistors from your sensor. When I read the instructions from the [ESPHome Documentation](https://www.esphome.io/components/sensor/dht.html), it tells us to use a 4.7kohm resistor. Relevant quote from the docs:

>The DHT22 and DHT11 require external pull up resistors on the data line. To do this, solder a resistor with about 4.7kΩ

So I followed that and added in my own 4.7kohm resistor, not realizing the minor difference between my DHT22 (which has a PCB attached to it) and the one pictured in the documentation (which does NOT have a PCB). So I _think_ the cause of the error (at least in my case) was essentially adding two resistors. The PCB likely has it's own resistor and then I added a second. When I removed the one I added in, everything started working properly. So if you didn't add in your own resistor, then I'm not sure what might be causing your issue. I just wanted to post what worked for me.

closed 2021-09-10 by stale[bot]