esphome/issues #1812: ESP32 Node connection & status dropping from green to red after a few seconds/minutes

Sleinous, 2021-02-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**

I have a single ESP-32 gateway from Olimex on ESPHome for over a year now - no issues and zero disconnections up until now.
Its purpose is to grab xiaomi temp/humidity/battery sensor ble data for use in HA.
Yesterday I noticed that the status bar for this node was red (as opposed to green) & seemed to be sporadically updating. (I noticed this because I also run a XiaomiGateway v3 parallel to this - grabbing the same data from the BLE temp/humidity sensors for redundancy purposes).
Spent about a day researching on the forums and github 'issues' pages thinking it may be to do with multicast and/or mDNS except this has always worked fine for me up until now and I haven't changed anything on my network (no VLANs in use here).
**_So here's what i've tried so far_**
I noticed that the device (esp32cuisine.local) was no longer able to grab it's IP - odd because it was always able to in the past so I added the manual_ip configuration to my yaml for this esp32 - was able to connect to the network but the status bar on ESPHome would stay green for about 5 seconds, then go orange & back to red. Again, sporadic updates of BLE tracking data afterwards. The device was still pingable even after ESPHome displayed the red bar across that node.
I read that this could be linked to multicast & mDNS, but again, i've never had this issue in well over a year of using esphome.

I then tried removing the manual_ip config & adding 'domain', declaring this device in my domain DNS. This time the device correctly connected to the network and resolved it's IP.
Didn't help with the node going offline (red) in ESPHome dashboard though.

Last tried activating VERY_VERBOSE in logger (connected to wifi and then later connected over USB)
I noticed that as soon as VERY_VERBOSE was activated and I was inspecting logs through ESPHome UI the node would stay green for a lot longer, upwards of 30mn, but could someones return to an 'offline' (red) state but still be receiving BLE data from the different sensors around the house...

At this stage I have run out of ideas on where this problem is coming from. Any ideas welcome ! Thanks 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32cuisine
  platform: ESP32
  board: esp32-gateway

wifi:
  ssid: "testssid"
  password: "testxxxx"
  domain: .xxxxxxx
  #manual_ip:
    # Set this to the IP of the ESP
    #static_ip: 192.168.X.X
    # Set this to the IP address of the router. Often ends with .1
    #gateway: 192.168.X.X
    # The subnet of the network. 255.255.255.0 works for most home networks.
    #subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32 Cuisine Fallback Hotspot"
    password: "FRcy10KIiQl1"

captive_portal:

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:

ota:

esp32_ble_tracker:

sensor:
  - platform: xiaomi_lywsdcgq
    mac_address: "58:2D:34:3B:52:F8"
    temperature:
      name: "Chambre Temperature"
    humidity:
      name: "Chambre Humidité"
    battery_level:
      name: "Chambre Batterie"
  - platform: xiaomi_lywsdcgq
    mac_address: "58:2D:34:3B:55:BC"
    temperature:
      name: "SdB Temperature"
    humidity:
      name: "SdB Humidité"
    battery_level:
      name: "SdB Batterie"
  - platform: xiaomi_lywsdcgq
    mac_address: "58:2D:34:3B:35:8B"
    temperature:
      name: "Salon Temperature"
    humidity:
      name: "Salon Humidité"
    battery_level:
      name: "Salon Batterie"
  - platform: xiaomi_lywsdcgq
    mac_address: "58:2D:34:3B:35:EE"
    temperature:
      name: "Couloir Temperature"
    humidity:
      name: "Couloir Humidité"
    battery_level:
      name: "Couloir Batterie"

```

**Logs (if applicable):**

```
debug over USB log incoming here

```

**Additional information and things you've tried:**

Sleinous, 2021-02-05:
I've found a (hopefully) temporary solution : add to the ESPHome configuration "status_use_ping: true" + FIXED IP
Now the production node (& the temp test node) are always green status in ESPHome and api connection between node and HA isn't dropping repeatedly.
I'm still totally clueless as to why this is happening now after nearly a year's uptime without any issues prior.
If anyone has any ideas, or needs more info I can provide as necessary - or if anyone else has experienced the same behaviour very recently ?
Hardware : EdgeRouter 12, 3x Managed switches, 2x Unifi NanoHD AP's

rradar, 2021-02-06:
If you didn't change your router/network configuration it could also be a new device in your network which could for example flooding your network with mDNS broadcasts.

ogiewon, 2021-02-18:
I am seeing the exact same issue.  The only thing that I have changed is upgrading ESPHome to 1.16.

ogiewon, 2021-02-19:
> I've found a (hopefully) temporary solution : add to the ESPHome configuration "status_use_ping: true"

@Sleinous - can you please explain exactly how you added this to the ESPHome Configuration?  I am running the latest version of Home Assistant on a RPi4, running the HASS image.  I cannot find any 'configuration' for ESPHome where I would add the "status_use_ping: true" option.  In the Supervisor -> AddOns -> ESPHome -> Configuration this is all I see.  

[image]

I don't see any other place to modify the ESPHome Configuration.  Any help would be appreciated.  I did fully remove, reboot, and reinstall ESPHome, but nothing has changed.  I have found a esphome.json file in /config/esphome/.esphome - is that where the configuration for ESPHome is stored?  I tried adding "status_use_ping": true to that file, and then restarted everything, but no change.  I am using a static IP on my ESP32 device.

```
{
  "storage_version": 1,
  "cookie_secret": "1f8ba2689f3e58dfb2e8e090a557a6b6d459d364dc962f0a102a667b4acea82b40338d9311486d4d14bbd4dd30f0a03c21b925b4d8b778413fb2e3f49b9ceb9d",
  "last_update_check": null,
  "remote_version": null,
  "status_use_ping": true
}
```

Sleinous, 2021-02-19:
Hi there,

Here’s my setup for conf:

[cid:image001.png@01D706E7.E37533C0]

[cid:image002.png@01D706E7.E37533C0]

[image]<https://user-images.githubusercontent.com/5206084/108533659-3b01a300-72a7-11eb-8dc1-d497b3141668.png>

ogiewon, 2021-02-19:
@Sleinous - those images did not come through.

Sleinous, 2021-02-19:
Sorry; here you go

[image]

[image]

ogiewon, 2021-02-19:
Interesting...  That entire Configuration Section is simply missing on my system.  Any idea how to get it to show up?  Is there a file for it in your /config/esphome folder?  If so, what is the exact name of that file?  Maybe I just need to add it to kick start the process?  Thank you for you assistance!

Sleinous, 2021-02-19:
I looned around but im not finding any manual yml for this - I have config section by default - very odd!

Télécharger Outlook pour Android<https://aka.ms/ghei36>

ogiewon, 2021-02-19:
Thank you for taking a look.  I'll create a new Issue to see if anyone from the ESPHome Team has any ideas.

ogiewon, 2021-02-20:
@Sleinous - So, after quite a bit of troubleshooting, I think I have found the cause of the "ESP32 Node connection & status dropping from green to red after a few seconds/minutes"....  It appears that a recent UniFi firmware update caused this issue.  I reverted all of my UniFi network Switches and Access Points back to the previous firmware release, and the status of the ESPHome devices now stays green.  I can also ping these devices using the mDNS name of _esp_device_.local.  So, in this case, it appears that the ESPHome release did not cause the issue (at least for me.)  I actually stood up another HomeAssistant RPi, and restored a Snapshot from my main HA system from about 6 weeks ago.  This was still running ESPHome 1.15.x, HassOS 5.10, and an older version of Home Assistant. The problem persisted.  It wasn't until I revered the firmware on my UniFi devices that things started working normally again.

Hope this help you, and anyone else who stumbles upon this.

I still have the issue of the Add-On Configuration Menu being missing, but that appears to be an issue that is being tracked elsewhere.

Sleinous, 2021-02-21:
Hello there, thanks for your reply - im using Unifi access points too, but all sites hace latest firmware and only one site was experiencing this issue.
I ended up bypassing this issue as I have olimex gateway esp32's so they have ethernet ports on the side too. Ive converted all of my gateways into esp32s with lan connectivity instead of wifi back to AP - problem solved
Its odd hiwever that one set of unifi APs in one site hasnt the same problem as the other site also on latest fw

Télécharger Outlook pour Android<https://aka.ms/ghei36>

ogiewon, 2021-02-21:
I was able to make some small UniFi network settings, and then upgrade all of my switches and access points to the latest UniFi Official Releases without any trouble this time.  Not sure what was going on, but all is well here now.  Thanks for your assistance.

arevindh, 2021-05-13:
@ogiewon what changes did you make to Unifi Network ? I have the same issue.

ogiewon, 2021-05-13:
@arevindh  - I am not exactly sure what I changed back then, as I was tweaking many things to try and get things to behave again.  I do believe that simply powering off the UniFi access point, and then restarting it may have helped get things to behave properly again.  Here are the current WiFi settings I use without any trouble.  Hope this helps.  Good luck!

[image]

arevindh, 2021-05-16:
@ogiewon I switched to `mqtt` instead of `api` in config .

bkvargyas, 2021-05-21:
I have a unifi network, and experiencing the same issues.   The firmware of my AP's are 5.63.0

Like others, this issue only started recently.

PablexXXXX, 2021-08-31:
> 
> 
> @ogiewon Nothing fixed the problem, so I switched to `mqtt` instead of `api` in config .

Have this fixed the problem for you?

arevindh, 2021-09-03:
> 
> 
> > @ogiewon Nothing fixed the problem, so I switched to `mqtt` instead of `api` in config .
> 
> Have this fixed the problem for you?

Yes, but now esphome native API is also working fine.