esphome/issues #1813: lamda loop not functioning as expected

danpowpow, 2021-02-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

esphome 1.16 in docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32cam
**ESPHome version (latest production, beta, dev branch)**

1.16
**Affected component:**

https://esphome.io/custom/custom_component.html

**Description of problem:**
i have some fairy/festoon lights, which i'm converting to smart control.  they require pulsed dc control, so i am feeding them power through a L298N H-Bridge. I am polarity reversing the signals into the L298N and this is functioning very nicely when using arduino, however using the equivalent simple loop sequence working in esphome is not working. I can get the speed down to about 50 before it starts messing with the wifi connection, but at speed 1 it simply doesnt work. Does esphome run fast enough to process this loop at a high enough frequency?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

#include "esphome.h"

class MyCustomComponent : public Component {
 public:
  void setup() override {
    pinMode(14, OUTPUT);
    pinMode(13, OUTPUT);
  }
  void loop() override {
  digitalWrite(14, HIGH);
  digitalWrite(13, LOW);
  delay(1);
  digitalWrite(14, LOW);
  digitalWrite(13, HIGH);  
  delay(1);
    }
  };

##########

esphome:
  name: festoon21
  platform: ESP32
  board: esp32cam

  includes:
    - mycustomcomponent.h

wifi:
  ssid: "HomeWifi"
  password: "Jellyfish"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
custom_component:
- lambda: |-
    auto my_custom = new festoon();
    return {my_custom};  

output:
  - platform: ledc
    pin: GPIO12
    id: gpio_12
    frequency: 19531

# Example usage in a light
light:
  - platform: monochromatic
    output: gpio_12
    name: "Festoon21"

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2021-02-09:
what you mean by speed 50 and speed 1? 
(Not familiar with fairy/festoon lights)

danpowpow, 2021-02-09:
hi @glmnet - thanks for the response.
in the .h file, where the custom component code lives, i use a delay function to run the polarity inversion. pin1 hi, pin2 low, delay, then invert so pin1 low, pin2 hi, then delay and loop.
by speed 50 i mean delay(50) in the custom component code, and by speed1 i mean delay(1). on an arduino pro micro, using delay(1) alows the polarity to be reversed/alternated fast enough so that flickering / alternating is not  seen, and the festoon lights to appear all lit at the same time. the esphome code however cannot seem to run fast enough to deal with delay(1). the code runs at delay(50), but so slow that the festoon lights are visibly turning on and off. Does that help?

glmnet, 2021-02-09:
I see, esphome's loop doesn't run continuously as quickly as the Arduino, esp32 is way faster than arduino so...

What is really happening is esphome not executing loop function as quickly as possible, it deliberately "slows" it down, this is per esphome design, however there are other ways to achieve what you're trying to do.

First option will be to ask esphome to run the loop quickly, add a `HighFrequencyLoopRequester high_freq_;` to your custom compoennt and call `start()` method of it

Another option will be to use PWM outputs, since you need alternating outputs, I'd try first to electrically invert the pin, however that will require transistors, resistors, etc.
Second option will be using two pwm outputs, one with the output inverted. I'm not sure that will work.
ESP32 has hardware PWM outputs which can go as slow as 10hz, so if you might want to try 500hz ~ 1khz https://esphome.io/components/output/ledc.html

Last, option but hardest to implement will be to setup a timer interrupt routine to update the outputs as you wish, exactly when it is needed. (This is how software PWM works)

wwebers, 2021-03-26:
Found this statement in the "contribution" section:

> One thing to note in these methods is that anything in loop() or setup() should not block. Specifically methods like delay(10) should be avoided and delays above ~10ms are not permitted. The reason for this is that ESPHome uses a central single-threaded loop for all components - if your component blocks the whole loop will be slowed down.

Than, why not simply using the existing H-Bridge light?

[https://esphome.io/components/light/hbridge.html](https://esphome.io/components/light/hbridge.html)

closed 2021-08-07 by stale[bot]