esphome/issues #1814: 1.16.0 gpio switch is not working after update

estevez-dev, 2021-02-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):** Wemos D1 mini

**ESPHome version (latest production, beta, dev branch)**: 1.16.0

**Affected component:** https://esphome.io/components/switch/gpio.html

**Description of problem:**
After upgrading to 1.16.0 all my boards with a relay connected as `gpio` switch doesn't work with no errors in log.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: multimedia_control
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "********"
  password: "*****************"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Multimedia Control"
    password: "************"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "************"
  services:
    - service: tv_volume_up
      then:
        - remote_transmitter.transmit_samsung:
            data: 0xE0E0E01F
    - service: tv_volume_down
      then:
        - remote_transmitter.transmit_samsung:
            data: 0xE0E0D02F

ota:
  password: "*************"

remote_transmitter:
  pin: D3
  carrier_duty_percent: 50%

remote_receiver:
  pin:
    number: D4
    inverted: true
  dump:
    - samsung

switch:
  - platform: gpio
    id: relay
    pin: D1
    restore_mode: ALWAYS_OFF
  - platform: template
    name: "Amplifier"
    optimistic: true
    id: amplifier
    turn_on_action:
      - switch.turn_on: relay
      - delay: 200ms
      - switch.turn_off: relay
    turn_off_action:
      - switch.turn_on: relay
      - delay: 200ms
      - switch.turn_off: relay

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/multimedia_control.yaml...
INFO Starting log output from multimedia_control.local using esphome API
INFO Connecting to multimedia_control.local:6053 (192.168.2.16)
INFO Successfully connected to multimedia_control.local
[21:35:29][I][app:105]: ESPHome version 1.16.0 compiled on Feb  5 2021, 21:08:49
[21:35:29][C][wifi:443]: WiFi:
[21:35:29][C][wifi:303]:   SSID: [redacted]
[21:35:29][C][wifi:304]:   IP Address: 192.168.2.16
[21:35:29][C][wifi:306]:   BSSID: [redacted]
[21:35:29][C][wifi:307]:   Hostname: 'multimedia_control'
[21:35:29][C][wifi:311]:   Signal strength: -5 dB ▂▄▆█
[21:35:29][C][wifi:315]:   Channel: 1
[21:35:29][C][wifi:316]:   Subnet: 255.255.255.0
[21:35:29][C][wifi:317]:   Gateway: 192.168.2.1
[21:35:29][C][wifi:318]:   DNS1: 192.168.2.1
[21:35:29][C][wifi:319]:   DNS2: (IP unset)
[21:35:29][C][switch.gpio:042]: GPIO Switch 'relay'
[21:35:29][C][switch.gpio:043]:   Pin: GPIO5 (Mode: OUTPUT)
[21:35:29][C][switch.gpio:059]:   Restore Mode: Always OFF
[21:35:29][C][template.switch:058]: Template Switch 'Amplifier'
[21:35:29][C][template.switch:059]:   Restore State: NO
[21:35:29][C][template.switch:060]:   Optimistic: YES
[21:35:29][C][logger:185]: Logger:
[21:35:29][C][logger:186]:   Level: DEBUG
[21:35:29][C][logger:187]:   Log Baud Rate: 115200
[21:35:29][C][logger:188]:   Hardware UART: UART0
[21:35:29][C][remote_transmitter:018]: Remote Transmitter...
[21:35:29][C][remote_transmitter:019]:   Carrier Duty: 50%
[21:35:29][C][remote_transmitter:020]:   Pin: GPIO0 (Mode: OUTPUT)
[21:35:29][C][remote_receiver.esp8266:059]: Remote Receiver:
[21:35:29][C][remote_receiver.esp8266:060]:   Pin: GPIO2 (Mode: INPUT, INVERTED)
[21:35:29][C][remote_receiver.esp8266:065]:   Buffer Size: 1000
[21:35:29][C][remote_receiver.esp8266:066]:   Tolerance: 25%
[21:35:29][C][remote_receiver.esp8266:067]:   Filter out pulses shorter than: 50 us
[21:35:29][C][remote_receiver.esp8266:068]:   Signal is done after 10000 us of no changes
[21:35:29][C][captive_portal:169]: Captive Portal:
[21:35:29][C][ota:029]: Over-The-Air Updates:
[21:35:29][C][ota:030]:   Address: multimedia_control.local:8266
[21:35:29][C][ota:032]:   Using Password.
[21:35:29][C][api:095]: API Server:
[21:35:29][C][api:096]:   Address: multimedia_control.local:6053
[21:35:35][D][switch:025]: 'Amplifier' Turning OFF.
[21:35:35][D][switch:021]: 'relay' Turning ON.
[21:35:35][D][switch:045]: 'relay': Sending state ON
[21:35:35][D][switch:045]: 'Amplifier': Sending state OFF
[21:35:35][D][switch:025]: 'relay' Turning OFF.
[21:35:35][D][switch:045]: 'relay': Sending state OFF

```

**Additional information and things you've tried:**
As you can see from logs, it gets the command from Home Assistant, but the relay is not working. Need to say that the `remote_transmitter` is still working on this device. I have the same issue with another device with a relay.

rradar, 2021-02-06:
Can you switch the relay using the web_server component of esphome @estevez-dev ?

estevez-dev, 2021-02-06:
@rradar , I didn't try that, but looks like the issue was resolved by removing the integration of those two devices come HA and adding it back.

closed 2021-02-14 by estevez-dev