esphome/issues #1815: SyntaxError: invalid syntax

chemelli74, 2021-02-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip (ubuntu)

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

v1.16.0

**Affected component:**

esp32_ble_tracker:

**Description of problem:**
Error on run :" SyntaxError: invalid syntax" in asyncio ( python v3.8.5)

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32-salotto-ble-gateway
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "xxxxxx"
  password: "xxxxxx"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: xxxxxx

ota:

esp32_ble_tracker:

sensor:
  - platform: xiaomi_hhccjcy01
    mac_address: 'xxxxxx'
    temperature:
      name: "Xiaomi HHCCJCY01 Temperature"
    moisture:
      name: "Xiaomi HHCCJCY01 Moisture"
    illuminance:
      name: "Xiaomi HHCCJCY01 Illuminance"
    conductivity:
      name: "Xiaomi HHCCJCY01 Soil Conductivity"
```

**Logs (if applicable):**

```
root@ubuntu:/esphome# esphome esp32-salotto-ble-gateway.yaml run
INFO Reading configuration esp32-salotto-ble-gateway.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d esp32-salotto-ble-gateway
Processing esp32-salotto-ble-gateway (board: nodemcu-32s; framework: arduino; platform: espressif32@1.12.4)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
  File "/usr/local/lib/python3.8/dist-packages/asyncio/base_events.py", line 296

    future = tasks.async(future, loop=self)

                   ^

SyntaxError: invalid syntax

======================================================================================================== [FAILED] Took 2.84 seconds ========================================================================================================
root@ubuntu:/esphome#

```

**Additional information and things you've tried:**

chemelli74, 2021-02-18:
Workarround-ed using esphome docker container

chemelli74, 2021-03-16:
No updates, closing

closed 2021-03-16 by chemelli74

eriknomitch, 2021-07-19:
I am also getting this error:

## Versions
* esphome, version 1.19.4
* PlatformIO Core, version 5.1.1

## Error
```Shell
❯ esphome run status-light.yaml
INFO Reading configuration status-light.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d status-light
Processing status-light (board: lolin32; framework: arduino; platform: platformio/espressif32@3.2.0)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
  File "/Users/erik/.asdf/installs/python/3.8.1/lib/python3.8/site-packages/asyncio/base_events.py", line 296

    future = tasks.async(future, loop=self)

                   ^

SyntaxError: invalid syntax

=========================================================================================== [FAILED] Took 0.12 seconds ===========================================================================================
```

## Device Config
```Yaml
esphome:
  name: status-light
  platform: ESP32
  # board: wemos_d1_mini32
  board: lolin32

<<: !include base.yaml

# Example configuration entry
light:
  - platform: fastled_clockless
    chipset: WS2812
    pin: GPIO23
    num_leds: 16
    rgb_order: GRB
    name: "Status Light RGB Ring"
```

---

I would use the Docker workaround but I need to use the local device for now (e.g., I can't OTA).

Thanks,
Erik

eriknomitch, 2021-07-19:
I got around this by compiling the firmware via Docker then flashing my ESP32 with ESPHome-Flasher but I think the issue should still be open.

dbrand666, 2021-12-09:
The asyncio backport project is no longer required. Remove it with `pip uninstall asyncio`.

OnFreund, 2021-12-10:
Thanks so much @dbrand666!!!