esphome/issues #1817: Nextion Received Unknown Filler end bytes

stevemutch, 2021-02-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio or Pi4

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 mini, also tried same on a nodemcu (ESP8266 and ESP32)

**ESPHome version (latest production, beta, dev branch)**

1.16.0

**Affected component:**

https://www.esphome.io/components/display/nextion.html?highlight=nextion

**Description of problem:**
I've loaded the yaml below to 3 ESP's (Wemos D1 Mini, ESP8266 Nodemcu and an ESP32. All 3 report back page and component id touches on the Nextion 3.2 Enhanced perfectly but on a 2.4 inch basic every press sends back the following:

[W][nextion:145]: Received unknown filler end bytes from Nextion!

and don't confirm back the usual info i get (see below) when a 3.2 enhanced nextion screen is attached to the same node:

[16:50:39][D][nextion:209]: Got touch page=0 component=2 type=PRESS
[16:50:42][D][nextion:209]: Got touch page=0 component=6 type=PRESS

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: nextion24_controller01
  platform: ESP8266
  board: d1_mini
          
wifi:
  ssid: "xxxxxxx"
  password: "xxxxxxx"
  fast_connect: true
    # Enable fallback hotspot (captive portal) in case wifi connection fails

  ap:
    ssid: "Nextion24_01 Fallback Hotspot"
    password: "xxxxxx"  
   
web_server:
  port: 80
  auth:
    username: xxxxxx
    password: xxxxxx    
  
mqtt:
  broker: mqtt.xxxxxx.com
  username: xxxxxx
  password: xxxxxx
  
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  password: "xxxxxx"

ota:
  password: "xxxxxx"

# Uart port definition - This works OK, but also HW UART0 works if Logs are disabled or changed to UART1
uart:
  rx_pin: D4
  tx_pin: D7
  baud_rate: 9600

#Print sensor data in relevant Nextion objects (number type)
display:
  - platform: nextion  
    id: touchpanel
    update_interval: 3s
    brightness: 1
    lambda: |-
      it.set_wait_for_ack(false);
      it.set_component_value("heaterinvalue", id(heaterin).state);  
      it.set_component_value("ambientvalue", id(ambient).state);  
      it.set_component_value("heateroutvalue", id(heaterout).state);  
      it.set_component_value("poolvalue", id(pool).state);  
      it.set_component_value("spavalue", id(spa).state);
      it.set_component_value("pumpvalue", id(pump).state);

#Read sensor data from MQTT
      
sensor:
  - platform: mqtt_subscribe
    id: heaterin
    topic: ds18b20_sensor02/sensor/ds18b20_sensor02/state
    
  - platform: mqtt_subscribe
    id: heaterout
    topic: ds18b20_sensor02/sensor/ds18b20_sensor02/state 
    
  - platform: mqtt_subscribe
    id: spa
    topic: ds18b20_sensor03/sensor/ds18b20_sensor03/state
    
  - platform: mqtt_subscribe
    id: pool
    topic: ds18b20_sensor03/sensor/ds18b20_sensor03/state    
    
  - platform: mqtt_subscribe
    id: ambient
    topic: ds18b20_sensor03/sensor/ds18b20_sensor03/state    
    
  - platform: mqtt_subscribe
    id: pump
    topic: ds18b20_sensor03/sensor/ds18b20_sensor03/state      
    
binary_sensor:
  - platform: nextion
    filters:
      - invert:
      - delayed_on: 50ms
      - delayed_off: 50ms
      - delayed_on_off: 50ms  
    page_id: 1
    component_id: 2
    id: spa_on
    on_press:
      then:
        - mqtt.publish:
            topic: 40a_relay01/switch/40a_relay01/command
            payload: "ON"
        - component.update: touchpanel   
    on_release:
      then:
        - component.update: touchpanel             

  - platform: nextion
    filters:
      - invert:
      - delayed_on: 50ms
      - delayed_off: 50ms
      - delayed_on_off: 50ms    
    page_id: 1
    id: spa_off
    component_id: 3
    on_press:
      then:
        - mqtt.publish:
            topic: 40a_relay01/switch/40a_relay01/command
            payload: "OFF"
        - component.update: touchpanel      
    on_release:
      then:
        - component.update: touchpanel  

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

stratus-ss, 2021-02-14:
For me I only encounter this about 5% of the time, but its enough to not trust whether the device has actually done anything

I have this occasionally with a Nextion 3.5 basic screen sample ESP Log output

```
[16:24:26][W][nextion:145]: Received unknown filler end bytes from Nextion!
[16:24:27][D][nextion:209]: Got touch page=5 component=4 type=PRESS
[16:24:27][D][binary_sensor:036]: 'proj_led_but': Sending state ON
[16:24:27][D][binary_sensor:036]: 'proj_led_but': Sending state OFF
[16:24:28][D][nextion:209]: Got touch page=5 component=4 type=PRESS
[16:24:28][D][binary_sensor:036]: 'proj_led_but': Sending state ON
[16:24:28][D][binary_sensor:036]: 'proj_led_but': Sending state OFF
[16:24:29][W][nextion:145]: Received unknown filler end bytes from Nextion!
[16:24:30][D][nextion:209]: Got touch page=5 component=5 type=PRESS
[16:24:30][D][binary_sensor:036]: 'steve_desk_but': Sending state ON
[16:24:30][D][binary_sensor:036]: 'steve_desk_but': Sending state OFF
[16:24:31][D][nextion:209]: Got touch page=5 component=5 type=PRESS
[16:24:31][D][binary_sensor:036]: 'steve_desk_but': Sending state ON
[16:24:31][D][binary_sensor:036]: 'steve_desk_but': Sending state OFF
[16:24:34][D][nextion:209]: Got touch page=5 component=5 type=PRESS
[16:24:34][D][binary_sensor:036]: 'steve_desk_but': Sending state ON
[16:24:34][D][binary_sensor:036]: 'steve_desk_but': Sending state OFF
[16:24:36][D][nextion:209]: Got touch page=5 component=5 type=PRESS
[16:24:36][D][binary_sensor:036]: 'steve_desk_but': Sending state ON
[16:24:36][D][binary_sensor:036]: 'steve_desk_but': Sending state OFF
[16:24:38][D][nextion:209]: Got touch page=5 component=5 type=PRESS
[16:24:38][D][binary_sensor:036]: 'steve_desk_but': Sending state ON
[16:24:38][D][binary_sensor:036]: 'steve_desk_but': Sending state OFF
[16:24:44][D][nextion:209]: Got touch page=3 component=4 type=PRESS
[16:24:44][D][binary_sensor:036]: 'kettle': Sending state ON
[16:24:44][D][binary_sensor:036]: 'kettle': Sending state OFF
[16:24:46][W][nextion:145]: Received unknown filler end bytes from Nextion!
[16:24:47][D][nextion:209]: Got touch page=3 component=7 type=PRESS
[16:24:47][D][binary_sensor:036]: 'kitchen_fan': Sending state ON
[16:24:47][D][binary_sensor:036]: 'kitchen_fan': Sending state OFF
[16:24:51][D][nextion:209]: Got touch page=3 component=7 type=PRESS
[16:24:51][D][binary_sensor:036]: 'kitchen_fan': Sending state ON
[16:24:51][D][binary_sensor:036]: 'kitchen_fan': Sending state OFF
[16:24:52][D][nextion:209]: Got touch page=3 component=3 type=PRESS
[16:24:52][D][binary_sensor:036]: 'laundry_over': Sending state ON
[16:24:52][D][binary_sensor:036]: 'laundry_over': Sending state OFF
[16:24:54][D][nextion:209]: Got touch page=3 component=3 type=PRESS
[16:24:54][D][binary_sensor:036]: 'laundry_over': Sending state ON
[16:24:54][D][binary_sensor:036]: 'laundry_over': Sending state OFF
[16:25:01][D][nextion:209]: Got touch page=6 component=1 type=PRESS
[16:25:01][D][binary_sensor:036]: 'bath_lights': Sending state ON
[16:25:01][D][binary_sensor:036]: 'bath_lights': Sending state OFF
[16:25:06][D][nextion:209]: Got touch page=6 component=1 type=PRESS
[16:25:06][D][binary_sensor:036]: 'bath_lights': Sending state ON
[16:25:06][D][binary_sensor:036]: 'bath_lights': Sending state OFF
[16:25:08][D][nextion:209]: Got touch page=6 component=1 type=PRESS
[16:25:08][D][binary_sensor:036]: 'bath_lights': Sending state ON
[16:25:08][D][binary_sensor:036]: 'bath_lights': Sending state OFF
[16:25:09][D][nextion:209]: Got touch page=6 component=1 type=PRESS
[16:25:09][D][binary_sensor:036]: 'bath_lights': Sending state ON
[16:25:09][D][binary_sensor:036]: 'bath_lights': Sending state OFF
[16:25:10][D][nextion:209]: Got touch page=6 component=2 type=PRESS
[16:25:10][D][binary_sensor:036]: 'bath_over': Sending state ON
[16:25:10][D][binary_sensor:036]: 'bath_over': Sending state OFF
[16:25:12][W][nextion:145]: Received unknown filler end bytes from Nextion!
```

Here is the yaml:

```
esphome:
  name: main_room_nextion
  platform: ESP8266
  board: d1_mini
 
[... 271 lines cut ...]
  - platform: nextion
    page_id: 6
    component_id: 2
    id: bath_over
    on_press:
      then:
        - homeassistant.service:
            service: input_boolean.toggle
            data:
              entity_id: input_boolean.bathroom_override
        - lambda: |-
            id(bath_over).publish_state(false);
  - platform: nextion
    page_id: 6
    component_id: 3
    id: bath_full
    on_press:
      then:
        - homeassistant.service:
            service: script.turn_on
            data:
              entity_id: script.main_br_full_white
        - lambda: |-
            id(bath_full).publish_state(false);            
```

SenexCrenshaw, 2021-02-19:
You can try setting wait for ack off. 
https://esphome.io/api/classesphome_1_1nextion_1_1_nextion.html#aa94c0403c6a6633fb1d839915f363966

Im writing a massive new driver which fixes and adds a ton of items. Tft Ota updates! 
https://github.com/esphome/esphome/pull/1464

tomjmul, 2021-03-05:
Also having this problem. @SenexCrenshaw great to see the changes coming down the line for Nextion. Is it anyway close to getting merged?

SenexCrenshaw, 2021-03-05:
Working with the esphome devs on that. Hopefully soon!

stevemutch, 2021-03-06:
Excellent news. Can't wait for this. TFT OTA updates will be a game changer for me.

tomjmul, 2021-03-07:
Thank @SenexCrenshaw. I'm testing the new functionality out with a new device I'm starting on and will let you know of any issues found

SenexCrenshaw, 2021-03-12:
Yes please post to the repo

stevemutch, 2021-04-10:
Do you know if this is going to be part of the 1.17 Release?

Thanks

bkaufx, 2021-06-22:
Is this fixed in the new driver update?  The command is coming in perfectly every time, it's just that for some reason there are only 1-2 filler bytes instead of 3 expected.  The Nextion screen is sending 3 but the read_until_ack_ function is only receiving 1 or 2.

I guess for now I'll just edit the nextion.cpp so that it ignores how many filler bytes come in.

cllatser, 2021-06-23:
Same problem with the 4.8" screen, I just have buttons to make them sensors and then in Node-network work with them. But when you click on one of them the error comes out:

[W] [nextion:145]: Received unknown filler end bytes from Nextion!

It would be fantastic to fix it, good job.

Thank you.

cllatser, 2021-06-23:
> Is this fixed in the new driver update?  The command is coming in perfectly every time, it's just that for some reason there are only 1-2 filler bytes instead of 3 expected.  The Nextion screen is [...]

Hello, can you explain to me how the nextion.cpp file is edited? Thank you.

bkaufx, 2021-06-23:
You just have to find out where python/pip puts it.  For me on linux its ~/.local/lib/python3.8/site-packages/esphome/components/nextion.

cllatser, 2021-06-26:
Good morning, I wanted to comment, that I have raised the baud to 115200 and I have to say that I no longer get so many errors "Nextion Received Unknown Filler end bytes". In normal use I have lowered them to less than 1%. Now I'm happy. Thanks a lot.

closed 2021-11-26 by github-actions[bot]