esphome/issues #1818: esp32_ble_beacon does not work

VanGog86, 2021-02-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
 pip install

**ESP (ESP32/ESP8266, Board/Sonoff):**
 esp32doit-devkit-v1

**ESPHome version (latest production, beta, dev branch)**
 Current version: 1.16.0

**Affected component:**
 https://esphome.io/components/esp32_ble_beacon.html

**Description of problem:**
After successful flashing, my iphone does not detect the ibeacon. I check through the bluetooth screen and don't see any new devices there. through the bleScanner application I see a new device but it is defined as N / A but not as a ibeacon

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: ble
  platform: ESP32
  board: esp32doit-devkit-v1

esp32_ble_beacon :
  type : iBeacon 
  uuid : '12a13a53-905b-0654-3468-3956014a7888'
  major : 234
  minor : 138
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

VanGog86, 2021-02-06:
I saw a similar problem, but I do not understand what I need to do https://github.com/esphome/issues/issues/243

VanGog86, 2021-02-27:
really no one faced such a problem?

koenvervloesem, 2021-03-06:
I just tried this, and on my Android phone the app [Beacon Scanner](https://play.google.com/store/apps/details?id=com.bridou_n.beaconscanner) finds the beacon and shows the correct UUID, major and minor. This is done on a LilyGO TTGO ESP32 T-Display.

patrickdk77, 2021-03-20:
This is really simple to resolve, ibeacons are broken due to #1844

gogoizzy, 2021-04-14:
Broken for me too

chrismetcalf, 2021-06-07:
I'm seeing the same issue on 1.19.0-dev. Compiles and loads fine, debug output looks fine, including esp32_ble_beacon reporting its major and minor number, but I can't discover it in any of the apps I've tried on iOS.

patrickdk77, 2021-06-07:
You are going have to open your own bug instead of using this one. This bug is about receiving beacons, not transmitting them.

closed 2021-10-20 by stale[bot]