esphome/issues #1820: PN532_SPI NFC/RFID does not work

gkfx01, 2021-02-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
 Home Assistant OS 5.10

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 / Wemos d1 mini

**ESPHome version (latest production, beta, dev branch)**
1.16.0

**Affected component:**
https://esphome.io/components/binary_sensor/pn532.html

**Description of problem:**
I'm using Pn532 controller with SPI. It worked well with esphome 1.15.3.
It hasn't worked since the update. I changed PN532 to PN532_SPI In the configuration but it doesn't do anything.
When I put the TAG there, nothing happens. The log is empty and nothing appears in it.
I put back 1.15.3 with it working fine again.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
spi:
  clk_pin: D0 
  miso_pin: D1 
  mosi_pin: D2 

pn532_spi:
  cs_pin: D3
  update_interval: 1s
  on_tag:
    then:
      - homeassistant.tag_scanned: !lambda 'return x;'
      - delay: 50ms
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

techmatik, 2021-02-07:
I have the same problem.

koenvervloesem, 2021-02-09:
This has been fixed by https://github.com/esphome/esphome/pull/1511. I can confirm that the fix works, just tried it two days ago with esphome-dev.

glmnet, 2021-02-10:
Duplicate of #1654

closed 2021-02-14 by jesserockz