esphome/issues #1822: pn532 doesn't communicate tag gone event

PaulAntonDeen, 2021-02-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home assistant 2021.02.1 with ESPHome 1.16.0

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 D1 mini

**ESPHome version (latest production, beta, dev branch)**

1.16.0

**Affected component:**

https://github.com/esphome/esphome/blob/bd66ea5740c5b175859df2ddde99fff33bfdfa3c/esphome/components/pn532/pn532.cpp

**Description of problem:**
The tag reader only communicates when a new tag has been found and does not communicate the removal of a tag. There is a function block in there that makes sure it does not repeatedly fire an event when the same tag is still present, but the first time that condition is no longer true it should fire a tag_gone event.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
pn532_i2c:
  id: pn532_board
  on_tag:
    then:
    - homeassistant.tag_scanned: !lambda 'return x;'
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

jesserockz, 2021-02-09:
Hi @PaulAntonDeen This should be a feature request, not an issue. But also there is an open PR to add this functionality: esphome/esphome#1436

jesserockz, 2021-02-09:
Feature request is here: esphome/feature-requests#172

closed 2021-02-09 by jesserockz