esphome/issues #1824: ST7735 + D1 mini = boot loop

franz2lyon, 2021-02-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on NUC

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini

**ESPHome version (latest production, beta, dev branch)**

v1.16.0

**Affected component:**

https://esphome.io/components/display/st7735.html

**Description of problem:**
After uploading, D1 mini is in boot loop and screen remains white

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test_st7735
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "Bbox-xxxxxxxxxx"
  password: "xxxxxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Tft Fallback Hotspot"
    password: "pHSxsI5yO44Y"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

spi:
  clk_pin: D5
  mosi_pin: D7

color:
  - id: my_red
    red: 100%
    green: 0%
    blue: 0%
  - id: my_yellow
    red: 100%
    green: 100%
    blue: 0%
  - id: my_green
    red: 0%
    green: 100%
    blue: 0%
  - id: my_blue
    red: 0%
    green: 0%
    blue: 100%
  - id: my_gray
    red: 50%
    green: 50%
    blue: 50%
[... 379 lines cut ...]
[23:46:10]3fffff80:  3fff26cc 00000014 3ffffed0 feefeffe  
[23:46:10]3fffff90:  feefeffe feefeffe feefeffe 3fff0a14  
[23:46:10]3fffffa0:  3fffdad0 00000000 3fff09d4 4021c5f4  
WARNING Decoded 0x4021c5f4: loop_wrapper() at core_esp8266_main.cpp
[23:46:11]3fffffb0:  feefeffe feefeffe 3ffe8520 40100675  
WARNING Decoded 0x40100675: cont_wrapper
[23:46:11]<<<stack<<<
[23:46:11]
[23:46:11]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[23:46:11]
[23:46:11] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[23:46:11]
[23:46:11]load 0x4010f000, len 3584, room 16 
[23:46:11]tail 0
[23:46:11]chksum 0xb0
[23:46:11]csum 0xb0
[23:46:11]v2843a5ac
[23:46:11]~ld
[23:46:11][I][logger:166]: Log initialized
[23:46:11][C][ota:366]: There have been 1 suspected unsuccessful boot attempts.
[23:46:11][I][app:029]: Running through setup()...
```

**Additional information and things you've tried:**
I've tried to program the D1 mini with an .ino using Adafruit_ST7735 and Adafruit_GFX librairies, and it works. So the D1 mini can drive the ST7735.

franz2lyon, 2021-02-07:
[image]

The model of tft display I use

rradar, 2021-02-07:
I can confirm this. Same for me with a 128x128 spi version (red) of the display.

Device boot loops with and without display attached:

```yaml
[21:30:03]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[21:30:03]
[21:30:03]Exception (28):
WARNING Exception type: Access to invalid address: LOAD (wild pointer?)
[21:30:03]epc1=0x4020d041 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d ./build/thermostat -t idedata
WARNING Decoded 0x4020d041: unsigned char esphome::spi::SPIComponent::transfer_<(esphome::spi::SPIBitOrder)1, (esphome::spi::SPIClockPolarity)0, (esphome::spi::SPIClockPhase)0, true, true>(unsigned char)
[21:30:05]
[21:30:05]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[21:30:05]
[21:30:05]ctx: cont
[21:30:05]sp: 3ffffb50 end: 3fffffc0 offset: 0190
[21:30:05]3ffffce0:  007a1200 00000000 3fff2af4 4020d03f  
WARNING Decoded 0x4020d03f: unsigned char esphome::spi::SPIComponent::transfer_<(esphome::spi::SPIBitOrder)1, (esphome::spi::SPIClockPolarity)0, (esphome::spi::SPIClockPhase)0, true, true>(unsigned char)
[21:30:05]3ffffcf0:  40105cfd 000453a4 3fff1444 40264a73  
WARNING Decoded 0x40105cfd: ets_timer_arm_new
WARNING Decoded 0x40264a73: sleep_reset_analog_rtcreg_8266
[21:30:05]3ffffd00:  00000003 40264a72 3fff317c 4020d492  
WARNING Decoded 0x40264a72: sleep_reset_analog_rtcreg_8266
WARNING Decoded 0x4020d492: esphome::st7735::ST7735::senddata_(unsigned char const*, unsigned char)
[21:30:05]3ffffd10:  0000000e 000000b1 3fff317c 000000ff  
[21:30:05]3ffffd20:  00000003 40264a72 3fff317c 4020d550  
WARNING Decoded 0x40264a72: sleep_reset_analog_rtcreg_8266
WARNING Decoded 0x4020d550: esphome::st7735::ST7735::sendcommand_(unsigned char, unsigned char const*, unsigned char)
[21:30:05]3ffffd30:  0000000d 40264a72 00000003 4020d5a4  
WARNING Decoded 0x40264a72: sleep_reset_analog_rtcreg_8266
WARNING Decoded 0x4020d5a4: esphome::st7735::ST7735::display_init_(unsigned char const*)
[21:30:05]3ffffd40:  3fff317c 00000003 00000064 40228264  
WARNING Decoded 0x40228264: esp_yield
[21:30:05]3ffffd50:  00000000 0000004c 00000040 00000040  
[21:30:05]3ffffd60:  3fff317c 3ffe9c28 3fff317c 4020d690  
WARNING Decoded 0x4020d690: esphome::st7735::ST7735::setup()
[21:30:05]3ffffd70:  3ffe964f 0000000f 3fff2bfc 4020e7ce  
WARNING Decoded 0x4020e7ce: esphome::version::VersionTextSensor::setup()
[21:30:05]3ffffd80:  3fff41d0 00000010 3fff317c 402152ec  
WARNING Decoded 0x402152ec: esphome::PollingComponent::call_setup()
[21:30:05]3ffffd90:  3fff2d7c 3fff36fc 3fff0edc 4021692d  
WARNING Decoded 0x4021692d: esphome::Scheduler::process_to_add()
[21:30:05]3ffffda0:  00000000 00000000 3fff0edc 4022b684  
WARNING Decoded 0x4022b684: esphome::Component::call_setup()
[21:30:05]3ffffdb0:  3fff3e64 00000010 3fff0edc 4022b705  
WARNING Decoded 0x4022b705: esphome::Component::call()
[21:30:05]3ffffdc0:  3fff2c2c 0000000f 3fff0edc 40215039  
WARNING Decoded 0x40215039: esphome::Application::setup()
[21:30:05]3ffffdd0:  3fff3e64 0000001c 00000020 40101596  
WARNING Decoded 0x40101596: malloc
[21:30:05]3ffffde0:  3fffff00 3fffff00 3ffea9ef 40237a7a  
WARNING Decoded 0x40237a7a: std::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string() at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:541
[21:30:05]3ffffdf0:  3fffff70 3fff15d0 00000010 3ffffef8  
[21:30:05]3ffffe00:  3fffff00 3fff2f1c 3fff3e4c 40218cbe  
WARNING Decoded 0x40218cbe: setup
[21:30:05]3ffffe10:  00000001 00000000 feefeffe feefeffe  
[21:30:05]3ffffe20:  3fff27d0 00000000 00000000 3fff27f8  
[21:30:05]3ffffe30:  00000000 00000000 00000000 4022d968  
WARNING Decoded 0x4022d968: precache
[21:30:05]3ffffe40:  00000000 4022d968 00000000 4022d968  
WARNING Decoded 0x4022d968: precache
WARNING Decoded 0x4022d968: precache
[21:30:05]3ffffe50:  00000000 4022d968 00000000 4022d968  
WARNING Decoded 0x4022d968: precache
WARNING Decoded 0x4022d968: precache
[21:30:05]3ffffe60:  00000000 00000000 3fff2820 00000000  
[21:30:05]3ffffe70:  00000000 3fff2838 00000000 00000000  
[21:30:05]3ffffe80:  00000000 4022d968 00000000 4022d968  
WARNING Decoded 0x4022d968: precache
WARNING Decoded 0x4022d968: precache
[21:30:05]3ffffe90:  00000000 4022d968 00000000 4022d968  
WARNING Decoded 0x4022d968: precache
WARNING Decoded 0x4022d968: precache
[21:30:05]3ffffea0:  00000000 4022d968 00000000 00000000  
WARNING Decoded 0x4022d968: precache
[21:30:05]3ffffeb0:  3fff2790 00000000 00000000 3fff27a8  
[21:30:05]3ffffec0:  00000000 00000000 00000000 4022d968  
WARNING Decoded 0x4022d968: precache
[21:30:05]3ffffed0:  00000000 4022d968 00000000 4022d968  
WARNING Decoded 0x4022d968: precache
WARNING Decoded 0x4022d968: precache
[21:30:05]3ffffee0:  00000000 4022d968 00000000 4022d968  
WARNING Decoded 0x4022d968: precache
WARNING Decoded 0x4022d968: precache
[21:30:05]3ffffef0:  00000000 00000000 3fff3ce4 00000001  
[21:30:05]3fffff00:  3fff15d0 00000000 00000000 00000000  
[21:30:05]3fffff10:  00000000 3fff15d0 706f6e6d 00000000  
[21:30:05]3fffff20:  78777675 3fff3ca8 3fff3b60 3fff3134  
[21:30:05]3fffff30:  3fff30f8 3fff3094 3fff300c 3fff2f1c  
[21:30:05]3fffff40:  3fff2ed4 3fff2e94 3fff2e44 3fff2ddc  
[21:30:05]3fffff50:  3fff2d2c 3fff2cb4 3fff2c44 3fff2b84  
[21:30:05]3fffff60:  3fff2b1c 3fff3234 3fff2bfc 3fff3df0  
[21:30:05]3fffff70:  feefeffe feefeffe feefeffe feefeffe  
[21:30:05]3fffff80:  3fffff6c 00000000 00000001 feefeffe  
[21:30:05]3fffff90:  feefeffe feefeffe feefeffe 3fff12b4  
[21:30:05]3fffffa0:  3fffdad0 00000000 3fff1274 40228374  
WARNING Decoded 0x40228374: loop_wrapper() at core_esp8266_main.cpp
[21:30:05]3fffffb0:  feefeffe feefeffe 3ffe856c 40100809  
WARNING Decoded 0x40100809: cont_wrapper
[21:30:05]<<<stack<<<
```

rradar, 2021-02-07:
Tag for ST7735 doesn't exist. Manual codeowner pinging :satellite:  @SenexCrenshaw :wink:

SenexCrenshaw, 2021-02-12:
Can you folks try the latest in dev?

adamcodemaker, 2021-02-12:
same here
I have Joy-It ST7735 1.8 128x160
It also works ok with Adafruit_ST7735

SenexCrenshaw, 2021-02-13:
So strange... I will dig in to it. 
What model? INITR_BLACKTAB? 
There are a few you can try. There are a lot of different copy cats of these boards
'INITR_GREENTAB',    'INITR_REDTAB',    'INITR_BLACKTAB',    'INITR_MINI160X80',    'INITR_18BLACKTAB',    'INITR_18REDTAB'

What did you try with the Adafruit one?

adamcodemaker, 2021-02-13:
I was using INITR_BLACKTAB from Adafruit

SenexCrenshaw, 2021-02-13:
Ok i will wire one up and try.

adamcodemaker, 2021-02-13:
I switched my board to esp32doit-devkit-v1 and it works :)

SenexCrenshaw, 2021-02-13:
yes esp32 works fine. there is an issue with 8266

SenexCrenshaw, 2021-02-13:
Found it
Line 410.
``    this->transfer_byte(pgm_read_byte(data_bytes++));  // write byte - SPI library``
to
``    this->write_byte(pgm_read_byte(data_bytes++));  // write byte - SPI library``

The transfer_byte works differently I suppose on the 8266. We didnt need the return anyway so write_byte is the way to go. 

PR submitted. Thanks for helping everyone
https://github.com/esphome/esphome/pull/1529

jesserockz, 2021-02-13:
I just tried the st7735 on my d1 mini 8266 and it works fine with the code in `dev`

```yaml
display:
  - platform: st7735
    model: INITR_BLACKTAB
    cs_pin: D8
    dc_pin: D1
    reset_pin: D2
    deviceheight: 160
    devicewidth: 128
    rowstart: 0
    colstart: 0
    eightbitcolor: true
    rotation: 270
    update_interval: 10s
    lambda: |-
      it.rectangle(0, 20, it.get_width(), it.get_height() - 20, id(my_blue));

      it.strftime((160 / 2), (128 / 3) * 1 + 5, id(font_24), COLOR_WHITE, TextAlign::CENTER, "%Y-%m-%d", id(esptime).now());
      it.strftime((160 / 2), (128 / 3) * 2 + 5, id(font_48), COLOR_WHITE, TextAlign::CENTER, "%H:%M", id(esptime).now());
      it.print(5, 5, id(font_12), id(my_yellow), TextAlign::TOP_LEFT, "ESPHome");

      if (id(system_status).state) {
        it.print(140, 5, id(font_12), id(my_green), TextAlign::TOP_RIGHT, "Online");
      } else {
        it.print(140, 5, id(font_12), id(my_red), TextAlign::TOP_RIGHT, "Offline");
      }
```

jesserockz, 2021-02-13:
Just looked at the original config in this issue, I think its because there is no MISO pin on SPI which is causing the `transfer` to crash when trying to `receive`. Just tested it and confirmed this on my side. The PR above will fix this, but also you can just set the `miso` pin on SPI to fix until that PR is released. (You don't actually have to wire the miso pin to anything.)

closed 2021-02-13 by jesserockz

franz2lyon, 2021-02-13:
> Just looked at the original config in this issue, I think its because there is no MISO pin on SPI which is causing the `transfer` to crash when trying to `receive`. Just tested it and confirmed this [...]

I try this immediately.

franz2lyon, 2021-02-13:
> I try this immediately.

I confirm it works after adding miso_pin: D6, without wiring anything on it.

Thanks a lot !

jesserockz, 2021-02-13:
I have also merged the PR that fixes this without the miso pin into dev and will release it in `1.16.1`