esphome/issues #1827: IR Receiver - codes not being recognized (buffer problems?)

Siridar, 2021-02-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-WROVER
**ESPHome version (latest production, beta, dev branch)**

1.16.0
**Affected component:**

https://esphome.io/components/remote_receiver.html

**Description of problem:**
Description of problem:
After receiving several - codes are not being recognized anymore.
Number of correctly recognized codes, after a reboot, seems to be strictly linked to the buffer_size
Increasing buffer_size solves the problem partially because after a (longer while) it fails to get the codes correctly anyway.
My guess is that has something to do with buffer overflow - received data being truncated or scrambled.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: livingroom_ir
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: ---
  password: ---

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ---
    password: ---

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ---

ota:
  password: ---

remote_receiver:
  pin:
    number: GPIO4
    inverted: true
    mode: INPUT_PULLUP
  dump: all
  buffer_size: 1kb
  idle: 25ms
  filter: 250us

```

**Logs (if applicable):**

```
    [12:19:04][C][remote_receiver.esp32:054]: Remote Receiver:
    [12:19:04][C][remote_receiver.esp32:055]: Pin: GPIO4 (Mode: INPUT_PULLUP, INVERTED)
    [12:19:04][C][remote_receiver.esp32:060]: Channel: 0
    [12:19:04][C][remote_receiver.esp32:061]: RMT memory blocks: 3
    [12:19:04][C][remote_receiver.esp32:062]: Clock divider: 80
    [12:19:04][C][remote_receiver.esp32:063]: Tolerance: 25%
    [12:19:04][C][remote_receiver.esp32:064]: Filter out pulses shorter than: 250 us
[... 63 lines cut ...]
    [12:19:26][D][remote.raw:041]: 553, -654, 554, -680, 478, -682, 2997, -649, 561, -622, 1103, -676, 557, -631, 578, -633, 1199, -598, 558, -630, 609, -622, 1150, -701, 587, -627, 582, -600, 584, -625, 2992, -677, 611, -622, 1205, -622, 582, -653, 532, -702, 1178, -625, 609
    [12:19:26][D][remote.raw:028]: Received Raw: 2288, -681, 1118, -683, 1148, -704, 504, -682, 477, -682, 1121, -658, 553, -657, 527, -680, 1119, -710, 553, -654, 554, -680, 478, -682, 2964, -658, 1170, -659, 1166, -706, 530, -706, 502, -684, 1147, -654, 505, -656, 552, -682, 1072, -655,
    [12:19:26][D][remote.raw:041]: 578, -607, 552, -681, 527, -681, 2937, -677, 611, -622, 1205, -622, 582, -653, 532, -702, 1178, -625, 609, -624, 585, -624, 1151, -627, 528, -657, 532, -652, 610, -626, 2971, -652, 561, -625, 1154, -623, 579, -634, 606, -650, 1125, -677, 610
    [12:19:26][D][remote.raw:028]: Received Raw: 2335, -658, 1170, -659, 1166, -706, 530, -706, 502, -684, 1147, -654, 505, -656, 552, -682, 1072, -655, 578, -607, 552, -681, 527, -681, 2937, -677, 611, -622, 1205, -622, 582, -653, 532, -702, 1178, -625, 609, -624, 585, -624, 1151, -627,
    [12:19:27][D][remote.raw:041]: 528, -657, 532, -652, 610, -626, 2971, -652, 561, -625, 1154, -623, 579, -634, 606, -650, 1125, -677, 610, -626, 604, -627, 1180, -627, 529, -681, 580, -625, 559, -626, 2974, -624, 582, -627, 1126, -653, 555, -628, 508, -679, 1151, -626, 557
    [12:19:30][D][remote.raw:028]: Received Raw: 2338, -654, 554, -659, 1146, -678, 532, -628, 581, -604, 1126, -655, 529, -632, 578, -627, 1203, -626, 581, -628, 606, -629, 556, -677, 2949, -652, 561, -625, 1154, -623, 579, -634, 606, -650, 1125, -677, 610, -626, 604, -627, 1180, -627,
    [12:19:30][D][remote.raw:041]: 529, -681, 580, -625, 559, -626, 2974, -624, 582, -627, 1126, -653, 555, -628, 508, -679, 1151, -626, 557, -631, 604, -628, 1172, -680, 582, -630, 553, -652, 533, -651, 2945, -655, 556, -629, 1149, -603, 554, -632, 527, -657, 1174, -654, 554
    [12:19:30][D][remote.raw:028]: Received Raw: 2312, -682, 530, -653, 1122, -655, 584, -627, 552, -682, 1152, -603, 581, -602, 529, -681, 1124, -656, 579, -629, 606, -629, 554, -680, 2951, -624, 582, -627, 1126, -653, 555, -628, 508, -679, 1151, -626, 557, -631, 604, -628, 1172, -680,
    [12:19:30][D][remote.raw:041]: 582, -630, 553, -652, 533, -651, 2945, -655, 556, -629, 1149, -603, 554, -632, 527, -657, 1174, -654, 554, -679, 556, -679, 1172, -654, 581, -653, 531, -656, 551, -660, 2992, -677, 533, -652, 1124, -656, 552, -659, 551, -637, 1193, -603, 555
    [12:19:30][D][remote.raw:028]: Received Raw: 2363, -675, 509, -654, 1120, -656, 556, -629, 532, -653, 1150, -629, 554, -633, 554, -655, 1147, -679, 581, -629, 606, -628, 553, -655, 2947, -655, 556, -629, 1149, -603, 554, -632, 527, -657, 1174, -654, 554, -679, 556, -679, 1172, -654,
    [12:19:30][D][remote.raw:041]: 581, -653, 531, -656, 551, -660, 2992, -677, 533, -652, 1124, -656, 552, -659, 551, -637, 1193, -603, 555, -657, 580, -628, 1171, -679, 557, -681, 528, -654, 556, -653, 2971, -628, 557, -603, 1174, -629, 552, -658, 605, -654, 1147, -658, 576
    [12:19:30][D][remote.raw:028]: Received Raw: 2449, -625, 561, -603, 1174, -624, 557, -655, 608, -623, 1205, -623, 609, -605, 605, -650, 1126, -702, 559, -656, 551, -651, 610, -626, 2974, -677, 533, -652, 1124, -656, 552, -659, 551, -637, 1193, -603, 555, -657, 580, -628, 1171, -679,
    [12:19:30][D][remote.raw:041]: 557, -681, 528, -654, 556, -653, 2971, -628, 557, -603, 1174, -629, 552, -658, 605, -654, 1147, -658, 576, -656, 552, -657, 1120, -657, 528, -659, 526, -658, 577, -656, 2968, -679, 507, -654, 1149, -631, 576, -658, 578, -678, 1150, -678, 581
    [12:19:30][D][remote.raw:028]: Received Raw: 2324, -701, 559, -601, 1176, -603, 580, -632, 611, -620, 1102, -677, 616, -550, 623, -579, 1185, -596, 587, -601, 609, -620, 589, -646, 3007, -649, 610, -621, 1207, -620, 572, -644, 607, -603, 1201, -620, 563, -646, 589, -596, 1182, -601,
    [12:19:30][D][remote.raw:041]: 584, -600, 610, -621, 534, -677, 2999, -679, 507, -654, 1149, -631, 576, -658, 578, -678, 1150, -678, 581, -627, 555, -679, 1125, -653, 481, -681, 581, -626, 530, -680, 2972, -603, 554, -654, 1097, -658, 555, -631, 578, -632, 1119, -730, 530
    [12:19:31][D][remote.raw:028]: Received Raw: 2366, -649, 610, -621, 1207, -620, 572, -644, 607, -603, 1201, -620, 563, -646, 589, -596, 1182, -601, 584, -600, 610, -621, 534, -677, 3032, -623, 586, -597, 1183, -600, 608, -622, 574, -639, 1123, -677, 539, -626, 608, -621, 1207, -601,
    [12:19:31][D][remote.raw:041]: 528, -676, 614, -624, 609, -621, 2980, -603, 554, -654, 1097, -658, 555, -631, 578, -632, 1119, -730, 530, -658, 602, -629, 1149, -676, 532, -679, 553, -658, 579, -652, 2896, -653, 529, -655, 1098, -707, 579, -653, 529, -682, 1199, -627, 555
    [12:19:31][D][remote.raw:028]: Received Raw: 2369, -623, 586, -597, 1183, -600, 608, -622, 574, -639, 1123, -677, 539, -626, 608, -621, 1207, -601, 528, -676, 614, -624, 609, -621, 3027, -625, 610, -623, 1205, -600, 608, -625, 582, -653, 1203, -624, 610, -624, 582, -632, 1198, -626,
    [12:19:31][D][remote.raw:041]: 581, -653, 557, -654, 582, -626, 2920, -653, 529, -655, 1098, -707, 579, -653, 529, -682, 1199, -627, 555, -680, 550, -660, 1145, -681, 581, -652, 556, -605, 529, -654, 559
    [12:19:31][D][remote.raw:028]: Received Raw: 2412, -625, 610, -623, 1205, -600, 608, -625, 582, -653, 1203, -624, 610, -624, 582, -632, 1198, -626, 581, -653, 557, -654, 582, -626, 2920, -653, 529, -655, 1098, -707, 579, -653, 529, -682, 1199, -627, 555, -680, 550, -660, 1145, -681,
    [12:19:31][D][remote.raw:041]: 581, -652, 556, -605, 529, -654, 559
    [12:19:51][D][remote.raw:041]: Received Raw: 632, -655, 1098, -707, 579
```

**Additional information and things you've tried:**

thedishking, 2021-02-20:
I am having the same issue.
I apologize if I am not formatting my log correctly.  Fairly new here and still learning :)
I've replicated on several ESP32 boards, including Lolin32 and NodeMCU-32S, and wESP32.

Initially codes are recognized as expected.  After a bunch of presses, the codes stop being recognized.  This can be replicated within 2 minutes, and can be seen using logs.

The only way to get unit to properly recognize codes again is to power cycle ESP32.

'[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 8': Sending state ON
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 8': Sending state OFF
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 9': Sending state ON
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 9': Sending state OFF
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 9': Sending state ON
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 9': Sending state OFF
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 9': Sending state ON
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 9': Sending state OFF
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 9': Sending state ON
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 9': Sending state OFF
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 0': Sending state ON
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 0': Sending state OFF
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 0': Sending state ON
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 0': Sending state OFF
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 0': Sending state ON
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 0': Sending state OFF
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 0': Sending state ON
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 0': Sending state OFF
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 0': Sending state ON
[11:20:07][D][binary_sensor:036]: 'R1 Sony TV 0': Sending state OFF
[11:20:08][D][binary_sensor:036]: 'R1 Sony TV Picture Mode': Sending state ON
[11:20:08][D][binary_sensor:036]: 'R1 Sony TV Picture Mode': Sending state OFF
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Picture Mode': Sending state ON
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Picture Mode': Sending state OFF
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Picture Mode': Sending state ON
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Picture Mode': Sending state OFF
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Picture Mode': Sending state ON
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Picture Mode': Sending state OFF
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Picture Mode': Sending state ON
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Picture Mode': Sending state OFF
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Up': Sending state ON
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Up': Sending state OFF
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Up': Sending state ON
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Up': Sending state OFF
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Up': Sending state ON
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Up': Sending state OFF
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Up': Sending state ON
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Up': Sending state OFF
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Up': Sending state ON
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Up': Sending state OFF
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Down': Sending state ON
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Down': Sending state OFF
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Down': Sending state ON
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Down': Sending state OFF
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Down': Sending state ON
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Down': Sending state OFF
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Down': Sending state ON
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Down': Sending state OFF
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Down': Sending state ON
[11:20:09][D][binary_sensor:036]: 'R1 Sony TV Volume Down': Sending state OFF
[11:20:10][D][remote.raw:028]: Received Raw: 2431, -585, 605, -594, 595, -604, 596, -604, 596, -603, 1197, -595, 595, -604, 596, -603, 1197, -594, 596, -604, 596, -603, 597, -603, 3075, -590, 1200, -592, 1198, -593, 597, -577, 623, -576, 1224, -593, 597, -577, 623, -576, 1224, -593, 
[11:20:10][D][remote.raw:041]:   597, -577, 623, -576, 624, -576, 3108, -585, 1205, -586, 1204, -587, 603, -596, 604, -571, 1229, -587, 603, -596, 604, -571, 1229, -587, 603, -571, 629, -571, 629, -570, 3110, -589, 1201, -590, 1200, -591, 599, -576, 624, -575, 1225, -591, 599
[11:20:10][D][remote.raw:028]: Received Raw: 2424, -592, 598, -601, 598, -602, 598, -601, 599, -601, 1199, -592, 598, -601, 599, -601, 1199, -592, 598, -601, 598, -602, 598, -601, 3083, -585, 1205, -586, 1204, -587, 603, -596, 604, -571, 1229, -587, 603, -596, 604, -571, 1229, -587, 
[11:20:10][D][remote.raw:041]:   603, -571, 629, -571, 629, -570, 3110, -589, 1201, -590, 1200, -591, 599, -576, 624, -575, 1225, -591, 599, -576, 624, -575, 1225, -591, 599, -601, 599, -575, 625, -574, 3103, -566, 624, -575, 1225, -567, 623, -576, 624, -575, 1225, -567, 623
[11:20:10][D][remote.raw:028]: Received Raw: 2426, -565, 625, -600, 600, -599, 601, -598, 601, -599, 1201, -590, 600, -599, 601, -599, 1201, -590, 600, -599, 601, -599, 600, -599, 3081, -589, 1201, -590, 1200, -591, 599, -576, 624, -575, 1225 [... cut ...]
[11:20:10][D][remote.raw:028]: Received Raw: 2426, -565, 1225, -591, 599, -601, 599, -600, 600, -599, 1201, -591, 599, -600, 600, -599, 1200, -592, 598, -601, 599, -600, 600, -600, 3082, -587, 602, -572, 1228, -588, 602, -572, 628, -572, 1228, -588, 602, -572, 628, -572, 1228, -588, 
[11:20:10][D][remote.raw:041]:   602, -572, 628, -572, 628, -571, 3106, -591, 1199, -593, 1206, -585, 595, -579, 621, -578, 1232, -585, 605, -569, 621, -578, 1232, -585, 605, -569, 620, -579, 621, -579, 3101, -589, 1201, -590, 1200, -591, 598, -576, 624, -576, 1224, -592, 598'

thedishking, 2021-06-21:
This issue is still happening. 
I have noticed I can get the Esp8266 to work, and this appears to happen only on Esp32. 

It would be nice to have it fixed, to allow us to move to hard wired Ethernet Esp32 units.

closed 2021-11-04 by github-actions[bot]