esphome/issues #1828: Remote transmitter is not working after update [1.15.3]->[1.16.0]

mihing, 2021-02-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant OS 5.11

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**ESPHome version (latest production, beta, dev branch)**

1.16.0

**Affected component:**

https://esphome.io/components/remote_transmitter.html#remote-transmitter-transmit-rc-switch-raw-action

**Description of problem:**
After updating ESPHome to version 1.16.0, 433MHz transmitter stopped working.
The console looks like everything is in order, but the controlled devices stopped responding to commands.
I rolled back to version 1.15.3 and everything worked again.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: nodemcuv3_rf433
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret Wifi_SSID
  password: !secret Wifi_PWD
  fast_connect: on

logger:
      esp8266_store_log_strings_in_flash: false

api:
  password: !secret HassPass

ota:
  password: !secret HassPass

web_server:
  port: 80

remote_receiver:
  pin:
    number: D1
    #inverted: True
    #mode: INPUT_PULLUP
  dump:
    - rc_switch
  # Settings to optimize recognition of RF devices
  tolerance: 50%
  filter: 250us
  idle: 4ms
  buffer_size: 2kb
      
remote_transmitter:
  pin:
    number: D2
    #inverted: True
    #mode: INPUT_PULLUP
  # RF uses a 100% carrier signal
  carrier_duty_percent: 100%
  
switch:

  - platform: template
    name: "RF Power Socket 1"
    icon: "mdi:power-socket-de"
    id: RFSocket1
    
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '101100000111110101011111'
          protocol: 1
          repeat:
            times: 5
            wait_time: 50ms
      - switch.template.publish:
          id: RFSocket1
          state: ON
    
    turn_off_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '101100000111110101010111'
          protocol: 1
          repeat:
            times: 5
            wait_time: 50ms
      - switch.template.publish:
          id: RFSocket1
          state: OFF
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

lipiec2, 2021-02-09:
How to downgrade 1.16.0 --> 1.15.3?

mihing, 2021-02-09:
> 
> 
> How to downgrade 1.16.0 --> 1.15.3?

Supervisor, ESPHome, Configuration
[image]

glmnet, 2021-02-10:
Try decreasing the wait time from 50ms to 3ms or other values

mihing, 2021-02-10:
> 
> 
> Try decreasing the wait time from 50ms to 3ms or other values

3ms, 1ms, 10ms -- no effect.
Previously, devices worked the first time, I added "repeat ..., time" for reliability.

40pints, 2021-02-10:
Having the same issue. I have noticed the log no longer states sending remote command when it should be transmitting

glmnet, 2021-02-10:
> Having the same issue. I have noticed the log no longer states sending remote command when it should be transmitting

can you post the log of both 1.15 and 1.16 for comparison? that will be very helpful if they are different

DieKatzchen, 2021-02-11:
Seems to be the same issue I was having with #1673 except that my ESP8266 works fine and my ESP32 exhibits this behavior (the log not showing that a signal was ever sent)

glmnet, 2021-02-11:
I just ordered DT-BL10 devkits.. let's see

DieKatzchen, 2021-02-11:
As far as logs go, here's the relevant code:

`remote_transmitter:
  pin:
    number: RX
    inverted: false
  # RF uses a 100% carrier signal
  carrier_duty_percent: 100%
  id: rf
  
switch:
  - platform: template
    name: Projector UP
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '100010110010010110000010'
          protocol: 2
          repeat: 3
          transmitter_id: rf`

Which when running on a generic esp8266 produces:

`[20:15:16][D][switch:021]: 'Projector UP' Turning ON.
[20:15:16][D][remote_transmitter:066]: Sending remote code...`

and on a generic ESP32 instead produces

`[20:18:55][D][switch:021]: 'Projector UP' Turning ON.`

TRon993, 2021-02-15:
Same for me. I`m using D1 Minis that used to work amazingly great until the upgrade today to 1.16.1. Also, reverting (downgrade) to 1.15.3 didnt work (no settings in Supervisor, changes in the config file are ignored as well). Couldnt find any hints regarding this component in the changelog. May this be related to the internal upgrades like plattform io or something like that?

Thanks for any hints on how to get it back working!

franz82, 2021-02-16:
Same for me too. Shelly 1 v3 here.

- Home Assistant OS 5.11 on RPI3b+
- Core ver: 2021.2.3
- Supervisor ver: supervisor-2021.02.9
- ESPHome: 1.16.1

During source compiling I got this warning about SPIFFS deprecated:

```
    /data/shelly1_template/.piolibdeps/shelly1_template/ESPAsyncWebServer-esphome/src/SPIFFSEditor.h:16:101: warning: ‘SPIFFS’ is deprecated (declared at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:269): SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
    SPIFFSEditor(const String& username=String(), const String& password=String(), const fs::FS& fs=SPIFFS);
    ^
```

it could be the culprit?

TRon993, 2021-02-16:
Ok so, I've tried to downgrade to 1.15.3 (docker via supervisor installation) but couldn't get it working. I thought I've seen the possibility to chose a certain version but this seems to have changed since Nov/Dec I think. Even manually setting this value on options.json doesn't do anything (besides failing)... 

Anyone any idea on how to downgrade to a working version? Can't believe that this seems to be Impossible.

Thanks in advance

glmnet, 2021-02-16:
I believe the issue should be either https://github.com/esphome/esphome/pull/1465 or most likely https://github.com/esphome/esphome/pull/1426

So you can test undoing those changes and see what happens.

I have some covers actuated by rc-switch and I don't see a problem here, all still works.

To proper fix this issue we have to line up with another implementation, if somebody would capture esphome outputs and compare with a known transmitter output that will be helpful to decide which is correct.

TRon993, 2021-02-16:
Thanks for the references. The upgrade by itself seems to be uncritical. What I did is to update all of my devices via "update all" function on the dashboard. I could see that a newer PlattformIO version was added as well as some of the components I'm using. I'm trying to clean up and recompile the firmware of my rf enabled device. All other devices are running as usual so far... Weird

OkhammahkO, 2021-02-17:
I think I'm having the same issue. I can't downgrade either.

TRon993, 2021-02-17:
Crazy... it seems to work.

Changed
`- remote_transmitter.transmit_rc_switch_raw:
          code: '10100101011100001000100100000000'
          protocol: 2 
          repeat: 
            times: 3
            wait_time: 650us`
to
`- remote_transmitter.transmit_rc_switch_raw:
          code: '10100101011100001000100100000000'
          protocol: 2 
          repeat: 
            times: 10
            wait_time: 0s`

Thanks a bunch for this solution. Nevertheless, i will be more cautious regarding updates in the future.

mihing, 2021-02-22:
> Crazy... it seems to work.

Yes!

Changed
wait_time: 50ms
to
wait_time: 0ms.
Its works!

(ESPHome 1.16.2)

erkme73, 2021-05-14:
Confirming that my RF Bridge also stopped working after 1.15.3, but setting wait_time to zero ms let me transmit again.  Thanks for the workaround!

closed 2021-09-25 by stale[bot]