esphome/issues #1829: Output Switch Always Restores

alexyao2015, 2021-02-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

https://esphome.io/components/switch/output.html

**Description of problem:**
https://github.com/esphome/esphome/blob/5c86f332b269fd3e4bffcbdf3359a021419effdd/esphome/components/output/switch/output_switch.cpp#L11

There is no way to disable restoring of the state. The code currently always creates a preference object which can be problematic with an esp8266 if `esp8266_restore_from_flash` is enabled. This would cause a preference object to be created using restore from flash as true. Should the switch change rapidly, it could degrade the flash. 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2021-02-10:
This more than an issue is a lack of a feature, it's documented that https://esphome.io/components/esphome.html#esp8266-restore-from-flash what you're reporting as an issue.

glmnet, 2021-02-10:
This https://github.com/esphome/esphome/pull/1063 and this https://github.com/esphome/esphome/pull/1469 are attempts to fix the issue in a broader aspect.

alexyao2015, 2021-02-10:
Thanks for pointing those out! I wanted to bring that specific component to attention because it lacks the restore mode feature that other components have. This could lead an unknowing user to including that and thinking it would be safe for rapid changes since restore mode isn't an option. Personally I don't use this component but found it while digging through the code.

alexyao2015, 2021-02-10:
For most other components, setting restore state to no is enough to prevent it from being stored to the flash and thus remove the caveats from enabling that option as it can be selectively enabled per component.

closed 2021-06-25 by stale[bot]