esphome/issues #1833: Start container addon_a0d7b954_esphome failed

Martinnygaard, 2021-02-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant install via Supervisor
Installed on Synology DS920+ with Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

No ESP yet. Issues with HA AddOn

**ESPHome version (latest production, beta, dev branch)**

Latest Release in HA Add-on store: ESP Home 1.16.0

**Affected component:**

HA Add-on (Docker container)

**Description of problem:**
I recently tried to install ESPHome for the first time. I can install without problems, but when I try to start the AddOn, i get the
following message. (See LOG section).
Any ideas what may happen?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

**The message from HA Addon: (And screenshot)**

> **Failed to start addon**
> 400 Client Error for http+docker://localhost/v1.39/containers/b48796972681369e9dcfe5a2b5da1c15a06786532b3149948bba6b2e62562355/start: Bad Request ("OCI runtime create failed: container_linux.go:345: starting container process caused "exec: \"/dev/init\": stat /dev/init: no such file or directory": unknown"

> [image]

**The Log from Docker:**

> **Error**
2021/02/11   11:29:32
admin
Start container addon_a0d7b954_esphome failed:   {"message":"OCI runtime create failed: container_linux.go:345:   starting container process caused \"exec: \\\"/dev/init\\\":   stat /dev/init: no such file or directory\": unknown"}.

**Additional information and things you've tried:**

DEADSEC-SECURITY, 2021-02-11:
Having the same problem with esphome. If anyone could help would be awesome.

I tried reinstalling HA, Esphome
Removing files that could stay after uninstall. 
Nothings seems to fix it.

bredberger, 2021-02-12:
Same issue too. 
Tried restoring multiple partial and full images with no success.
DS216+II

Martinnygaard, 2021-02-12:
I am in fact having the same problem with the Sambe Backup Addon.
Ind if I am not mistaken, the same issue with Samba, happened after upgrading HA Core from 2020.2.1 to 2020.2.2..
Iwil try to upgrade to 2020.2.3, if still not working, I will try to revert to Home Assistant Core 2020.2.1.

Martinnygaard, 2021-02-12:
@bredberger , you are also on Synology.
What about you, @DEADSEC-SECURITY ? Synology?

Martinnygaard, 2021-02-12:
It seams it is a general issue on multiple Addons. 

https://community.home-assistant.io/t/hass-io-on-synology-dsm-native-package/125559/1413

https://github.com/home-assistant/supervisor/issues/2514

andilge, 2021-02-13:
+1 with exactly the same problem on a Synology DS-218+. jobs.yaml with ignore health is active. As @DEADSEC-SECURITY I uninstalled everything, esphome and HA, cleaned up manually after this and did a fresh install from scratch => no joy

DEADSEC-SECURITY, 2021-02-13:
> @bredberger , you are also on Synology.
> What about you, @DEADSEC-SECURITY ? Synology?

Yes I am

andilge, 2021-02-13:
CPU architekcure x86_64 on your Syno @DEADSEC-SECURITY?

andilge, 2021-02-13:
got this answer from another channel:

it's Synology and Synology is officially unsupported in this type of installation.

This error happens with all addon's - I got it with MQTT few days ago, same error. **Fix is to downgrade Supervisor to 2021.01.x. And then it's catch 22 since some new addons require 2021.02.x**

From what I've found, error can be related to API version that is too low on Synology.

hass.io community installation package definitely got deprecated with this => new installation with docker is the way to go.

DEADSEC-SECURITY, 2021-02-14:
Try creating the file /dev/init. Could work. Still dint teste it but could work.

DEADSEC-SECURITY, 2021-02-15:
It didn't work cuz docker doesn't have permission, and Synology works strangely for permissions. I tried other things, but none work including creating a docker group. It seems like it's a problem with version 1.16.0. Maybe there is a way to choose the version of esphome in HA to test to see if it a bug from this new version?

papipascal, 2021-04-03:
I've the same issue with ESP Home and several other addons...
Snology DS415+, DSM 6.2.3, Docker 18.09.8, HA Core 21.3.4, ESP Home 1.16.2
addon_a0d7b954_esphome : I can't start it
even when i try to start it with portainer...
same with mosquitto, grafana, ESP Home DEV 
Others are OK 
message is: 
21-04-03 18:06:20 ERROR (SyncWorker_4) [supervisor.docker] Can't start addon_a0d7b954_esphome: 400 Client Error for http+docker://localhost/v1.39/containers/e1fc4eb93c45d5ff53b74c96e7127c3c97bd538abfcc2e8a9501bc2f3a44a66c/start: Bad Request ("OCI runtime create failed: container_linux.go:345: starting container process caused "exec: \"/dev/init\": stat /dev/init: no such file or directory": unknown")

OttoWinter, 2021-04-10:
This sounds like an issue in https://github.com/home-assistant/supervisor, not esphome (as this error appears before the esphome container is even started). Please open an issue there (or reopen if it really is just esphome)

closed 2021-04-10 by OttoWinter

ihipop, 2021-06-13:
I figure out  out the problem and make a Temporary solution
`cp -av `which docker-init` /dev/init`
if you want to know why :

newer docker version create docker-init as /dev/init in host too ,but docker 17/18 doesn’t .newer hassio supervisor mount /dev/ into container which make the the /dev/init in container created by docker --init disappear.

see detail  https://community.home-assistant.io/t/failed-to-start-addon-mosquitto-broker/276551/49?u=ihipop