esphome/issues #1835: BME280 relative pressure is wrong

Protoncek, 2021-02-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io, running in Docker on Synology NAS

**ESP (ESP32/ESP8266, Board/Sonoff):**
 Sonoff RE5V1 board with BME280

  
**ESPHome version (latest production, beta, dev branch)**
 ESPHome latest production version

  
**Affected component:**
BME280, environment calculations
[https://esphome.io/cookbook/bme280_environment.html](url)

**Description of problem:**
I think that formula for "equivalent sea-level pressure" may be wrong here. I have two sensors, one is outside and one inside and they show different relative pressure, depending on temperature, while they shouldn't. 
Example:
OUTSIDE:
temp: -4,7 degrees Celsius
abs. pressure: 996hPa
rel.pressure: 1037hPa

INSIDE:temp: 17,4 deg. C
abs.pressure: 996 hPa
rel.pressure: 1034 hPa

So, the difference is 3 hPa, which is wrong, because it's clearly that absolute pressures are equal. I think that problem is in formula itself, because it has temperature in calculation, while it really shouldn't have. Absolute pressure is already calculated inside BME with temperature taken into account, so relative pressure should be calculated WITHOUT temperature in formula. 
I found a formula in BME180 datasheet (page 17). Interesting, though, that relative pressure is not mentioned at all in BME/P 280 datasheets... 
 
[image]

output from this formula in above conditions show 1034hPa, so it seems that ESP formula is pretty accurate in room temperatures, but way off when temps are low (freezing...). I didn't test it in hot conditions (say, 30 and more deg.C), though but i  guess that rel.pressure would be lower than correct one in that case.

Any thoughts, suggestions?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
  - platform: template
    name: "Relative pressure 1"
    icon: "mdi:gauge"

//WRONG formula
    lambda: |-
      const float STANDARD_ALTITUDE = 315; // in meters, see note
      return id(abs_pressure_out).state / powf(1 - ((0.0065 * STANDARD_ALTITUDE) /
        (id(temperature_out).state + (0.0065 * STANDARD_ALTITUDE) + 273.15)), 5.257); // in hPa
    update_interval: 60s
    unit_of_measurement: 'hPa'
    accuracy_decimals: 0  
    
  - platform: template
    name: "Relative pressure 2"
    icon: "mdi:gauge"

//CORRECT formula
    lambda: |-
      const float STANDARD_ALTITUDE = 315; // in meters, see note
      return (id(abs_pressure_out).state)/powf((1-((float)(STANDARD_ALTITUDE))/44330), 5.255);
    update_interval: 60s
    unit_of_measurement: 'hPa'
    accuracy_decimals: 0   

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
Before ESPHome i used Tasmota firmware in my sonoff's and relative pressure was ok there. I often compare my readings with official weather station ones and with Tasmota they were always ok, while with ESPHome are getting more and more off with lower temperatures, that's why i suspected wrong formula.

I found ESPHome FW way better and more flexible than Tasmota, that's why i switched...

MichielioZ, 2021-02-20:
Let me start by saying: I am not an expert on this and I have no BME280 running outside to compare pressure with.
What I can say however is that it seems to me you did your homework.
The [code for the BME280](https://github.com/esphome/esphome/tree/dev/esphome/components/bmp280) seems pretty straightforward, so perhaps you could try to make pull request ?
Just a suggestion of course, don't know your situation.

Protoncek, 2021-02-20:
Well, sadly i'm not so experienced in programming so i don't really know how or what to do...
I'm still learning how to program Home Assistant, but i guess all that is needed is change web page ESPHome Environment, since relative pressure calculation is not part of basic BME280 code, but you must enter formula later in configuration. Although it would be nice to have relative pressure already in basic code, since relative pressure is the one that average user needs, not absolute.

closed 2021-07-04 by stale[bot]