esphome/issues #1837: Not using ESP and getting Warning on supervisorHASSOs

hellresistor, 2021-02-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant OS .vmdk

**ESP (ESP32/ESP8266, Board/Sonoff):**
SonOff 

**ESPHome version (latest production, beta, dev branch)**
NOT USING!

**Affected component:**
no..

**Description of problem:**
just dont have this installed and keep showing the error..

**Problem-relevant YAML-configuration entries:**
```yaml

```

**Logs (if applicable):**

```
21-02-14 13:08:33 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-observer with version 2020.10.1
21-02-14 13:08:33 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-multicast with version 3
21-02-14 13:08:33 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
21-02-14 13:08:33 INFO (SyncWorker_0) [supervisor.docker.multicast] Starting Multicast homeassistant/amd64-hassio-multicast with version 3 - Host
21-02-14 13:08:33 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-02-14 13:08:33 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/qemux86-64-homeassistant with version 2021.2.3
21-02-14 13:08:33 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 1 Home Assistant secrets
21-02-14 13:08:33 INFO (MainThread) [supervisor.hassos] Detect HassOS 5.11 / BootSlot B
21-02-14 13:08:33 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome
21-02-14 13:08:33 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
21-02-14 13:08:33 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
21-02-14 13:08:33 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/f13797af repository
21-02-14 13:08:33 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome
21-02-14 13:08:33 INFO (MainThread) [supervisor.store] Loading add-ons from store: 66 all - 66 new - 0 remove
21-02-14 13:08:33 INFO (MainThread) [supervisor.addons] Found 8 installed add-ons
```

**Additional information and things you've tried:**
Nothing.. Never installed this addon. never used before. cant understand why keep warning.. A single reboot to reproduce.

samnewman86, 2021-02-14:
Also having this issue. I do use esphome though. I've updated to latest version

Mikefila, 2021-02-15:
I am using esphome and receiving this message in the supervisor log

```
 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome
```

Using version 1.16.1, working nodes updated.
Log from esphome

```
INFO 200 GET / (0.0.0.0) 7.07ms
INFO 200 GET /static/fonts/material-icons/material-icons.css?hash=ead4c276 (0.0.0.0) 3.77ms
INFO 304 GET /static/js/vendor/materialize/materialize.min.js?hash=4be20daf (0.0.0.0) 3.11ms
INFO 304 GET /static/js/vendor/jquery-ui/jquery-ui.min.js?hash=cab3392d (0.0.0.0) 3.74ms
INFO 200 GET /static/css/vendor/materialize/materialize.min.css?hash=ec1df3ba (0.0.0.0) 6.28ms
INFO 304 GET /static/css/vendor/materialize-stepper/materialize-stepper.min.css?hash=f4d254e9 (0.0.0.0) 3.41ms
INFO 304 GET /static/css/esphome.css?hash=50406740 (0.0.0.0) 10.99ms
INFO 304 GET /static/js/vendor/jquery/jquery.min.js?hash=12108007 (0.0.0.0) 10.08ms
INFO 304 GET /static/js/vendor/jquery-validate/jquery.validate.min.js?hash=aad785d4 (0.0.0.0) 8.23ms
INFO 304 GET /static/js/vendor/materialize-stepper/materialize-stepper.min.js?hash=ef217c59 (0.0.0.0) 3.77ms
INFO 304 GET /static/js/vendor/ace/ace.js?hash=65322f49 (0.0.0.0) 3.33ms
INFO 304 GET /static/js/esphome.js?hash=b418707c (0.0.0.0) 3.75ms
INFO 200 GET /static/fonts/material-icons/MaterialIcons-Regular.woff2 (0.0.0.0) 2.66ms
INFO 304 GET /static/js/vendor/ace/theme-dreamweaver.js (0.0.0.0) 5.61ms
INFO 101 GET /ace (0.0.0.0) 3.05ms
INFO Running command 'esphome --dashboard -q /config/esphome vscode --ace'
INFO 304 GET /static/js/vendor/ace/mode-yaml.js (0.0.0.0) 3.69ms
```

adnn-hassio, 2021-02-18:
+1 same here:
21-02-18 09:11:07 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome

And i cant see hassio graph in lovelace card. My room temperature always  27.5 cantigrad. But when i check node from its ipi it is working good.

hellresistor, 2021-03-04:
Well I can Say. Not showing anymore!!!

💯

closed 2021-03-04 by hellresistor