esphome/issues #1839: Failed to start at setting up wifi

maxperron, 2021-02-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

[ESP32](https://www.banggood.com/ESP32-Development-Board-WiFi+bluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP-32-ESP-32S-Board-p-1109512.html?rmmds=myorder&cur_warehouse=CN)

**ESPHome version (latest production, beta, dev branch)**

latest production
**Affected component:**

wifi I believe

**Description of problem:**
After compiling from esphome in HA and downloading the bin to Windows 10 and flashing with ESPHome-Flasher-1.3.0-Windows-x64, the board show the log below and doesn't boot. Always ending by Serial port closed!

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: flora_1
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "mywifissid"
  password: "mypassword"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Flora 1 Fallback Hotspot"
    password: "qR0RAW61v0SH"

captive_portal:

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:

ota:

```

**Logs (if applicable):**

```
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[15:01:14][I][logger:166]: Log initialized
[15:01:14][V][preferences:260]: nvs_get_blob('0'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[15:01:14][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[15:01:14][VV][preferences:051]: SAVE 0: 0=0x00000001 1=0x42F4583F (Type=233825507, CRC=0x42F4583F)
[15:01:14][I][app:029]: Running through setup()...
[15:01:14][V][app:030]: Sorting components by setup priority...
[15:01:14][V][preferences:260]: nvs_get_blob('1'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[15:01:14][C][wifi:033]: Setting up WiFi...
[15:01:14][V][wifi_esp32:036]: Enabling STA.
[15:01:14][VV][esp-idf:000]: I (109) wifi: wifi driver task: 3ffb6030, prio:23, stack:3584, core=0
[15:01:14]
[15:01:14][VV][esp-idf:000]: I (423) wifi: wifi firmware version: 7997e4b
[15:01:14]
[15:01:14][VV][esp-idf:000]: I (423) wifi: config NVS flash: enabled
[15:01:14]
[15:01:14][VV][esp-idf:000]: I (425) wifi: config nano formating: disabled
[15:01:14]
[15:01:14][VV][esp-idf:000]: I (432) wifi: Init dynamic tx buffer num: 32
[15:01:14]
[15:01:14][VV][esp-idf:000]: I (439) wifi: Init data frame dynamic rx buffer num: 32
[15:01:14]
[15:01:14][VV][esp-idf:000]: I (446) wifi: Init management frame dynamic rx buffer num: 32
[15:01:14]
[15:01:15][VV][esp-idf:000]: I (454) wifi: Init management short buffer num: 32
[15:01:15]
[15:01:15][VV][esp-idf:000]: I (462) wifi: Init static rx buffer size: 1600
[15:01:15]
[15:01:15][VV][esp-idf:000]: I (469) wifi: Init static rx buffer num: 16
[15:01:15]
[15:01:15][VV][esp-idf:000]: I (475) wifi: Init dynamic rx buffer num: 32
[15:01:15]
[15:01:15]
Serial port closed!

```

**Additional information and things you've tried:**
I have tried multiple device type and it still stop at the same place.
I have flashed multiple ESP8266 using this solution and it worked flawlessly. It my first time with a ESP32, maybe I'm not doing something right.

Patrickmaartense, 2021-02-16:
can confirm the issue
when compiling 1.16.1 version wifi is broken
using OTA from a 1.14 version will render the device inoperable and needs to revert via serial 
flashing the old version (same yaml file) will bring the device back to life

maxperron, 2021-02-19:
@Patrickmaartense , were you able to flash again? I bought new ESP32 and worry about bricking them.

Patrickmaartense, 2021-02-20:
yes, using serial, no issues reflashing, just used an older version (1.14.5)

maxperron, 2021-02-28:
I bought 2 brand new ESP32 from a different site and a different vendor and I get the same error. There is definitely an issue with esphome and ESP32...

I was not able to find a way to downgrade esphome to 1.14.5. Hopefully, this issue will be resolve one day.

Patrickmaartense, 2021-03-01:
You can uninstall esphome and then use pip to install a certain version

closed 2021-06-29 by maxperron