esphome/issues #1841: Cannot Set Configuration - Hass.io

alex-shaw, 2021-02-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

Not applicable to the problem (though I have used both ESP8266 and ESP32 with this instance)
**ESPHome version (latest production, beta, dev branch)**

1.16.1
**Affected component:**

Not applicable to the problem

**Description of problem:**
Accessing the configuration page of the addon in Hass.io supervisor (/hassio/addon/DOCKERCONTAINERID_esphome/config) shows only the generic network configuration - no access to the actual configuration options. Vanilla Hass.io install with ESPHome installed as per readme instructions

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
N/A
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-banner.sh: executing... 
-----------------------------------------------------------
 Hass.io Add-on: ESPHome
 ESPHome Hass.io add-on for intelligently managing all your ESP8266/ESP32 devices.
-----------------------------------------------------------
 Add-on version: 1.16.1
 You are running the latest version of this add-on.
 System: Debian GNU/Linux 10 (buster)  (amd64 / qemux86-64)
 Home Assistant Core: 2021.2.3
 Home Assistant Supervisor: 2021.02.9
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] 10-requirements.sh: executing... 
[cont-init.d] 10-requirements.sh: exited 0.
[cont-init.d] 20-nginx.sh: executing... 
[cont-init.d] 20-nginx.sh: exited 0.
[cont-init.d] 30-esphome.sh: executing... 
[cont-init.d] 30-esphome.sh: exited 0.
[cont-init.d] 40-migrate.sh: executing... 
[cont-init.d] 40-migrate.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[22:36:10] INFO: Waiting for dashboard to come up...
[22:36:10] INFO: Starting ESPHome dashboard...
INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[22:36:12] INFO: Starting NGINX...
```

No additional errors in Homeassistant

**Additional information and things you've tried:**
I've connected to the Hass.io box and can see a configuration file exists in the addon director and consists of "{}" - if i alter this file to add custom configuration, it reverts to the empty braces when restarting the addon.

alex-shaw, 2021-02-15:
[image]

alex-shaw, 2021-02-15:
[image]
This is what i had expected from the config screen (from the grafana add on)

Adrian-at-CrimsonAuzre, 2021-02-17:
I am also having this issue with a clean install of ESPHome. I think it has something to do with the config not being set correctly?

config.json for ESPHome:
```json
{
  "arch": [
    "amd64",
    "armv7",
    "aarch64"
  ],
  "auth_api": true,
  "auto_uart": true,
  "boot": "auto",
  "description": "ESPHome Hass.io add-on for intelligently managing all your ESP8266/ESP32 devices.",
  "hassio_api": true,
  "hassio_role": "default",
  "homeassistant_api": false,
  "host_network": true,
  "image": "esphome/esphome-hassio-{arch}",
  "ingress": true,
  "ingress_port": 0,
  "map": [
    "ssl:ro",
    "config:rw"
  ],
  "name": "ESPHome",
  "options": {},
  "panel_icon": "mdi:chip",
  "ports": {
    "6052/tcp": null
  },
  "ports_description": {
    "6052/tcp": "Web interface (Not required for Home Assistant Ingress)"
  },
  "schema": {
    "certfile": "str?",
    "esphome_version": "str?",
    "keyfile": "str?",
    "leave_front_door_open": "bool?",
    "relative_url": "str?",
    "ssl": "bool?",
    "status_use_ping": "bool?",
    "streamer_mode": "bool?"
  },
  "slug": "esphome",
  "stage": "stable",
  "startup": "application",
  "url": "https://esphome.io/",
  "version": "1.16.1",
  "webui": "http://[HOST]:[PORT:6052]"
}
```
Node-Red config.json:
```json
{
  "name": "Node-RED",
  "version": "8.1.1",
  "slug": "nodered",
  "description": "Flow-based programming for the Internet of Things",
  "url": "https://github.com/hassio-addons/addon-node-red",
  "ingress": true,
  "ingress_port": 0,
  "panel_icon": "mdi:sitemap",
  "init": false,
  "homeassistant": "0.92.0b2",
  "arch": [
    "aarch64",
    "amd64",
    "armhf",
    "armv7",
    "i386"
  ],
  "ports": {
    "80/tcp": 1880
  },
  "ports_description": {
    "80/tcp": "Web interface"
  },
  "hassio_api": true,
  "hassio_role": "manager",
  "homeassistant_api": true,
  "host_network": true,
  "auth_api": true,
  "uart": true,
  "gpio": true,
  "privileged": [
    "SYS_RAWIO"
  ],
  "devices": [
    "/dev/mem"
  ],
  "apparmor": false,
  "map": [
    "config:rw",
    "media:rw",
    "share:rw",
    "ssl"
  ],
  "options": {
    "credential_secret": "",
    "dark_mode": false,
    "http_node": {
      "username": "",
      "password": ""
    },
    "http_static": {
      "username": "",
      "password": ""
    },
    "ssl": true,
    "certfile": "fullchain.pem",
    "keyfile": "privkey.pem",
    "require_ssl": true,
    "system_packages": [],
    "npm_packages": [],
    "init_commands": []
  },
  "schema": {
    "log_level": "list(trace|debug|info|notice|warning|error|fatal)?",
    "credential_secret": "password",
    "dark_mode": "bool",
    "http_node": {
      "username": "str",
      "password": "password"
    },
    "http_static": {
      "username": "str",
      "password": "password"
    },
    "ssl": "bool",
    "certfile": "str",
    "keyfile": "str",
    "require_ssl": "bool",
    "system_packages": [
      "str"
    ],
    "npm_packages": [
      "str"
    ],
    "init_commands": [
      "str"
    ],
    "i_like_to_be_pwned": "bool?",
    "leave_front_door_open": "bool?"
  },
  "image": "ghcr.io/hassio-addons/node-red/{arch}"
}
```

Notice how, while schema is filled out for both, `options` is not filled out for ESPHome. I believe this is supposed to be the default options for the package and if it is blank it just doesn't generate a config.

mpepping, 2021-02-17:
Same, also after reinstalling the add-on.  Running core-2021.2.3 with supervisor-2021.02.9

ogiewon, 2021-02-19:
Same issue here as well.

jaddel, 2021-02-23:
Same here, is there a manual fix we can apply until it has been updated? 

ESPHome Current version: 1.16.2
Supervisor Version supervisor-2021.02.11

Adrian-at-CrimsonAuzre, 2021-02-23:
I _need_ to use `status_use_ping` on my network because the IOT VLAN is locked down tight and I can't easily do mDNS forwarding. No way of setting it because the HASS config doesn't work.

rkhanso, 2021-03-01:
I can't set ESPHome to use the secrets.yaml because of this bug - 
I'm getting the same problem as those previously reported.
 Add-on version: 1.16.2
 You are running the latest version of this add-on.
 System: Home Assistant OS 5.12  (amd64 / qemux86-64)
 Home Assistant Core: 2021.2.3
 Home Assistant Supervisor: 2021.02.11

ced1442, 2021-03-02:
Same problem here, all up to date and no access to the configuration file, only the network.

JanVanr, 2021-03-02:
Same problem, all up to date. Only network section in config and that cannot be “enabled” either. (Doesn’t save). I’m not sure if it is related, but if I try to save the config of an esp module it sometimes return: an error occurred saving esp_xxxx.yaml. Running on a pi4 with external hdd (no sd)

glmnet, 2021-03-03:
The fix to this issue is coming in tomorrow release of home assistant is an UI issue. 

Until then 

Dev version works
Beta home assistant works

Also other solution will be adding the options to the file directly but that's hard to explain and the fix is right there. 

Thanks for reporting

closed 2021-03-03 by glmnet