esphome/issues #1842: Cannot get Thermostat Yaml to validate

jaysalazar, 2021-02-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**ESPHome version (latest production, beta, dev branch)**

1.16.1
**Affected component:**

https://esphome.io/components/climate/thermostat.html

**Description of problem:**
Will not validate, getting the following message ..... Platform not found: 'climate.thermostat'.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
climate:
  - platform: thermostat
    name: "Thermostat Climate Controller"
    sensor: BME280_Temp
    default_target_temperature_low: 20 °C
    default_target_temperature_high: 22 °C
    cool_action:
      - switch.turn_on: air_cond
    heat_action:
      - switch.turn_on: heater
    idle_action:
      - switch.turn_off: air_cond
      - switch.turn_off: heater
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

SenexCrenshaw, 2021-02-20:
this worked for me but I just used  this section. Can you post your full yaml (remove wifi information)?

jaysalazar, 2021-02-20:
esphome:
  name: hydro3
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: "XXXXXX"
  password: "XXXXXXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Hydro3 Fallback Hotspot"
    password: "v8R11voGhj2g"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

sensor:
  - platform: ultrasonic
    trigger_pin: GPIO25
    echo_pin: GPIO26
    unit_of_measurement: "Inches from Top"
    icon: "mdi:water-percent"
    name: "Ultrasonic Sensor"
    update_interval: 10s
    filters:
    - lambda: return x * 39.37;
  - platform: wifi_signal
    name: "External Signal"
    update_interval: 10s

climate:
  - platform: thermostat
    name: "Thermostat Climate Controller"
    sensor: my_temperature_sensor
    default_target_temperature_low: 20 °C
    default_target_temperature_high: 22 °C
    cool_action:
      - switch.turn_on: air_cond
    heat_action:
      - switch.turn_on: heater
    idle_action:
      - switch.turn_off: air_cond
      - switch.turn_off: heater

kbx81, 2021-03-23:
```yaml
sensor: my_temperature_sensor
```

...does not exist in the config you pasted. Add `id: my_temperature_sensor` to the (temperature) sensor you want the `thermostat` component to use for control. You also have:

```yaml
sensor: BME280_Temp
```

...in your original snippet but I don't see that sensor in your config, either. Did you omit part of it?

jaysalazar, 2021-03-25:
Good catch @kbx81 I updated my code however I still get the "Platform not found" error message while trying to validate. Here i my code.

`esphome:`
  name: hydro3
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: "XXXXXXXXXXXXX"
  password: "XXXXXXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Hydro3 Fallback Hotspot"
    password: "v8R11voGhj2g"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
i2c:
  sda: 21
  scl: 22
  scan: True
  id: bus_a

sensor:
  - platform: ultrasonic
    trigger_pin: GPIO25
    echo_pin: GPIO26
    unit_of_measurement: "Inches from Top"
    icon: "mdi:water-percent"
    name: "Ultrasonic Sensor"
    update_interval: 10s
    filters:
    - lambda: return x * 39.37;
  - platform: wifi_signal
    name: "External Signal"
    update_interval: 10s
  - platform: bme280
    address: 0x76
    temperature:
     name: BME280_Temp
     id: BME280_Temp
     oversampling: 16x
    pressure:
     name: BME280_Pressure
     id: BME280_Pressure
    humidity:
     name: BME280_Humidity
     id: BME280_Humidity
    update_interval: 10s

climate:
  - platform: thermostat
    name: "Thermostat Climate Controller"
    sensor: BME280_Temp 
    default_target_temperature_low: 20 °C
    default_target_temperature_high: 22 °C
    cool_action:
      - switch.turn_on: air_cond
    heat_action:
      - switch.turn_on: heater
    idle_action:
      - switch.turn_off: air_cond
      - switch.turn_off: heater
     ``

kbx81, 2021-03-25:
> Platform not found

What version of ESPHome are you attempting to use?

jaysalazar, 2021-03-25:
I am using version: 1.16.1

kbx81, 2021-03-25:
Could you paste the complete error message when you try to compile/run this?

Pro tip: wrap blocks of text in triple ticks to format it correctly. See https://guides.github.com/features/mastering-markdown/

jaysalazar, 2021-03-25:
Thanks for the tip!!

```
INFO Reading configuration /config/esphome/hydro3.yaml...
Failed config

climate.thermostat: [source /config/esphome/hydro3.yaml:59]
  
  Platform not found: 'climate.thermostat'.
  platform: thermostat
  name: Thermostat Climate Controller
  sensor: BME280_Temp
  default_target_temperature_low: 20 °C
  default_target_temperature_high: 22 °C
  cool_action: 
    - switch.turn_on: air_cond
  heat_action: 
    - switch.turn_on: heater
  idle_action: 
    - switch.turn_off: air_cond
    - switch.turn_off: heater
  ```

jaysalazar, 2021-03-25:
I think this may have something to do with it, I don't see the climate.thermostat .h or .cpp

[image]

kbx81, 2021-03-25:
The `thermostat` component isn't in the `climate` directory; it has its own directory.

I could be overlooking something, but from what I can see it looks fine. Are you building this config directly in the ESPHome UI/Dashboard within Home Assistant or are you building it in a terminal on your local machine? The "Platform not found" error is very odd and makes me wonder if your source is somehow not up-to-date. If you're running it in HA, I wonder if reinstalling the ESPHome add-on would be a worthwhile move.

jaysalazar, 2021-03-25:
I am building it directly in the ESPHome Dashboard within Home Assistant. I can try to reinstall the add-on,

kbx81, 2021-03-25:
Yeah, aside from reinstalling the add-on, I'm kind of out of ideas. From what I can see, your yaml looks fine -- although it never hurts to double-check the spacing and alignment of things. The dashboard editor is pretty good at highlighting errors but it may not catch _everything_ -- and yaml is very sensitive to spaces/spacing. The only other thing I can think of is if something didn't paste correctly (maybe you got some special character and aren't realizing it). If you were copy/pasting, maybe try re-typing what you have in the editor. This _should_ be working for you -- lots of others using this component now so I don't think it's a bug.

jaysalazar, 2021-05-07:
It's working now. I had to reinstall the add on like you suggested. Thank you for the help!

closed 2021-05-22 by ssieb