esphome/issues #1843: RF Sync header being sent at the end of the code payload instead of at the beggining

Rodrigoah, 2021-02-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMcu

**ESPHome version (latest production, beta, dev branch)**

1.15.3

**Affected component:**

remote_transmitter

**Description of problem:**

Header of custom protocol is sent at the end of the code instead of at the beggining

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

      remote_transmitter.transmit_rc_switch_raw:
        code: '1010001100110000000101000010010000000001000000000000101101110100'
        protocol:
          pulse_length: 510
          sync: [25, 3]
          zero: [1,3]
          one: [3,1]

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Rodrigoah, 2021-02-15:
[image]

As you can see, the sync is being sent at the end, while it is starting with the code

Rodrigoah, 2021-02-18:
more info, 

it does not happen if "times" is set to a value greater than one

glmnet, 2021-02-21:
Can you update with 1.16.2?

Rodrigoah, 2021-03-03:
> Can you update with 1.16.2?

This issue is solved in 1.16.2

The issue can be closed

closed 2021-03-03 by Rodrigoah