esphome/issues #1847: DHT: add support for setting the 'device_class'

StijnVdd, 2021-02-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**ESPHome version (latest production, beta, dev branch)**

1.16.1

**Affected component:**

DHT and other sensor components

**Description of problem:**
The sensor components like the DHT, do not report the device_class to HA. It would be great to set the device class correctly for temperature/humidity...

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

SenexCrenshaw, 2021-02-20:
https://github.com/esphome/esphome/pull/1525 has been merged into dev

glmnet, 2021-02-21:
This will be supported in 1.17

closed 2021-02-21 by glmnet