esphome/issues #1850: Build fails at installing ESPAsyncTCP-esphome @ 1.2.3 due to timeouts

njfberg, 2021-02-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home assistant supervisor-2021.02.9, core-2021.2.3
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 d1_mini
**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.16.1 
**Affected component:**

https://esphome.io/guides/getting_started_hassio.html

**Description of problem:**
When uploading code, the build fails at this step:

Library Manager: Installing ESPAsyncTCP-esphome @ 1.2.3

It's a bit random as it does not occur during all builds. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: deurbel
  platform: ESP8266
  board: d1_mini

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/deurbel.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting /data/deurbel/.pioenvs
INFO Deleting /data/deurbel/.piolibdeps
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/deurbel
Processing deurbel (board: d1_mini; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Library Manager: Installing ESPAsyncTCP-esphome @ 1.2.3
HTTPClientError: HTTPSConnectionPool(host='dl.registry.platformio.org', port=443): Max retries exceeded with url: /download/ottowinter/library/ESPAsyncTCP-esphome/1.2.3/ESPAsyncTCP-esphome-1.2.3.tar.gz (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xb5b50850>: Failed to establish a new connection: [Errno -2] Name or service not known',)):
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "/root/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/root/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/root/.platformio/platforms/espressif8266/builder/main.py", line 238:
    target_elf = env.BuildProgram()
  File "/root/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/platformio.py", line 62:
    env.ProcessProjectDeps()
  File "/root/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "/root/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/piolib.py", line 1062:
    project.install_dependencies()
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/piolib.py", line 898:
    lm.install(spec)
  File "/usr/local/lib/python3.6/dist-packages/platformio/package/manager/_install.py", line 49:
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "/usr/local/lib/python3.6/dist-packages/platformio/package/manager/library.py", line 86:
    force=force,
  File "/usr/local/lib/python3.6/dist-packages/platformio/package/manager/_install.py", line 99:
    pkg = self.install_from_registry(spec, search_filters, silent=silent)
  File "/usr/local/lib/python3.6/dist-packages/platformio/package/manager/_registry.py", line 102:
    for url, checksum in RegistryFileMirrorIterator(pkgfile["download_url"]):
  File "/usr/local/lib/python3.6/dist-packages/platformio/package/manager/_registry.py", line 56:
    else None,
  File "/usr/local/lib/python3.6/dist-packages/platformio/util.py", line 73:
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/clients/http.py", line 131:
    raise HTTPClientError(str(e))
========================= [FAILED] Took 65.82 seconds =========================
```

**Additional information and things you've tried:**

NateLogan, 2021-02-28:
I had the similiar problem today with new build (or after running clean build). No idea what's going on. But manually adding those missing dependancies helped:
```
esphome:
...
  platformio_options:
    lib_deps: 
      - AsyncTCP
      - me-no-dev/ESP Async WebServer
  ```

glmnet, 2021-02-28:
This should be a transient issues, probably caused by dns servers or some weird issues with internet, it is not recommended to use these versions of these libraries as they are known to have issues, this is why ESPHome uses other versions

glmnet, 2021-02-28:
BTW, I just run a test and get the library, no problems

````
ESPAsyncTCP-esphome @ 1.2.3
````

closed 2021-02-28 by glmnet

njfberg, 2021-03-01:
@glmnet , can you share with us why you closed this issue? As stated in the description, the issue does not occur each time. It happened here yesterday and also @NateLogan faced it.

glmnet, 2021-03-01:
> This should be a transient issues, probably caused by dns servers or some weird issues with internet, it is not recommended to use these versions of these libraries as they are known to have issues, this is why ESPHome uses other versions