esphome/issues #1851: impossible to have 2 servos on nodemcu

hokagegano, 2021-02-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Nodemcu
**ESPHome version (latest production, beta, dev branch)**

1.16.1
**Affected component:**

https://esphome.io/components/servo.html

**Description of problem:**
Only one servo works fine the other doesn't work but no error even in debug

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: servo
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "mysuperwifi"
  password: "mysuperpassword"
  manual_ip:
    static_ip: 192.168.xx.xx
    gateway: 192.168.xx.xx
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Servo Fallback Hotspot"
    password: "xxxxxxxxxxxx"

captive_portal:

# Enable logging
logger:
  level: debug

debug:

api:
  password: "ganopacha"
  services:
    - service: control_servo
      variables:
        level: float
      then:
        - servo.write:
            id: my_servo
            level: !lambda 'return level / 100.0;'
    - service: control_servo2
      variables:
        level: float
      then:
        - servo.write:
            id: my_servo2
            level: !lambda 'return level / 100.0;'            

ota:
  password: "ganopacha"

# Example configuration entry
servo:
  - id: my_servo
    output: pwm_output
  - id: my_servo2
    output: pwm_output2
    
# Example output platform
# On ESP32, use ledc output
output:
  - platform: esp8266_pwm
    id: pwm_output
    pin: D5
    frequency: 50 Hz
  - platform: esp8266_pwm
    id: pwm_output2
    pin: D1
    frequency: 50 Hz

```

**Logs (if applicable):**

```
[15:20:20][D][debug:023]: ESPHome version 1.16.1
[15:20:20][D][debug:025]: Free Heap Size: 33968 bytes
[15:20:20][D][debug:053]: Flash Chip: Size=4096kB Speed=40MHz Mode=DIO
[15:20:20][D][debug:190]: Chip ID: 0x00D61960
[15:20:20][D][debug:191]: SDK Version: 2.2.2-dev(38a443e)
[15:20:20][D][debug:192]: Core Version: 2_7_4
[15:20:20][D][debug:193]: Boot Version=6 Mode=1
[15:20:20][D][debug:194]: CPU Frequency: 80
[15:20:20][D][debug:195]: Flash Chip ID=0x0016405E
[15:20:20][D][debug:196]: Reset Reason: External System
[15:20:20][D][debug:197]: Reset Info: External System
[15:20:27][D][servo:059]: Servo new target: 1.000000
[15:20:27][D][servo:049]: Servo reached target
[15:20:29][D][servo:059]: Servo new target: 1.000000
[15:20:29][D][servo:049]: Servo reached target

```

**Additional information and things you've tried:**
The firtst servo is good 
the second nothing happens

glmnet, 2021-02-27:
Does it work if you disable the first one? (Just to check is a sw issue)

glmnet, 2021-03-02:
This is working ok for me

closed 2021-07-14 by stale[bot]