esphome/issues #1856: Captive Portal will not start with a static IP

penright, 2021-02-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

VirtualBox (.vdi) (supported OS)

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32 D1 mini
esp32doit-devkit-v1

**ESPHome version (latest production, beta, dev branch)**

HA ESPHome add on Version 1.16.1

**Affected component:**

https://esphome.io/components/wifi.html
https://esphome.io/components/captive_portal.html

**Description of problem:**
I was testing the ability to recover from an issue where you could not connect to WIFI. I was working through it while the ESP32 was on the bench and therefore easy to reflash via USB. So I was on purpose setting the password incorrectly. Because of various reasons, I was using static IP because the wifi was on a different VLan, and xxxx.local would not resolve. I did not want to chase whatever IP DHCP gave out at the time, so I added static IP to the WIFI section. I was expecting to be able to connect via the AP fall back and the captive portal. If I removed the static IP, then I could connect to the captive portal and set the correct password. If I was connected to the flasher, I could see the IP that DHCP assigned. Then I could add use_address instead of static IP and I could OTA update. For when I get this to my house, I will be able to lock the IP to the mac, otherwise, I would be in the same boat since all my IOT are on their own VLan.
Because of the socket error, the AP keeps resetting itself.

The captive portal will try but will not start after the time out connecting to wifi. As you will see in the "Expected Behavior" I had a desire to have a static IP. When I get home into production, it will not be an issue for me but I thought I would post this in case someone else needed the static IP and use_address is not an option. This is my first ESPHome, but my guess is this is an edge case and between the static IP and "maybe" the captive portal expecting 192.168.4.1 it is getting confused.

[10:58:30][I][wifi:116]: Starting fallback AP!
[10:58:30][C][wifi:161]: Setting up AP...
[10:58:30][C][wifi:163]: AP SSID: 'FB Portal'
[10:58:30][C][wifi:164]: AP Password: 'xxxx'
[10:58:30][C][wifi:167]: AP Static IP: '172.25.92.100'
[10:58:30][C][wifi:168]: AP Gateway: '172.25.92.1'
[10:58:30][C][wifi:169]: AP Subnet: '255.255.255.0'
[10:58:30][C][wifi:173]: IP Address: 172.25.92.100
[10:58:30][E][WiFiUdp.cpp:52] begin(): could not create socket: 23
[10:58:30][E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9

**Problem-relevant YAML-configuration entries:**
```yaml
wifi:
  ssid: !secret wifi_ssid_work
#  password: !secret wifi_password_work
  password: 'testtest'
  use_address: 172.25.92.113

  manual_ip:
    static_ip: 172.25.92.100
    gateway: 172.25.92.1
    subnet: 255.255.255.0  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: !secret wifi_esphome_fallback_ssid
    password: !secret wifi_esphome_fallback_password

captive_portal:

```

**Logs (if applicable):**

```
[11:16:16][E][WiFiUdp.cpp:52] begin(): could not create socket: 23
[11:16:16][E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9
[11:16:16][I][wifi:116]: Starting fallback AP!
[11:16:16][C][wifi:161]: Setting up AP...
[11:16:16][C][wifi:163]:   AP SSID: 'FB Portal'
[11:16:16][C][wifi:164]:   AP Password: 'xxxxx'
[11:16:16][C][wifi:167]:   AP Static IP: '172.25.92.100'
[11:16:16][C][wifi:168]:   AP Gateway: '172.25.92.1'
[11:16:16][C][wifi:169]:   AP Subnet: '255.255.255.0'
[11:16:16][C][wifi:173]:   IP Address: 172.25.92.100
[11:16:16][E][WiFiUdp.cpp:52] begin(): could not create socket: 23
[11:16:16][E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9
[11:16:16][I][wifi:116]: Starting fallback AP!
[11:16:16][C][wifi:161]: Setting up AP...
[11:16:16][C][wifi:163]:   AP SSID: 'FB Portal'
[11:16:16][C][wifi:164]:   AP Password: 'xxxxxx'
[11:16:16][C][wifi:167]:   AP Static IP: '172.25.92.100'
[11:16:16][C][wifi:168]:   AP Gateway: '172.25.92.1'
[11:16:16][C][wifi:169]:   AP Subnet: '255.255.255.0'
[11:16:16][C][wifi:173]:   IP Address: 172.25.92.100
[11:16:16][E][WiFiUdp.cpp:52] begin(): could not create socket: 23
[11:16:16]abort() was called at PC 0x4015024f on core 1
[11:16:16]
[11:16:16]Backtrace: 0x4008c744:0x3ffb1a20 0x4008c975:0x3ffb1a40 0x4015024f:0x3ffb1a60 0x40150296:0x3ffb1a80 0x4014f327:0x3ffb1aa0 0x4014f43e:0x3ffb1ac0 0x4014f3dd:0x3ffb1ae0 0x400e13ab:0x3ffb1b00 0x400e65d5:0x3ffb1b40 0x400d62e3:0x3ffb1b70 0x400d62ee:0x3ffb1b90 0x4015dc79:0x3ffb1bb0 0x4015dd0d:0x3ffb1bd0 0x400dc281:0x3ffb1bf0 0x400de3f2:0x3ffb1c40 0x400e8b5b:0x3ffb1fb0 0x40088e8d:0x3ffb1fd0
[11:16:16]
[11:16:16]Rebooting...
[11:16:16]ets Jun  8 2016 00:22:57
[11:16:16]
[11:16:16]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[11:16:16]configsip: 0, SPIWP:0xee
[11:16:16]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[11:16:16]mode:DIO, clock div:2
[11:16:16]load:0x3fff0018,len:4
[11:16:16]load:0x3fff001c,len:1044
[11:16:16]load:0x40078000,len:8896
[11:16:16]load:0x40080400,len:5828
[11:16:16]entry 0x400806ac
[11:16:16][I][logger:166]: Log initialized
[11:16:16][C][status_led:014]: Setting up Status LED...
[11:16:16][C][ota:366]: There have been 4 suspected unsuccessful boot attempts.
[11:16:16][I][app:029]: Running through setup()...
[11:16:16][C][wifi:033]: Setting up WiFi...
[11:16:16]
[11:16:16]Brownout detector was triggered
[11:16:16]
[11:16:17]ets Jun  8 2016 00:22:57
[11:16:17]
[11:16:17]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[11:16:17]configsip: 0, SPIWP:0xee
[11:16:17]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[11:16:17]mode:DIO, clock div:2
[11:16:17]load:0x3fff0018,len:4
[11:16:17]load:0x3fff001c,len:1044
[11:16:17]load:0x40078000,len:8896
[11:16:17]load:0x40080400,len:5828

```

**Additional information and things you've tried:**
The easiest way to duplicate the issue is to create a base project and on purpose type, the password wrong needed to connect to the AP. Then it will fall back and you will get a captive portal. Then add a static IP to the WIFI component. Now when it tries to start the captive portal you will see the socket errors.

closed 2021-07-03 by stale[bot]

srnoth, 2021-09-15:
This is still an issue, should not have been closed. Just discovered the captive portal function is completely useless when using static IPs.

srnoth, 2021-09-15:
@penright do you have the option to reopen this?

srnoth, 2021-09-16:
Well... @penright and others seems the solution is you also have to specify a static IP under the "ap" section. 

Example:
```
wifi:
  ssid: xxx
  password: xxx
  reboot_timeout: 0s
  manual_ip:
    static_ip: x.x.x.x.x
    gateway: x.x.x.x.x
    subnet: x.x.x.x.x
  ap:
    password: xxx
    channel: 6
    ap_timeout: 20s
    manual_ip:
      static_ip: x.x.x.x.x
      gateway: x.x.x.x.x
      subnet: x.x.x.x.x
```

Failover captive portal works perfectly this way.