esphome/issues #1862: ESPHome HA Addon 1.16.2 missing configuration section

pcartwright81, 2021-02-20:
DO NOT DELETE ANY TEXT from this template! Otherwise the issue may be closed without a comment.
-->

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

I hit reset on the configuration section, now it is missing.

[image]
[image]

I have been looking for a while, and I can't seem to find where the actual file is because I have always used the editor to change the version.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

closed 2021-02-20 by pcartwright81

Protoncek, 2021-02-20:
@pcartwright81 : Hello!
Did you solve the problem? How? Because i have the same situation...

pcartwright81, 2021-02-20:
Since this is a home assistant issue I opened it on the home assistant issue tracker.
It is a supervisor issue and will be fixed in the latest beta.
https://github.com/home-assistant/core/issues/46826

Protoncek, 2021-02-20:
Ok, thanks a lot!