esphome/issues #1863: st7735 has various small problems preventing support of M5Stick-C

wjcarpenter, 2021-02-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Standalone pip3 esphome 1.16.2

Standalone pip3 esphome 1.16.2

**ESP (ESP32/ESP8266, Board/Sonoff):** 
ESP32, M5Stick-C

**ESPHome version (latest production, beta, dev branch)**
latest production

1.16.2
**Affected component:**

https://esphome.io/components/display/st7735.html

**Description of problem:**
Support for the st7735 was recently added, but it does not quite support that display in the M5Stick-C. 

It most closely resembles the model INITR_MINI160X80. but that model has "colstart" hardcoded to 24 in the C++ code. M5Stick-C needs a value of 26. It would be a good enough fix to eliminate the hardcoding and accept the configured value, as is done for other models.

The display must also have the ST77XX_INVON command sent to it during initialization. A reasonable thing would be to have a new optional boolean configuration property, for example, `invert` with a default value of false. When set to true,
```
  if (this->invert_) {
    sendcommand_(ST77XX_INVON, &data, 0);
  }
```
during device initialization. Assuming these changes, a YAML config for the display part of the M5Stick-C is given in the YAML section below.

**Problem-relevant YAML-configuration entries:**
```
display:
  - platform: st7735
    model: INITR_MINI160X80
    rowstart: 0
    colstart: 26
    devicewidth: 0
    deviceheight: 0
    cs_pin: GPIO5
    dc_pin: GPIO23
    reset_pin: GPIO18
    rotation: 270
    usebgr: true
    invert: true
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I have made these changes locally and gotten the display working completely on M5Stick-C. I could supply a pull request if that would be helpful. (The only reason I didn't automatically send a PR is because I see that some additional work has been done in the `dev` branch for the st7735 since the 1.16.2 release.

wjcarpenter, 2021-02-23:
In case it is useful, here is a diff from the esphome 1.16.2 component and my changes:
```
diff -u /usr/local/lib/python3.8/dist-packages/esphome/components/st7735/display.py ./display.py
--- /usr/local/lib/python3.8/dist-packages/esphome/components/st7735/display.py	2021-02-20 15:22:15.631630420 -0800
+++ ./display.py	2021-02-21 16:11:12.005000912 -0800
@@ -17,6 +17,7 @@
 CONF_COLSTART = 'colstart'
 CONF_EIGHTBITCOLOR = 'eightbitcolor'
 CONF_USEBGR = 'usebgr'
+CONF_INVERT = 'invert'
 
 SPIST7735 = st7735_ns.class_('ST7735', cg.PollingComponent, display.DisplayBuffer, spi.SPIDevice)
 ST7735Model = st7735_ns.enum('ST7735Model')
@@ -46,6 +47,7 @@
     cv.Required(CONF_ROWSTART):  cv.int_,
     cv.Optional(CONF_EIGHTBITCOLOR, default=False): cv.boolean,
     cv.Optional(CONF_USEBGR, default=False):  cv.boolean,
+    cv.Optional(CONF_INVERT, default=False):  cv.boolean,
 }).extend(cv.COMPONENT_SCHEMA).extend(spi.spi_device_schema()),
                        cv.has_at_most_one_key(CONF_PAGES, CONF_LAMBDA))
 
@@ -67,7 +69,7 @@
 def to_code(config):
     var = cg.new_Pvariable(config[CONF_ID], config[CONF_MODEL], config[CONF_DEVICEWIDTH],
                            config[CONF_DEVICEHEIGHT], config[CONF_COLSTART], config[CONF_ROWSTART],
-                           config[CONF_EIGHTBITCOLOR], config[CONF_USEBGR])
+                           config[CONF_EIGHTBITCOLOR], config[CONF_USEBGR], config[CONF_INVERT])
     yield setup_st7735(var, config)
     yield spi.register_spi_device(var, config)
 
Common subdirectories: /usr/local/lib/python3.8/dist-packages/esphome/components/st7735/__pycache__ and ./__pycache__
diff -u /usr/local/lib/python3.8/dist-packages/esphome/components/st7735/st7735.cpp ./st7735.cpp
--- /usr/local/lib/python3.8/dist-packages/esphome/components/st7735/st7735.cpp	2021-02-20 15:22:15.631630420 -0800
+++ ./st7735.cpp	2021-02-21 16:23:05.625580311 -0800
@@ -221,7 +221,7 @@
 static const char *TAG = "st7735";
 
 ST7735::ST7735(ST7735Model model, int width, int height, int colstart, int rowstart, boolean eightbitcolor,
-               boolean usebgr) {
+               boolean usebgr, boolean invert) {
   model_ = model;
   this->width_ = width;
   this->height_ = height;
@@ -229,10 +229,11 @@
   this->rowstart_ = rowstart;
   this->eightbitcolor_ = eightbitcolor;
   this->usebgr_ = usebgr;
+  this->invert_ = invert;
 }
 
 void ST7735::setup() {
-  ESP_LOGCONFIG(TAG, "Setting up ST7735...");
+  ESP_LOGCONFIG(TAG, "Setting up ST7735 m5...");
   this->spi_setup();
 
   this->dc_pin_->setup();  // OUTPUT
@@ -264,8 +265,8 @@
     height_ == 0 ? height_ = ST7735_TFTHEIGHT_160 : height_;
     width_ == 0 ? width_ = ST7735_TFTWIDTH_80 : width_;
     display_init_(RCMD2GREEN160X80);
-    colstart_ = 24;
-    rowstart_ = 0;  // For default rotation 0
+    colstart_ = (colstart_ == 0) ? 24 : colstart_;
+    //rowstart_ = (rowstart_ == 0) ? 0 : rowstart_;  // becomes a no-op
   } else {
     // colstart, rowstart left at default '0' values
     display_init_(RCMD2RED);
@@ -274,7 +275,10 @@
 
   uint8_t data = 0;
   if (this->model_ != INITR_HALLOWING) {
-    uint8_t data = ST77XX_MADCTL_MX | ST77XX_MADCTL_MY;
+    data = ST77XX_MADCTL_MX | ST77XX_MADCTL_MY;
+  }
+  if (this->invert_) {
+    sendcommand_(ST77XX_INVON, &data, 0);
   }
   if (this->usebgr_) {
     data = data | ST7735_MADCTL_BGR;
@@ -379,6 +383,9 @@
   ESP_LOGD(TAG, "  Width: %d", this->width_);
   ESP_LOGD(TAG, "  ColStart: %d", this->colstart_);
   ESP_LOGD(TAG, "  RowStart: %d", this->rowstart_);
+  ESP_LOGD(TAG, "  EightBitColor: %s", this->eightbitcolor_ ? "true" : "false");
+  ESP_LOGD(TAG, "  UseBGR: %s", this->usebgr_ ? "true" : "false");
+  ESP_LOGD(TAG, "  Invert: %s", this->invert_ ? "true" : "false");
   LOG_UPDATE_INTERVAL(this);
 }
 
diff -u /usr/local/lib/python3.8/dist-packages/esphome/components/st7735/st7735.h ./st7735.h
--- /usr/local/lib/python3.8/dist-packages/esphome/components/st7735/st7735.h	2021-02-20 15:22:15.631630420 -0800
+++ ./st7735.h	2021-02-21 16:13:50.987359454 -0800
@@ -37,7 +37,7 @@
                public spi::SPIDevice<spi::BIT_ORDER_MSB_FIRST, spi::CLOCK_POLARITY_LOW, spi::CLOCK_PHASE_LEADING,
                                      spi::DATA_RATE_8MHZ> {
  public:
-  ST7735(ST7735Model model, int width, int height, int colstart, int rowstart, boolean eightbitcolor, boolean usebgr);
+  ST7735(ST7735Model model, int width, int height, int colstart, int rowstart, boolean eightbitcolor, boolean usebgr, boolean invert);
   void dump_config() override;
   void setup() override;
 
@@ -77,6 +77,7 @@
   uint8_t colstart_ = 0, rowstart_ = 0;
   boolean eightbitcolor_ = false;
   boolean usebgr_ = false;
+  boolean invert_ = false;
   int16_t width_ = 80, height_ = 80;  // Watch heap size
 
   GPIOPin *reset_pin_{nullptr};
```

geiseri, 2021-06-04:
Any forward movement here?  It looks like the colors might be correct now but the background is solid white.

closed 2021-10-16 by stale[bot]

geiseri, 2021-10-17:
still broken