esphome/issues #1864: Cannot use A0 and VCC at same time

jbrepogmailcom, 2021-02-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASSIO

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**ESPHome version (latest production, beta, dev branch)**

latest prod

**Affected component:**

A0 and VCC

**Description of problem:**

Hello. I would like to measure voltage on A0 pin and also VCC battery voltage. If I program it in C++, it works. But following configuration only sends one mqtt sensor - I suspect that one definition overrides the other. Is there any other way how to define both sensors?

```
sensor:
  - platform: adc
    id: pracka2_vcc
    pin: VCC
    name: "Pracka2 VCC"
    retain: true
    update_interval: 30s
    accuracy_decimals: 2

  - platform: adc
    id: pracka2_pr
    pin: A0
    name: "Pracka2 Photo Resistor"
    retain: true
    update_interval: 30s
    accuracy_decimals: 3
```

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2021-02-27:
Docs says that if you want to measure HCC you must connect nothing to A0. Isn't it accurate?

jbrepogmailcom, 2021-03-06:
Actually, it is not correct. Lets say you have two digital pins GPIO12 and GPIO13 and analog pin A0. You connect photoresistor between A0 and GPIO12 and you connect GPIO13 to A0.
Then you set both GPIO12 and GPIO13 as INPUT. Therefore nothing is connected to A0 and you can measure battery Vcc. Then you set GPIO13 INPUT_PULLUP and GPIO12 OUTPUT LOW. This way you create voltage divider from + voltage internal resistor to GPIO13 to A0 to photoresistor to ground on GPIO12 and you can measure variable voltage on A0 that depends on light intensity.
This way you can measure both Vcc on disconnected A0 and voltage on A0. Therefore I would request to enable both VCC and A0 pins in one yaml configuration for sensor.