esphome/issues #1865: SIM800L: SMS decoding issue

andriej, 2021-02-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HA Supervised

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**ESPHome version (latest production, beta, dev branch)**

1.16.1

**Affected component:**

https://esphome.io/components/sim800l.html

**Description of problem:**
After receiving SMS with emoji, SIM800l is not decoding text from it:
[image]

**Problem-relevant YAML-configuration entries:**
```yaml
n/a
```

**Logs (if applicable):**

```
[17:31:50][D][sim800l:174]: Received SMS from: +48xxxxxxxxx
[17:31:50][D][sim800l:175]: 0046006900720065D83DDD25

```

**Additional information and things you've tried:**
It's "Fire" with emoji message (after [decoding](https://gchq.github.io/CyberChef/#recipe=From_Hex('Auto')&input=MDA0NjAwNjkwMDcyMDA2NUQ4M0RERDI1))

andriej, 2021-02-22:
Same goes with polish alphabet letters: https://en.wikipedia.org/wiki/Polish_alphabet

andriej, 2021-06-23:
Still valid

andriej, 2021-09-04:
@glmnet any chance to easy fix this? 
In my (not dev) opinion it could be just matter of detecting those and re-formatting them.

It applies also for i.e. polish alphabet letters if sender's app send those as (I guess) UTF

ingria, 2024-06-11:
Also non-numeric sender decoding is broken.

```
[D][sim800l:322]: Received SMS from: 4556p65623
```
It should've been decoded as Tele2.

It looks like that string is a correct ascii sequence (`54 65 6c 65 32`) but with swapped endianness (`45 56 p6 56 23`). Not sure about the `p` letter in the third octet though.