esphome/issues #1866: display component has UnicodeDecodeError

randompee, 2021-02-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip on Raspi 3B+
**ESP (ESP32/ESP8266, Board/Sonoff):**

e.g. esp32doit-devkit-v1
**ESPHome version (latest production, beta, dev branch)**

1.17.0-dev
**Affected component:**

https://esphome.io/components/display/index.html

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
display:
...
         lambda: |-
              it.printf(0,  6, id(my_font5), "CO""\u2082");
...
font:
   - file: "DejaVuSans.ttf"
     id: my_font5
     size: 20
```

**Logs (if applicable):**

```
  File "/home/pi/.local/lib/python3.7/site-packages/google/protobuf/internal/decoder.py", line 572, in _ConvertToUnicode
    value = local_unicode(byte_str, 'utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe2 in position 80: 'utf-8' codec can't decode byte 0xe2 in position 80: invalid continuation byte in field: 

```

**Additional information and things you've tried:**

when I added the (optional!!) line 

        glyphs: " ()CHMOPRT%\u2082\u00B0"

to the font, all was well, the UnicodeDecodeError went away. I did not give more info on the ESPs and so on because I tested with two different ones and two different displays, and the results were the same.

OttoWinter, 2021-04-10:
Are you sure that is related to font (can you double check if it really only happens when you don't add that line)?

That stacktrace you posted is not loaded by esphome during compile time, and especially not by the font system. Also please attach a complete stacktrace (usually they contain many lines listing each stack frame).

randompee, 2021-04-12:
I commented the "glyphs:" entry for one of the fonts and compiled. Compilation showed no exception. After succesful OTA and reboot the expected exception showed in the log:
`[13:32:21][I][app:105]: ESPHome version 1.18.0-dev compiled on Apr 12 2021, 13:25:32

...
13:32:21][C][i2c:028]: I2C Bus:
[13:32:21][C][i2c:029]:   SDA Pin: GPIO21
[13:32:21][C][i2c:030]:   SCL Pin: GPIO22
[13:32:21][C][i2c:031]:   Frequency: 50000 Hz
[13:32:21][I][i2c:033]: Scanning i2c bus for active devices...
[13:32:21][I][i2c:040]: Found i2c device at address 0x3C
[13:32:21][I][i2c:040]: Found i2c device at address 0x77
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/home/pi/.local/lib/python3.7/site-packages/esphome/api/client.py", line 407, in run
    self._run_once()
  File "/home/pi/.local/lib/python3.7/site-packages/esphome/api/client.py", line 397, in _run_once
    msg.ParseFromString(raw_msg)
  File "/home/pi/.local/lib/python3.7/site-packages/google/protobuf/message.py", line 199, in ParseFromString
    return self.MergeFromString(serialized)
  File "/home/pi/.local/lib/python3.7/site-packages/google/protobuf/internal/python_message.py", line 1145, in MergeFromString
    if self._InternalParse(serialized, 0, length) != length:
  File "/home/pi/.local/lib/python3.7/site-packages/google/protobuf/internal/python_message.py", line 1212, in InternalParse
    pos = field_decoder(buffer, new_pos, end, self, field_dict)
  File "/home/pi/.local/lib/python3.7/site-packages/google/protobuf/internal/decoder.py", line 619, in DecodeField
    field_dict[key] = _ConvertToUnicode(buffer[pos:new_pos])
  File "/home/pi/.local/lib/python3.7/site-packages/google/protobuf/internal/decoder.py", line 572, in _ConvertToUnicode
    value = local_unicode(byte_str, 'utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe2 in position 80: 'utf-8' codec can't decode byte 0xe2 in position 80: invalid continuation byte in field: SubscribeLogsResponse.message
`
I am not sure what additional information I can provide to help analyze this situation. If a stack trace still makes sense I will find out how to produce that...

closed 2021-08-24 by stale[bot]