esphome/issues #1868: Nextion sends val instead of pic/picc for set_component_picture()

pashdown, 2021-02-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**ESPHome version (latest production, beta, dev branch)**

v1.16.2

**Affected component:**

Nextion

**Description of problem:**
Nextion has the ability to set picture on a picture component with:
```
componentname.pic=1
```
and on a crop component with:
```
componentname.picc=1
```
In set_component_picture as it stands now, it sends 'val' which is wrong in either case.  Ideally there would be two new functions, one for crop and one for picture, that sends the proper postfix.

The crop component makes layout toggles very easy, so it would be nice if it was functional.

**Problem-relevant YAML-configuration entries:**
```yaml
display:
  - platform: nextion
    lambda: |-
      it.set_component_picture("westmotion" , "1");
      it.set_component_picture("eastmotion" , "1");
```

**Logs (if applicable):**

```
[13:25:55][W][nextion:177]: Nextion reported variable name invalid!
[13:25:55][W][nextion:037]: Waiting for ACK timed out!
[13:25:55][W][nextion:084]: Sending command 'westmotion.val=0' failed because no ACK was received
[13:25:55][W][nextion:177]: Nextion reported variable name invalid!
[13:25:55][W][nextion:037]: Waiting for ACK timed out!
[13:25:55][W][nextion:084]: Sending command 'eastmotion.val=0' failed because no ACK was received

```

**Additional information and things you've tried:**

kbx81, 2021-03-23:
Maybe this is fixed here?: https://github.com/esphome/esphome/pull/1464

rarroyo6, 2021-03-23:
I am very interested in testing this, but I have never installed from a PR.
How do I install this into ESPHome in order to test it?
Is there a time frame on when this will be merged into the ESPHome?
I'm new at doing this, so any help is appreciated.

Thank you.

pashdown, 2021-04-09:
Fixed by [ca151eb](https://github.com/esphome/esphome/pull/1464/commits/ca151ebcbcdd253ed3f9aded1884d4eeb5d69b4e)

closed 2021-04-09 by pashdown

projectdimpreza, 2021-04-14:
how did u fix this?

pashdown, 2021-04-14:
Wait for the new Nextion code to be merged.

projectdimpreza, 2021-04-15:
> Wait for the new Nextion code to be merged.

im new to github.. what to u mean when u say merged? sorry im really new to this...

pashdown, 2021-04-15:
The code is in review right now. It will probably be a month or so until it is merged and part of the mainline release.  When that happens, just update and you should be able to use the new features.