esphome/issues #1874: esp32 is unable to get values from HA sensors or upload sensor values to HA if a display component (ssd1327_i2c) is used

bea30, 2021-02-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
link: https://www.ebay.co.uk/itm/ESP32-Development-Board-ESP-32-38-pin-DevKitC-Layout-UK-Seller-/322484359954
**ESPHome version (latest production, beta, dev branch)**

ESPHome version 1.16.2
**Affected component:**

https://esphome.io/components/display/ssd1327.html
https://esphome.io/components/binary_sensor/homeassistant.html

**Description of problem:**
When a display component is defined and used over i2c all attached sensors on the board stops to uploading their states to HA and their status in HA is unavailable. (MHZ19, DHT22, BME680, MQ2 sensors) No matter if display only shows static text or if display is showing their values.  Display component is able to show values of sensors without any problem.

If a display component is commented or removed all states are uploaded to HA without any problem.

I am using following display marked as gme128128-02 v3.1 with 4 pins - GNC, VCC, SCL,SDA
https://www.aliexpress.com/item/1005001462777917.html?spm=a2g0o.productlist.0.0.3b8427dd7eZSAs&algo_pvid=337d1302-6f6b-4b02-9f3e-1a6db797d95b&algo_expid=337d1302-6f6b-4b02-9f3e-1a6db797d95b-4&btsid=0bb0623616142669305485604e3fb9&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

Display connection:
SDA -> 21
SCL -> 22
VCC -> 3v3
GND -> GND

or 

SDA -> 21
SCL -> 22
VCC -> 5V
GND -> GND

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test_esp32
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: "xxxx"
  password: "xxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Esp32 Fallback Hotspot"
    password: "xxxx"

captive_portal:

# Enable logging
logger:
  level: VERY_VERBOSE
  logs:
    i2c: DEBUG

# Enable Home Assistant API
api:
  password: "xxxx"

ota:
  password: "xxxx"

# i2c - displej
i2c:
 - id: bus_a
   sda: 21
[... 981 lines cut ...]
```

Error from HA:
```
Logger: homeassistant.components.esphome
Source: components/esphome/__init__.py:197
Integration: ESPHome (documentation, issues)
First occurred: 2:13:20 PM (216 occurrences)
Last logged: 4:02:21 PM

Error getting initial data for 192.x.x.154: Timeout while waiting for API response!
```

**Additional information and things you've tried:**

WHAT I ' VE TRIED:
1. show sensor values to the display. All values are showed properly and are updated on oled display, but their values are not uploaded to HA. HA marked them as unavailable
2. print only static text to the display - hello world. All sensor values are not uploaded to HA. HA marked them as unavailable
3. I tried to display not values from attached sensors but read their values as a sensors of a homeassistant platform and then print them to display. Same behaviour, values of sensors are not uploaded to HA. HA marked them as unavailable. And display shows nand values.
4. I tried to use i2c another bus (esp32 has a two of them), same issue
5. I tried to use another dev board - esp32 - same issue
6. I tried to attach display to another esp32 display, without any sensors and read values from HA as a homeassistant platform sensors
7. Also tried do change board type to esp32dev

I will be thankful for any advice.

pasiz, 2021-02-26:
Have you tried supply stable 3.3v directly to board. Seems like power issue in first sight...

bea30, 2021-02-26:
Thank you for your reply. No I did not try 3.3v power source. I am using 5v/2A power supply.  I ve tried multiple 5v adapters with max 2A output. Now I tried stronger adapter - 5v-3A with same behaviour. 

So I think that is not related with power isssue. 2A as a max output seems to me as a sufficient power source for only esp32 and one display. There is a same behaviour as when was multiple sensors connected with display to esp. Now as you can see in yaml I am using only a display connected to esp. Also I tried to powering up a display with 5v - (VCC).

bea30, 2021-02-26:
Also tried a another esp32 board - wroom-32u with external wifi antena connected with display.
https://www.tme.eu/sk/details/esp32-devkitc-32u/supravy-na-prenos-dat/espressif/?brutto=1&gclid=EAIaIQobChMIzPuSl6uH7wIVibLVCh0VZwToEAQYASABEgJIDfD_BwE

Same issue. It cant retreive values from HA if a display component(ssd1327_i2c) is defined.
Same logs output.

pasiz, 2021-02-26:
> Thank you for your reply. No I did not try 3.3v power source. I am using 5v/2A power supply. I ve tried multiple 5v adapters with max 2A output. Now I tried stronger adapter - 5v-3A with same behaviour.
> 
> So I think that it is not related to power isssue. 2A as a max output seems to me as a sufficient power source for only esp32 and one display. There is a same behaviour as when was multiple sensors connected with display to esp. Now as you can see in yaml I am using only a display connected to esp. Also I tried to powering up a display with 5v - (VCC).

ESP32 is powered with 3.3v. If your internal regulator is not sufficient to provide enough 3.3v power, it would escalate like the behavior you described.

bea30, 2021-02-26:
I thought that internal voltage regulator is not used for the display powering if the displays vcc port is connected to 5V esp port for a display. So behaviour of the esp board should be the same if the display is connected or not.

Ok then because I dont have any 3.3V power supply I made a another test. Display was powered up from one 5v/2a (vcc, gnd) power supply and esp board was powered up from another power suply 5v/2a through a usb port.

 Is this kind of test enought to exclude a problem with internal regulator?

pasiz, 2021-02-28:
Okay, understanded. 

For some reason, like your logs say:
`[15:58:34][V][app:081]: A component took a long time in a loop() cycle (1.84 s).`
`[15:58:34][V][app:082]: Components should block for at most 20-30ms in loop().`

So i suggest there is something wrong, in the display, or connection, thus blocking the mainloop.

Do you see those lines if you just draw something in the display without parsing any data. Have you tried different bus options on the display?

pasiz, 2021-02-28:
Also try without very_verbose logging, that is causing often troubles on esp devices as stated on documentation.

https://esphome.io/components/logger.html

bea30, 2021-03-01:
> Okay, understanded.
>
> For some reason, like your logs say:
> `[15:58:34][V][app:081]: A component took a long time in a loop() cycle (1.84 s).`
> `[15:58:34][V][app:082]: Components should block for at [...]

No matter what is on screen. Same behaviour spotted if there is only static text or values of the attached sensors (if there some). It is able to display values of attached sensors but as I described in original post after that is sensors are marked as unavailable in HA. In this mode there is also same verbose line:

```
[15:58:34][V][app:081]: A component took a long time in a loop() cycle (1.84 s).
[15:58:34][V][app:082]: Components should block for at most 20-30ms in loop().
```

> 
> Do you see those lines if you just draw something in the display without parsing any data. Have you tried different bus options on the display?

No, but I tried different bus on ESP32. I think that this is bug because when I wrote a simple peace of code aimed to test using wifi and display at the same time it works without any problem. The main purpose of the code was connect to wifi, try GET to some url and print out to screen IP adress. Displaying values was in for cycle, and printed out with changing possition from top to bottom in one loop() cycle.

So according to this observation I think that oled screen is OK. Also standard examples from arduinoIDE for ssd1327 driver works.

>Also try without very_verbose logging, that is causing often troubles on esp devices as stated on documentation.
>
>https://esphome.io/components/logger.html

That high logger severity was enabled only for a troubleshooting purposes. My standard logger level is WARNING with a same problem.  I tried VERY_VERBOSE to gain all possible logs.

pasiz, 2021-03-01:
Sorry to ask one more time stupid questions, but you have home assistant sensors that are marked as internal, is that purposeful?

bea30, 2021-03-01:
I am not consider any question as stupid. Every question or pointing me to the right way to resolve this issue is highly appreciated. Esspecialy if the problem may be from a category of PEPKAC 🙂 

Sensors from this published config are purposely marked as internal because this values are get from HA and should be only displayed on the oled screen and not send back to HA. 

On the other esp32 (where are sensors connected) are not configured as internal and their values are sent and stored in HA. 

My first intend was add a display to a esp32 (device A) in living room that can permanently showing air quality values without needed to looking on phone or computer. But when I added a display , printed out values of sensors I noticed that their values are not transmitted to HA. 

So configuration from this esp32 (device b) is my another try to make device that can show values of air quality because on the first device it is not working. Now with architecture that all values will be get from HA (but source of values is device A). But it is also not working.

pasiz, 2021-03-02:
Could you try not to render special characters (degree signs, etc) while debugging. If that makes no difference, then could you provide i2c VV log.

bea30, 2021-03-02:
I found a problem.
### LOGS:
```
[15:56:09][C][ssd1327_i2c:023]: I2C SSD1327
[15:56:09][C][ssd1327_i2c:023]:   Rotations: 0 °
[15:56:09][C][ssd1327_i2c:023]:   Dimensions: 128px x 128px
[15:56:09][C][ssd1327_i2c:024]:   Address: 0x3C
[15:56:09][C][ssd1327_i2c:025]:   Model: SSD1327 128x128
[15:56:09][C][ssd1327_i2c:027]:   Update Interval: 1.0s
```

I noticed in my logs that there is a for some reason **update interval for a display set to 1s** and according to documentation **default value should be 5s** and I did not change this value or explicitly defined in my configuration.

https://esphome.io/components/display/ssd1327.html?highlight=ssd1327

> update_interval (Optional, Time): The interval to re-draw the screen. Defaults to 5s.

So I tried to set **update_interval to 5s** explicitly and communication with HA started to work. Now esp32 is able to to get values from HA.

Here is my working configuration:
```
...
i2c:
 - id: bus_a
   sda: 21
#oranzovy 22, zlty 21
   scl: 22 
   scan: True
#   frequency: 200kHz

sensor:
  - platform: homeassistant
    id: temp
    entity_id: sensor.spalna_bme_teplota
    internal: true

  - platform: homeassistant
    id: humidity
    entity_id: sensor.spalna_bme_vlhkost
    internal: true
  
  - platform: homeassistant
    id: co2
    entity_id: sensor.mh_z19_co2
    internal: true

font:
  - file: "fonts/ComicSansMS3.ttf"
    id: my_font
    size: 20
  - file: "fonts/ComicSansMS3.ttf"
    id: small
    size: 18

display:
  - platform: ssd1327_i2c
    i2c_id: bus_a
    model: "SSD1327 128x128"
    #reset_pin: 
    address: 0x3C
    update_interval: 5s
    lambda: |-
      if (id(temp).has_state()) {
        it.printf(0, 0, id(my_font), "T: %.1fC°", id(temp).state);
      }
      if (id(humidity).has_state()) {
        it.printf(0, 21, id(my_font), "H: %.1f %", id(humidity).state);
      }
      if (id(co2).has_state()) {
        it.printf(0, 41, id(my_font), "CO: %.0f ppm", id(co2).state);
[... 43 lines cut ...]
[13:49:15][V][sensor:013]: 'temp': Received new state 22.100000
[13:49:15][D][sensor:092]: 'temp': Sending state 22.10000  with 1 decimals of accuracy
[13:49:15][VV][api.service:326]: on_home_assistant_state_response: HomeAssistantStateResponse {
  entity_id: 'sensor.spalna_bme_vlhkost'
  state: '41.2'
}
[13:49:15][D][homeassistant.sensor:019]: 'sensor.spalna_bme_vlhkost': Got state 41.20
[13:49:15][V][sensor:013]: 'humidity': Received new state 41.200001
[13:49:15][D][sensor:092]: 'humidity': Sending state 41.20000  with 1 decimals of accuracy
[13:49:15][V][app:081]: A component took a long time in a loop() cycle (1.87 s).
[13:49:15][V][app:082]: Components should block for at most 20-30ms in loop().
[13:49:17][VV][api.service:220]: on_ping_request: PingRequest {}
[13:49:17][VV][api.service:032]: send_ping_response: PingResponse {}
[13:49:18][VV][scheduler:152]: Running interval 'update' with interval=5000 last_execution=451547 (now=456547)
[13:49:20][V][app:081]: A component took a long time in a loop() cycle (1.80 s).
[13:49:20][V][app:082]: Components should block for at most 20-30ms in loop().
[13:49:22][VV][api.service:220]: on_ping_request: PingRequest {}
[13:49:22][VV][api.service:032]: send_ping_response: PingResponse {}
```

**Question is why is a default value of update_interval set to 1s when a documentation says it is 5s.**  Apparently 1s interval breaks network communication between esp32 and HA and I also noticed that sometimes breaks wifi network connection as well.

With a explicitly defined update_interval to 5s display component works also on a device A (mentioned a few comments above) where are attached sensors (BME680, MHZ-19) and renders values straight from them. Their values are also transmitted to HA without any problem.

pasiz Thank you for pushing me to making new tests.

closed 2021-03-02 by bea30