esphome/issues #1875: Getting Kernel panic

justlikeef, 2021-02-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

WSL2 git clone

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 WROVER

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**
Kernel Panic

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```substitutions:
  device_name: "Filter Pump Dev"
  emer_temp: "34"

esphome:
  name: filter_pump_dev
  platform: ESP32
  board: esp-wrover-kit
  on_boot:
    priority: -100
    then:
      - script.execute: show_oper_mode
      - script.execute: show_run_mode
      - script.execute: rotate_display

globals:
  # 0 - Off   1-Local Only  2-Manual   3-Automatic
  - id: oper_mode
    type: int
    restore_value: yes
    initial_value: "3"
  - id: auto_run_state
    type: int
    restore_value: yes
    initial_value: "0"
  - id: man_run_state
    type: int
    restore_value: yes
    initial_value: "0"
  - id: loc_run_state
    type: int
    restore_value: yes
    initial_value: "0"
  - id: emer_run_state
    type: int
    restore_value: yes
    initial_value: "0"

wifi:
  ssid: !secret wireless_ssid
  password: !secret wireless_password
  fast_connect: true
  manual_ip:
    static_ip: 192.168.1.120
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1
    dns2: 8.8.8.8

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
[... 608 lines cut ...]
[20:45:59][D][binary_sensor:034]: 'off_button': Sending initial state OFF
[20:45:59][C][dallas.sensor:032]: Setting up DallasComponent...
[20:45:59][D][text_sensor:015]: 'Filter Pump Dev ESPHome Version': Sending state '1.17.0-dev Feb 25 2021, 20:34:12'
[20:45:59][D][text_sensor:015]: 'Filter Pump Dev Mac Wifi Address': Sending state '3C:71:BF:83:D3:14'
[20:45:59][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[20:46:00][C][wifi:033]: Setting up WiFi...
[20:46:00][I][wifi:194]: WiFi Connecting to 'justlikeef'...
[20:46:00][D][text_sensor:015]: 'Filter Pump Dev Oper Mode': Sending state '3'
[20:46:00][D][text_sensor:015]: 'Filter Pump Dev Uptime': Sending state '1s'
[20:46:00][D][sensor:092]: 'uptime_seconds': Sending state 0.58300 s with 0 decimals of accuracy
[20:46:00][D][switch:045]: 'Filter Pump Dev Run by Schedule': Sending state OFF
[20:46:00][D][main:183]: show_run_mode
[20:46:00][D][main:186]: show_run_mode: call update_pump_status_text
[20:46:00]***ERROR*** A stack overflow in task loopTask has been detected.
[20:46:00]abort() was called at PC 0x4008d580 on core 1
[20:46:00]
[20:46:00]Backtrace: 0x4008d338:0x3ffaff20 0x4008d569:0x3ffaff40 0x4008d580:0x3ffaff60 0x4008a628:0x3ffaff80 0x4008c2b0:0x3ffaffa0 0x4008c266:0xabba1234
[20:46:00]
[20:46:00]Rebooting...

**Additional information and things you've tried:**
Power supply
Multiple development boards

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

justlikeef, 2021-02-26:
Verbose logging:
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[I][logger:166]: Log initialized
[C][heartbeat_led:020]: Setting up Heartbeat LED...
[C][status_led:014]: Setting up Status LED...
[C][ota:366]: There have been 2 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[V][app:030]: Sorting components by setup priority...
[C][uart_esp32:072]: Setting up UART...
[C][switch.gpio:011]: Setting up GPIO Switch 'phase_a'...
[D][switch:025]: 'phase_a' Turning OFF.
[D][switch:045]: 'phase_a': Sending state OFF
[D][switch:025]: 'phase_a' Turning OFF.
[C][switch.gpio:011]: Setting up GPIO Switch 'phase_b'...
[D][switch:025]: 'phase_b' Turning OFF.
[D][switch:045]: 'phase_b': Sending state OFF
[D][switch:025]: 'phase_b' Turning OFF.
[D][binary_sensor:034]: 'on_button': Sending initial state OFF
[D][binary_sensor:034]: 'off_button': Sending initial state OFF
[V][preferences:260]: nvs_get_blob('3'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[V][preferences:260]: nvs_get_blob('4'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[V][preferences:260]: nvs_get_blob('5'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[V][preferences:260]: nvs_get_blob('6'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[V][preferences:260]: nvs_get_blob('7'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[V][preferences:260]: nvs_get_blob('8'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[V][preferences:260]: nvs_get_blob('9'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[C][dallas.sensor:032]: Setting up DallasComponent...
[D][text_sensor:015]: 'Filter Pump Dev ESPHome Version': Sending state '1.17.0-dev Feb 26 2021, 05:14:17'
[D][text_sensor:015]: 'Filter Pump Dev Mac Wifi Address': Sending state '3C:71:BF:83:D3:14'
[C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[V][preferences:260]: nvs_get_blob('12'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[C][wifi:033]: Setting up WiFi...
[V][wifi_esp32:036]: Enabling STA.
[V][esp-idf:000]: I (535) wifi: wifi driver task: 3ffcbe90, prio:23, stack:3584, core=0

[V][esp-idf:000]: I (1339) wifi: wifi firmware version: 7997e4b

[V][esp-idf:000]: I (1340) wifi: config NVS flash: enabled

[V][esp-idf:000]: I (1342) wifi: config nano formating: disabled

[V][esp-idf:000]: I (1349) wifi: Init dynamic tx buffer num: 32

[V][esp-idf:000]: I (1355) wifi: Init data frame dynamic rx buffer num: 32

[V][esp-idf:000]: I (1363) wifi: Init management frame dynamic rx buffer num: 32

[V][esp-idf:000]: I (1371) wifi: Init management short buffer num: 32

[V][esp-idf:000]: I (1378) wifi: Init static rx buffer size: 1600

[V][esp-idf:000]: I (1385) wifi: Init static rx buffer num: 16

[V][esp-idf:000]: I (1392) wifi: Init dynamic rx buffer num: 32

[V][esp-idf:000]: I (1467) wifi: mode : softAP (3c:71:bf:83:d3:15)

[V][esp-idf:000]: I (1468) wifi: Total power save buffer number: 16

[V][esp-idf:000]: I (1470) wifi: Init max length of beacon: 752/752

[V][esp-idf:000]: I (1478) wifi: Init max length of beacon: 752/752

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
[V][wifi_esp32:419]: Event: WiFi AP start
[V][wifi_esp32:352]: Event: WiFi ready
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask)
Core 1 register dump:
PC      : 0x40139672  PS      : 0x00060c36  A0      : 0x8008aaa0  A1      : 0x3ffb0070
A2      : 0x3f413e84  A3      : 0x00000069  A4      : 0x3f4141b8  A5      : 0x3f413f10
A6      : 0x00060820  A7      : 0x0000cdcd  A8      : 0x8013965d  A9      : 0x3ffb0050
A10     : 0x3ffb6678  A11     : 0x3f414cd9  A12     : 0x3f413f10  A13     : 0x3f413e84
A14     : 0x00000069  A15     : 0x3f414ccc  SAR     : 0x00000004  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Backtrace: 0x40139672:0x3ffb0070 0x4008aa9d:0x3ffb00a0 0x4008c913:0x3ffb00d0 0x4008cf87:0x3ffb00f0 0x400858fc:0x3ffb0110 0x40085931:0x3ffb0130 0x40085df5:0x3ffb0150 0x4000beaf:0x3ffb0170 0x4008930e:0x3ffb0190 0x40089554:0x3ffb01b0 0x40085ec8:0x3ffb01d0 0x40085ef8:0x3ffb01f0 0x40086049:0x3ffb0220 0x4013fd7b:0x3ffb0240 0x4013c095:0x3ffb0500 0x40139675:0x3ffb0550 0x4008aa9d:0x3ffb0580 0x4008c913:0x3ffb05b0 0x4008cf87:0x3ffb05d0 0x400858fc:0x3ffb05f0 0x40085931:0x3ffb0610 0x40085df5:0x3ffb0630 0x4000beaf:0x3ffb0650 0x40085e33:0x3ffb0670 0x4000bef5:0x3ffb0690 0x4008253e:0x3ffb06b0 0x40082565:0x3ffb06d0 0x4011c794:0x3ffb06f0 0x400f8be8:0x3ffb0710 0x400e83b1:0x3ffb0810 0x400e8410:0x3ffb0840 0x400e74e5:0x3ffb0870 0x4017e559:0x3ffb0890 0x4017e615:0x3ffb08b0 0x400eac3d:0x3ffb08d0 0x400f5732:0x3ffb0920 0x40103fa3:0x3ffb1fb0 0x40089a75:0x3ffb1fd0

justlikeef, 2021-02-27:
May be the same as #1605

justlikeef, 2021-03-01:
Looks like the text_sensor.publish action causes the stack overflow.

justlikeef, 2021-03-01:
Calling the update function using id(text_sensor)->update() will also cause the stack overflow.

kbx81, 2021-03-23:
I've run into this before with large configs such as this. I edited `main.cpp in my `~/.platformio/packages/framework-arduinoespressif32/cores/esp32` folder to enlarge the stack. YMMV.

```c
xTaskCreateUniversal(loopTask, "loopTask", 8192 * 4, NULL, 1, &loopTaskHandle, CONFIG_ARDUINO_RUNNING_CORE);
```

...I added the `* 4` to quadruple the stack size to ~32KB.

justlikeef, 2021-03-23:
Thank you!!! I haven't had time to dig through code yet.

Maybe this should be the default for the ESP32 where there are more
resources?

Thanks,
Rob

kbx81, 2021-03-23:
Unfortunately that piece isn't a part of the ESPHome codebase...so I guess go complain to the Arduino folks? 😁😇

glmnet, 2021-03-23:
I wonder if there is a PR for that. A #define constant which could be redefined in platformio_options would work great

trvrnrth, 2021-03-26:
@glmnet It would appear so. And it's merged! https://github.com/espressif/arduino-esp32/pull/4564. Looks like it made it into https://github.com/espressif/arduino-esp32/releases/tag/1.0.5 / https://github.com/platformio/platform-espressif32/releases/tag/v3.1.0

glmnet, 2021-03-27:
Cool then a mechanism too set should be documented. 

I don't know is ESPHome is merging build_flags inside platformio_options

trvrnrth, 2021-03-29:
@glmnet At the moment it [overwrites them](https://github.com/esphome/esphome/blob/69879920eb21d92a8e06a16d559a5729a538456a/esphome/writer.py#L275) rather than merging which is problematic. If a special case was added to merge in build flags then something like this would work:
```
esphome:
  ...
  arduino_version: latest
  platformio_options:
    build_flags: 
      - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768
```

trvrnrth, 2021-03-29:
I've [drafted a PR](https://github.com/esphome/esphome/pull/1651) which works but may not necessarily be the best overall behaviour so open to input.

marrobHD, 2021-06-26:
Any fix?

closed 2021-07-26 by OttoWinter