esphome/issues #1876: DHT12 component fails, Received NACK on transmit of address 0x5C

WhistleMaster, 2021-02-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Python - Command line - pip
esphome config/ dashboard

**ESP (ESP32/ESP8266, Board/Sonoff):**

  platform: ESP32
  board: pico32

[https://docs.m5stack.com/#/en/core/atom_lite](M5Stack Atom Lite) with [https://docs.m5stack.com/#/en/unit/env](ENV unit)

**ESPHome version (latest production, beta, dev branch)**

v1.16.2

**Affected component:**

https://esphome.io/components/sensor/dht12.html

**Description of problem:**
Same as #278, the DHT12 sensor is detected during the scan but fails to setup. Adding setup_priority does provide the same error.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
========================= [SUCCESS] Took 52.66 seconds =========================
INFO Successfully uploaded program.
INFO Starting log output from /dev/cu.usbserial-2952B613F3 with baud rate 115200
[18:35:55]\xff[I][logger:166]: Log initialized
[18:35:55][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[18:35:55][I][app:029]: Running through setup()...
[18:35:55][C][fastled:010]: Setting up FastLED light...
[18:35:55][C][light:097]: Setting up light 'RGB LED'...
[18:35:55][D][light:265]: 'RGB LED' Setting:
[18:35:55][D][light:278]:   Brightness: 100%
[18:35:55][D][light:287]:   Red=100%, Green=100%, Blue=100%
[18:35:55][C][bmp280.sensor:056]: Setting up BMP280...
[18:35:55][C][dht12:037]: Setting up DHT12...
[18:35:55][W][i2c:070]: Received NACK on transmit of address 0x5C
[18:35:55][W][dht12:056]: Updating DHT12 failed!
[18:35:55][E][component:092]: Component was marked as failed.
[18:35:55][C][wifi:033]: Setting up WiFi...
[18:35:55][D][wifi:324]: Starting scan...
[18:35:56][D][sensor:092]: 'Ultrasonic': Sending state 1491.23999 mm with 2 decimals of accuracy
[18:35:56][D][bmp280.sensor:149]: Got temperature=23.9°C pressure=971.9hPa
[18:35:56][D][sensor:092]: 'BMP280 Temperature': Sending state 23.89000 °C with 1 decimals of accuracy
[18:35:56][D][sensor:092]: 'BMP280 Pressure': Sending state 971.94104 hPa with 1 decimals of accuracy
[18:35:57][D][wifi:339]: Found networks:
[18:35:57][I][wifi:385]: - [redacted]▂▄▆█
[18:35:57][D][wifi:386]:     Channel: 11
[18:35:57][D][wifi:387]:     RSSI: -23 dB
[18:35:57][I][wifi:385]: - [redacted]▂▄▆█
[18:35:57][D][wifi:386]:     Channel: 6
[18:35:57][D][wifi:387]:     RSSI: -82 dB
[18:35:57][D][wifi:389]: - [redacted]▂▄▆█
[18:35:57][D][wifi:389]: - [redacted]▂▄▆█
[18:35:57][D][wifi:389]: - [redacted]▂▄▆█
[18:35:57][D][wifi:389]: - [redacted]▂▄▆█
[18:35:57][I][wifi:194]: WiFi Connecting to [redacted]...
[18:35:58][I][wifi:457]: WiFi Connected!
[18:35:58][C][wifi:303]:   SSID: [redacted]
[18:35:58][C][wifi:304]:   IP Address: [redacted]
[18:35:58][C][wifi:306]:   BSSID: [redacted]
[18:35:58][C][wifi:307]:   Hostname: [redacted]
[18:35:58][C][wifi:311]:   Signal strength: -15 dB ▂▄▆█
[18:35:58][C][wifi:315]:   Channel: [redacted]
[18:35:58][C][wifi:316]:   Subnet: [redacted]
[18:35:58][C][wifi:317]:   Gateway:[redacted]
[18:35:58][C][wifi:318]:   DNS1: [redacted]
[18:35:58][C][wifi:319]:   DNS2: [redacted]
[18:35:58][C][ota:029]: Over-The-Air Updates:
[18:35:58][C][ota:030]:   Address: [redacted].local:3232
[18:35:58][C][api:022]: Setting up Home Assistant API server...
[18:35:58][I][app:059]: setup() finished successfully!
[18:35:58][D][sensor:092]: 'WiFi Signal': Sending state -15.00000 dB with 0 decimals of accuracy
[18:35:58][I][app:105]: ESPHome version 1.16.2 compiled on Feb 26 2021, 18:34:58
[18:35:58][C][wifi:443]: WiFi:
[18:35:58][C][wifi:303]:   SSID: [redacted]
[18:35:58][C][wifi:304]:   IP Address: [redacted]
[18:35:58][C][wifi:306]:   BSSID: [redacted]
[18:35:58][C][wifi:307]:   Hostname:[redacted]
[18:35:58][C][wifi:311]:   Signal strength: -15 dB ▂▄▆█
[18:35:58][C][wifi:315]:   Channel: [redacted]
[18:35:58][C][wifi:316]:   Subnet: [redacted]
[18:35:58][C][wifi:317]:   Gateway:[redacted]
[18:35:58][C][wifi:318]:   DNS1: [redacted]
[18:35:58][C][wifi:319]:   DNS2: [redacted]
[18:35:58][C][i2c:028]: I2C Bus:
[18:35:58][C][i2c:029]:   SDA Pin: GPIO26
[18:35:58][C][i2c:030]:   SCL Pin: GPIO32
[18:35:58][C][i2c:031]:   Frequency: 50000 Hz
[18:35:58][I][i2c:033]: Scanning i2c bus for active devices...
[18:35:58][I][i2c:040]: Found i2c device at address 0x57
[18:35:58][I][i2c:040]: Found i2c device at address 0x76
[18:35:58][C][fastled:019]: FastLED light:
[18:35:58][C][fastled:020]:   Num LEDs: 1
[18:35:58][C][fastled:021]:   Max refresh rate: 400
[18:35:58][C][logger:185]: Logger:
[18:35:58][C][logger:186]:   Level: DEBUG
[18:35:58][C][logger:187]:   Log Baud Rate: 115200
[18:35:58][C][logger:188]:   Hardware UART: UART0
[18:35:58][C][light:178]: Light 'RGB LED'
[18:35:58][C][light:180]:   Default Transition Length: 1.0s
[18:35:58][C][light:181]:   Gamma Correct: 2.80
[18:35:58][C][bmp280.sensor:098]: BMP280:
[18:35:58][C][bmp280.sensor:099]:   Address: 0x76
[18:35:58][C][bmp280.sensor:111]:   IIR Filter: OFF
[18:35:58][C][bmp280.sensor:112]:   Update Interval: 60.0s
[18:35:58][C][bmp280.sensor:114]:   Temperature 'BMP280 Temperature'
[18:35:58][C][bmp280.sensor:114]:     Unit of Measurement: '°C'
[18:35:58][C][bmp280.sensor:114]:     Accuracy Decimals: 1
[18:35:58][C][bmp280.sensor:114]:     Icon: 'mdi:thermometer'
[18:35:58][C][bmp280.sensor:115]:     Oversampling: 16x
[18:35:59][C][bmp280.sensor:116]:   Pressure 'BMP280 Pressure'
[18:35:59][C][bmp280.sensor:116]:     Unit of Measurement: 'hPa'
[18:35:59][C][bmp280.sensor:116]:     Accuracy Decimals: 1
[18:35:59][C][bmp280.sensor:116]:     Icon: 'mdi:gauge'
[18:35:59][C][bmp280.sensor:117]:     Oversampling: 16x
[18:35:59][D][dht12:045]: DHT12:
[18:35:59][C][dht12:046]:   Address: 0x5C
[18:35:59][E][dht12:048]: Communication with DHT12 failed!
[18:35:59][C][dht12:050]:   Temperature 'DHT12 Temperature'
[18:35:59][C][dht12:050]:     Unit of Measurement: '°C'
[18:35:59][C][dht12:050]:     Accuracy Decimals: 1
[18:35:59][C][dht12:050]:     Icon: 'mdi:thermometer'
[18:35:59][C][dht12:051]:   Humidity 'DHT12 Humidity'
[18:35:59][C][dht12:051]:     Unit of Measurement: '%'
[18:35:59][C][dht12:051]:     Accuracy Decimals: 1
[18:35:59][C][dht12:051]:     Icon: 'mdi:water-percent'
[18:35:59][C][ota:029]: Over-The-Air Updates:
[18:35:59][C][ota:030]:   Address: [redacted].local:3232
[18:35:59][C][api:095]: API Server:
[18:35:59][C][api:096]:   Address: [redacted].local:6053
[18:35:59][C][wifi_signal.sensor:009]: WiFi Signal [redacted]
[18:35:59][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[18:35:59][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[18:35:59][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
```

**Additional information and things you've tried:**

ogiewon, 2021-03-02:
I am using a AM2320 I2C Temp/Humid sensor (same as DHT12?), and could not get it to work reliably with an ESP32 until I added two 10Kohm pull-up resistors to the I2C lines.

WhistleMaster, 2021-03-03:
Thanks, I will try that.

agners, 2021-06-22:
@WhistleMaster did this resolve your issue?

WhistleMaster, 2021-06-23:
Unfortunately, I was not able to get a hand on those resistors, so I moved to SHT30 and BMP280 sensors.

closed 2021-06-23 by agners