esphome/issues #1877: ESP32 Nodemcu remains in boot loop after flashing basic configuration

shyney7, 2021-02-27:
DO NOT DELETE ANY TEXT from this template! Otherwise the issue may be closed without a comment.
-->

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Windows 10 Build: 19042.804
Esphome 1.16.1 (pip)
Python 3.9.0
pip 20.3
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 DevBoard NodeMCU 38pin Version
**ESPHome version (latest production, beta, dev branch)**

1.16.1
**Affected component:**

https://esphome.io/devices/nodemcu_esp32.html

**Description of problem:**
After flashing a basic configuration to my esp32 so that is just connects to wifi it gets stuck in the booting loop.
(see logs)

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32mijia
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "BLABLAWIFI"
  password: "supersavepassword"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32Mijia Fallback Hotspot"
    password: "supersavepassword"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "supersavepassowrd"

ota:
  password: "supersavepassword"

```

**Logs (if applicable):**

```
[13:33:14]Brownout detector was triggered
[13:33:14]
[13:33:14]ets Jun  8 2016 00:22:57
[13:33:14]
[13:33:14]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:33:14]configsip: 0, SPIWP:0xee
[13:33:14]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:33:14]mode:DIO, clock div:2
[13:33:14]load:0x3fff0018,len:4
[13:33:14]load:0x3fff001c,len:1044
[13:33:14]load:0x40078000,len:8896
[13:33:14]load:0x40080400,len:5828
[13:33:14]entry 0x400806ac
[13:33:14][I][logger:166]: Log initialized
[13:33:14][C][ota:366]: There have been 1 suspected unsuccessful boot attempts.
[13:33:14][I][app:029]: Running through setup()...
[13:33:14][C][wifi:033]: Setting up WiFi...
[13:33:14]
[13:33:14]Brownout detector was triggered
[13:33:14]
[13:33:14]ets Jun  8 2016 00:22:57
[13:33:14]
[13:33:14]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:33:14]configsip: 0, SPIWP:0xee
[13:33:14]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:33:14]mode:DIO, clock div:2
[13:33:14]load:0x3fff0018,len:4
[13:33:14]load:0x3fff001c,len:1044
[13:33:14]load:0x40078000,len:8896
[13:33:14]load:0x40080400,len:5828
[13:33:14]entry 0x400806ac
[13:33:15][I][logger:166]: Log initialized
[13:33:15][C][ota:366]: There have been 2 suspected unsuccessful boot attempts.
[13:33:15][I][app:029]: Running through setup()...
[13:33:15][C][wifi:033]: Setting up WiFi...
```

**Additional information and things you've tried:**
I've tried to clean everything with the clean command and recompiling it. I also tried to clean and delete the .platformio folder as it was suggested here https://github.com/esphome/issues/issues/58 put with no success.

Even after reconnecting the USB it still remains in the boot loop.

randybb, 2021-02-27:
Probably you have not noticed there is a "Brownout detector was triggered" message, that gets triggered when the system voltage is below a threshold, also known as the "brownout voltage" and then processor just shut downs by self. This is to preserve memory contents and avoid corruption. And this has nothing to do with ESPHome.

You are getting this message because of a problem with power:
-  the USB cable is of poor quality, or too long;
- USB port is not able to support enough power;
- or you have something wired up wrongly that is taking too much power, for example last time I have seen such error when the poor guy connected a display with swapped VCC and GND pins directly to a TTGO T-Beam board.

shyney7, 2021-02-28:
@randybb thank you! I just missed that log indeed it was a power problem.

closed 2021-02-28 by shyney7