esphome/issues #1879: Neopixelbus with multiple SK6812 LED strips

NateLogan, 2021-02-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 (Wemos Mini D1 ESP32)

**ESPHome version (latest production, beta, dev branch)**

v1.16.1

**Affected component:**

https://esphome.io/components/light/neopixelbus.html
Using version NeoPixelBus@2.6.1

**Description of problem:**
Single SK6812 LED strip works fine, adding second strip to another GPIO makes all lights act strange. With two strips they are merged into one strip overwriting each other in one strip that is defined last in the config file (confirmed by changing their order).  

It's not a HW problem, because using 4 channel WLED works just fine.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  platform: ESP32
  board: wemos_d1_mini32
  platformio_options:
    lib_deps: NeoPixelBus@2.6.1
...

light:
  - platform: neopixelbus
    id: light_led1
    type: GRBW
    pin: GPIO16
    num_leds: 28
    name: "test_led1"
  - platform: neopixelbus
    id: light_led2    
    type: GRBW
    pin: GPIO03
    num_leds: 71
    name: "test_led2"
```

**Logs (if applicable):**

```
[18:07:51][D][debug:023]: ESPHome version 1.16.1
[18:07:51][D][debug:025]: Free Heap Size: 228812 bytes
[18:07:51][D][debug:053]: Flash Chip: Size=4096kB Speed=40MHz Mode=DIO
[18:07:51][D][debug:086]: Chip: Model=ESP32, Features=WIFI_BGN,BLE,BT, Cores=2, Revision=1
[18:07:51][D][debug:088]: ESP-IDF Version: v3.2.3-14-gd3e562907
[18:07:51][D][debug:091]: EFuse MAC: 0000EC3735B267AC
[18:07:51][D][debug:143]: Reset Reason: Software Reset CPU
[18:07:51][D][debug:186]: Wakeup Reason: Unknown
[18:08:36][D][light:265]: 'test_led1' Setting:
[18:08:36][D][light:274]:   State: OFF
[18:08:36][D][light:304]:   Transition Length: 1.0s
[18:08:37][D][light:265]: 'test_led1' Setting:
[18:08:37][D][light:274]:   State: ON
[18:08:37][D][light:304]:   Transition Length: 1.0s
```

**Additional information and things you've tried:**
Different Neopixel library versions, differnet GPIO pins.

closed 2021-07-12 by stale[bot]

RAYs3T, 2021-11-08:
@NateLogan So I have the exact same issue now (on latest release).
Did you find any solution for this?

RAYs3T, 2021-11-08:
@NateLogan If you're still searching for a solution...

If you use two different methods to access the lights, it works.

See:
```yaml
light:    
  - platform: neopixelbus
    variant: LC8812 
    pin: GPIO23
    num_leds: 72
    type: GRBW
    method: ESP32_I2S_0                  # 0 Here
    id: "bathroom_mirror_light_left"
    name: "Bathroom Mirror Left Bar"
          
  - platform: neopixelbus
    variant: LC8812 
    pin: GPIO26
    num_leds: 72
    type: GRBW
    method: ESP32_I2S_1                  # 1 Here
    id: "bathroom_mirror_light_right"
    name: "Bathroom Mirror Right Bar"
```

Found this while looking at the esp home documentation.

Also a good read: https://github.com/Makuna/NeoPixelBus/wiki/ESP32-NeoMethods