esphome/issues #1882: ESP8266 WemosD1 Boot failure - Suspected I2c error

PrathikGopal, 2021-02-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 - Wemos D1 mini
**ESPHome version (latest production, beta, dev branch)**

v1.16.2
**Affected component:**

ssd1306

**Description of problem:**
Boot Failure, Error on logs. This was working previously and broke only after upgrade recently

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: kbedroom
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  fast_connect: true
  manual_ip:
    static_ip: !secret kbedroomsensor_ip
    gateway: !secret ip_gateway
    subnet: !secret ip_subnet
    dns1: !secret ip_dns1
    dns2: !secret ip_dns2
    
# Comment out and/or remove MQTT if you want to use the ESPHome API    
mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password
  
# Enable logging
logger:

ota:
  safe_mode: True
  password: !secret esphome_otapass

font:
  - file: "fonts/Comic Sans MS.ttf"
    id: comic
    size: 23
  - file: "fonts/Comic Sans MS.ttf"
    id: comic1
    size: 16
  - file: "fonts/Comic Sans MS.ttf"
    id: comic2
    size: 31
  - file: "fonts/OpenSans-Semibold.ttf"
    id: opensans
    size: 11
    
time:
  - platform: sntp
    id: time1
    timezone: Asia/Kolkata
      
sensor:
  - platform: dht
    pin:
      number: GPIO5 # D1
[... 574 lines cut ...]
[21:23:52][C][wifi:306]:   BSSID: [redacted]
[21:23:52][C][wifi:307]:   Hostname: 'kbedroom'
[21:23:52][C][wifi:311]:   Signal strength: -39 dB ▂▄▆█
[21:23:52][C][wifi:315]:   Channel: 11
[21:23:52][C][wifi:316]:   Subnet: 255.255.255.0
[21:23:52][C][wifi:317]:   Gateway: 192.168.0.1
[21:23:52][C][wifi:318]:   DNS1: 192.168.0.1
[21:23:52][C][wifi:319]:   DNS2: 192.168.0.1
[21:23:52][C][logger:185]: Logger:
[21:23:52][C][logger:186]:   Level: DEBUG
[21:23:52][C][logger:187]:   Log Baud Rate: 115200
[21:23:52][C][logger:188]:   Hardware UART: UART0
[21:23:52][C][ota:029]: Over-The-Air Updates:
[21:23:52][C][ota:030]:   Address: 192.168.0.23:8266
[21:23:52][C][ota:032]:   Using Password.
[21:23:52][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts

```

**Additional information and things you've tried:**
1. Suspect the Display component SSD1306 (Received NACK on transmit of address 0x3C) - No go
2. Removed all sensors and flashed - No go
3. Flashed a new Wemos D1 as suspected old chip - No go
4. Changed Pin config - No go
5. Ran out of ideas. - Please help.

mikosoft83, 2021-03-01:
I seem to be affected by this as well. It works for a while (a while = random time interval between 10 minutes and two days) and then just gets stuck at setup(). Tried two D1 minis, same hardware setup, same screen as above, same result. If I remove I2C seems to work just fine.

ioios-io, 2021-03-02:
Same here, not sure which update it was but having the exact same issues as Mikosoft83. All 3 of us are using the ssd1306_i2c screen, I think that's the only thing in common.
```
i2c:
  sda: ${i2cData}
  scl: ${i2cClock}
  scan: true
  frequency: 200MHz
  id: bus_a

font:
  - file: "fonts/LiberationSans-Regular.ttf"
    id: title_font
    size: 12
    glyphs: ":/HTPS"

image:
  - file: "images/return.png"
    id: return_image
    resize: 48x48

display:
  - platform: ssd1306_i2c
    model: "SH1106 128x64"
    address: 0x3C
    update_interval: 0.2s
    id: i2cDisplay
    pages:
```

wwebers, 2021-03-24:
Has anyone of you tried to raise the update intervall to 10s for the display? What I realized is that the communication with the display over I2C is quiet slow and very talkative. Which results the display driver sending a lot of data over the I2C bus to the display (even when not showing anything) and blocking the WiFi routines and sometimes it seem to prevent the ESP8266 watchdog timer running nicely.

When I raised the update_interval to 10s everything went nice. However, 10s is quiet a lot. Better solution would be to "shut up" that talkative display library...

kk-sonata, 2021-03-26:
I wonder if there's not enough heap memory to allocate the buffer for the display. I'm working on a similar issue with the st7789 display. ESPhome should have more elegant error messages in these cases.

glmnet, 2021-03-27:
Duplicate of #1628

FreshJonte, 2021-05-03:
Having a similar issue with the ST7735 display on D1 mini pro. When "update_interval" triggers and the display updates, it loses network connectivity. Drops packets and makes OTA fail for example.

This is the log (VERBOSE):
[21:56:16][V][app:081]: A component took a long time in a loop() cycle (0.95 s).
[21:56:16][V][app:082]: Components should block for at most 20-30ms in loop().

wwebers, 2021-05-04:
> Duplicate of #1628

Don't think its the same issue. The one you refer to is about a boot loop. This issue is not a boot loop as the board perfectly boots, but WiFi is dead because of too much traffic on the I2C bus produced by the display library.

PrathikGopal, 2021-05-04:
So, v1.17.0 doesn't fix it. Hopefully someone can resolve this issue.

mikosoft83, 2021-05-05:
Same with me, I updated yesterday, reflashed my device but it got stuck just like before.

PrathikGopal, 2021-05-05:
Boot is ok on v1.17.1, but the device is not found

```
[23:57:03][I][i2c:033]: Scanning i2c bus for active devices...
[23:57:03][I][i2c:049]: Found no i2c devices!
[23:57:04][C][ssd1306_i2c:023]: I2C SSD1306
[23:57:04][C][ssd1306_i2c:023]:   Rotations: 0 °
[23:57:04][C][ssd1306_i2c:023]:   Dimensions: 128px x 64px
[23:57:04][C][ssd1306_i2c:024]:   Address: 0x3C
[23:57:04][C][ssd1306_i2c:025]:   Model: SSD1306 128x64
[23:57:04][C][ssd1306_i2c:026]:   Reset Pin: GPIO16 (Mode: OUTPUT)
[23:57:04][C][ssd1306_i2c:028]:   Update Interval: 1.0s
[23:57:04][E][ssd1306_i2c:031]: Communication with SSD1306 failed!
```

PrathikGopal, 2021-05-09:
Well what do you know, it was a the damn jumper wires.... The whole bunch of them. I used a wire from a newer lot and it works. 
Thankfully the errors prompted me to relook on the connected cables.

closed 2021-05-09 by PrathikGopal