esphome/issues #1883: Request: Sorry not a bug but trying to get a kelvinator split system working with the remote transmitter

Techknowledgeman, 2021-03-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

home assistant supervised in docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32
**ESPHome version (latest production, beta, dev branch)**

lastest prod
**Affected component:**

https://esphome.io/components/remote_transmitter.html

**Description of problem:**
I have been able to determine the settings to get this working and have confirmed this by successfully capturing a remote code, and I can see the receiver flashing picking up it's own transmission, but I can't for th elife of me get the code to work. 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```
```
remote_transmitter:
  pin: GPIO25
  # RF uses a 100% carrier signal
  carrier_duty_percent: 50%
  
switch:
  - platform: template
    name: AC Power Button
    turn_on_action:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [4354, -4492, 458, -1728, 430, -676, 405, -1754, 404, -1729, 430, -650, 430, -675, 404, -1727, 431, -649, 455, -651, 404, -1729, 430, -676, 405, -673, 406, -1752, 405, -1780, 379, -675, 404, -1729, 430, -675, 432, -1702, 430, -1756, 402, -1745, 415, -1752, 405, -676, 404, -1754, 431, -1702, 431, -1753, 405, -675, 405, -650, 429, -675, 405, -675, 405, -1753, 406, -649, 430, -675, 429, -1729, 406, -1727, 431, -1728, 431, -650, 429, -701, 378, -675, 405, -675, 429, -626, 428, -675, 405, -676, 403, -677, 404, -1727, 456, -1703, 432, -1729, 429, -1729, 432, -1752, 430, -5310, 4328, -4494, 480, -1730, 405, -675, 404, -1754, 405, -1754, 405, -674, 405, -676, 403, -1730, 430, -648, 455, -625, 430, -1753, 431, -623, 431, -650, 454, -1728, 406, -1754, 404, -677, 404, -1753, 405, -676, 403, -1728, 432, -1728, 431, -1754, 405, -1781, 378, -675, 405, -1727, 431, -1754, 432, -1702, 431, -675, 403, -703, 378, -676, 402, -676, 404, -1730, 429, -676, 429, -625, 430, -1729, 430, -1727, 432, -1727, 456, -650, 405, -675, 429, -649, 406, -650, 430, -676, 403, -674, 406, -701, 377, -650, 456, -1701, 432, -1756, 404, -1754, 430, -1728, 406, -1726, 432]
  
remote_receiver:
  pin:
    number:  GPIO05
    inverted: yes
  dump: all
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Logs show the butting firing and dump the following code when using the receiver dump:

```
[21:40:14][D][remote.raw:028]: Received Raw: 4354, -4492, 458, -1728, 430, -676, 405, -1754, 404, -1729, 430, -650, 430, -675, 404, -1727, 431, -649, 455, -651, 404, -1729, 430, -676, 405, -673, 406, -1752, 405, -1780, 379, -675, 404, -1729, 430, -675, 432, -1702, 430, -1756, 402, 
[21:40:15][D][remote.raw:028]:   -1745, 415, -1752, 405, -676, 404, -1754, 431, -1702, 431, -1753, 405, -675, 405, -650, 429, -675, 405, -675, 405, -1753, 406, -649, 430, -675, 429, -1729, 406, -1727, 431, -1728, 431, -650, 429, -701, 378, -675, 405, -675, 429, -626, 428, -675, 405, 
[21:40:15][D][remote.raw:028]:   -676, 403, -677, 404, -1727, 456, -1703, 432, -1729, 429, -1729, 432, -1752, 430, -5310, 4328, -4494, 480, -1730, 405, -675, 404, -1754, 405, -1754, 405, -674, 405, -676, 403, -1730, 430, -648, 455, -625, 430, -1753, 431, -623, 431, -650, 454, -1728, 
[21:40:15][D][remote.raw:028]:   406, -1754, 404, -677, 404, -1753, 405, -676, 403, -1728, 432, -1728, 431, -1754, 405, -1781, 378, -675, 405, -1727, 431, -1754, 432, -1702, 431, -675, 403, -703, 378, -676, 402, -676, 404, -1730, 429, -676, 429, -625, 430, -1729, 430, -1727, 432, 
[21:40:15][D][remote.raw:041]:   -1727, 456, -650, 405, -675, 429, -649, 406, -650, 430, -676, 403, -674, 406, -701, 377, -650, 456, -1701, 432, -1756, 404, -1754, 430, -1728, 406, -1726, 432
```

glmnet, 2021-03-02:
I don't see a problem right away, you might want to setup another node, one as transmitter and another as receiver so you double check your hardware is transmitting ok

Techknowledgeman, 2021-03-03:
I know it's transmitting because I see the receiver pick up the signal.
I think the remote code may be the issue.
I noticed that the code changes between raw and Samsung but it's pretty erratic.

closed 2021-07-15 by stale[bot]