esphome/issues #1884: espcam fails to compile with espressif@3.1.0

comdata, 2021-03-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io, Mac OS pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESPcam32

**ESPHome version (latest production, beta, dev branch)**

v1.16.2

**Affected component:**

https://esphome.io/components/esp32_camera.html

**Description of problem:**

Build fails

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32cam
  platform: ESP32
  board: esp32cam
  arduino_version: espressif32@3.1.0
  build_path: esp32cam
  platformio_options: {}
  includes: []
  libraries: []

```

**Logs (if applicable):**

```
Compiling .pioenvs/esp32cam/src/esphome/components/json/json_util.cpp.o
src/esphome/components/esp32_camera/esp32_camera.cpp: In member function 'virtual void esphome::esp32_camera::ESP32Camera::dump_config()':
src/esphome/components/esp32_camera/esp32_camera.cpp:58:10: error: 'FRAMESIZE_QQVGA2' was not declared in this scope
     case FRAMESIZE_QQVGA2:
          ^
src/esphome/components/esp32_camera/esp32_camera.cpp: In member function 'void esphome::esp32_camera::ESP32Camera::set_frame_size(esphome::esp32_camera::ESP32CameraFrameSize)':
src/esphome/components/esp32_camera/esp32_camera.cpp:213:34: error: 'FRAMESIZE_QQVGA2' was not declared in this scope
       this->config_.frame_size = FRAMESIZE_QQVGA2;
                                  ^
*** [.pioenvs/esp32cam/src/esphome/components/esp32_camera/esp32_camera.cpp.o] Error 1

```

**Additional information and things you've tried:**

crossan007, 2021-05-21:
I'm seeing this too

bernardovalente, 2021-05-26:
I'm having this problem too

closed 2021-05-31 by jesserockz

pszeremeta86, 2021-06-30:
The same problem!