esphome/issues #1885: MQTT Hostname

jamesgoodhouse, 2021-03-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Running MQTT (mosquitto) in Kubernetes behind Traefik Proxy

**ESP (ESP32/ESP8266, Board/Sonoff):**

`ESP32-WROOM-32D`

**ESPHome version (latest production, beta, dev branch)**

latest production

**Affected component:**

[MQTT](https://esphome.io/components/mqtt.html)

**Description of problem:**
I have the broker field for the MQTT component set to the hostname of my MQTT installation. Upon checking the logs I see that it's resolving the hostname to an IP. Unfortunately this means it's coming into Traefik by that IP instead of the hostname, resulting in a connection never being established.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
mqtt:
  broker: mqtt.example.com
  port: 1883
```

**Logs (if applicable):**

```
.
.
.
PASTE DEBUG LOG HERE
[12:56:02][C][mqtt:025]: Setting up MQTT...
[12:56:02][D][mqtt:103]: Resolving MQTT broker IP address...
[12:56:02][D][mqtt:136]: Resolved broker IP address to 172.31.3.200
[12:56:02][I][mqtt:162]: Connecting to MQTT...
[12:57:02][D][mqtt:103]: Resolving MQTT broker IP address...
[12:57:02][D][mqtt:136]: Resolved broker IP address to 172.31.3.200
[12:57:02][I][mqtt:162]: Connecting to MQTT...
[12:58:02][D][mqtt:103]: Resolving MQTT broker IP address...
[12:58:02][D][mqtt:136]: Resolved broker IP address to 172.31.3.200
[12:58:02][I][mqtt:162]: Connecting to MQTT...
[12:59:02][D][mqtt:103]: Resolving MQTT broker IP address...
[12:59:02][D][mqtt:136]: Resolved broker IP address to 172.31.3.200
[12:59:02][I][mqtt:162]: Connecting to MQTT...
[13:00:02][D][mqtt:103]: Resolving MQTT broker IP address...
[13:00:02][D][mqtt:136]: Resolved broker IP address to 172.31.3.200
[13:00:02][I][mqtt:162]: Connecting to MQTT...
```

**Additional information and things you've tried:**
I got the MQTT component working fine by removing my MQTT installation from behind Traefik and having it's port available on the actual host. This is non-ideal and only a workaround, but it proved to me that I could get things working.

jamesgoodhouse, 2021-03-03:
I was able to work around this by changing my Traefik config to forward all requests on the MQTT port to mosquitto.

closed 2021-03-03 by jamesgoodhouse

randybb, 2021-03-03:
But this is how communication on top of the Internet Protocol works - hostname gets resolved and then only IP address is used. You can double check how an [IP packet looks ](https://en.wikipedia.org/wiki/IPv4), especially its address field, looks like :)
That means - fix your DNS!