esphome/issues #1886: Flicker Light Effect Doesn't Respect Primary Color - Changes to White

rozza-m, 2021-03-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

esphome 1.16.2

**Affected component:**
RGBWW
https://esphome.io/components/light/index.html#light-effects

**Description of problem:**
When on flicker light effect, ESP does not respect primary color and turns to white. Any change to primary color is ignored on flicker. The light does flicker, but always white.
**This appears to be a duplicate or regression of https://github.com/esphome/issues/issues/562**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: bathroom_light_1
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "..."
  password: "..."
  reboot_timeout: 1min

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bathroom Light 1"
    password: "..."

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

web_server:
  port: 80

output:
  - platform: esp8266_pwm
    id: output_red
    pin: GPIO5
  - platform: esp8266_pwm
    id: output_green
    pin: GPIO4    
  - platform: esp8266_pwm
    id: output_blue
    pin: GPIO13
  - platform: esp8266_pwm
    id: output_warm_white
    pin: GPIO12
  - platform: esp8266_pwm
    id: output_cold_white
    pin: GPIO14

light:
  - platform: rgbww
    name: bathroom_light_1
    id: bathroom_light_1
    red: output_red
    green: output_green
    blue: output_blue
    warm_white: output_warm_white
    cold_white: output_cold_white
    cold_white_color_temperature: 6200 K
    warm_white_color_temperature: 2800 K
    color_interlock: true # avoids simultaneous RGB and W/W
    constant_brightness: false
    effects:
      - flicker

```

**Logs (if applicable):**
```
[D][light:265]: 'bathroom_light_3' Setting:
[D][light:287]:   Red=75%, Green=0%, Blue=100%
[D][light:290]:   White Value: 0%
[D][light:304]:   Transition Length: 1.0s
[D][light:265]: 'bathroom_light_3' Setting:
[D][light:327]:   Effect: 'Flicker'
[I][ota:046]: Boot seems successful, resetting boot loop counter.
```

Webserver reports:
```
{"id":"light-bathroom_light_3","state":"ON","effect":"Flicker","brightness":166,"color":{"r":255,"g":255,"b":255},"white_value":0,"color_temp":161}
```

This appears to be exactly the same issue as https://github.com/esphome/issues/issues/562. Has anyone else seen this again? I've verified the code and I am seeing the (new) code merged through https://github.com/esphome/esphome/pull/749 which fixed the issue previously.

regoras, 2021-03-06:
Pretty sure this is related to #1790, if not a duplicate. I experienced the same issue with RGBCCT bulbs and color_interlock on.

rozza-m, 2021-03-11:
Closing then as dup of #1790.

closed 2021-03-11 by rozza-m