esphome/issues #1888: Will ESPHome connect on a network with a router using WEP only?

SteveRMann, 2021-03-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant (HassOS) on an Intel NUC
**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**
My network is ancient and the router only supports WEP. (Yes I know I need to upgrade, but I have some 50 devices around the house and opening each of them to flash a new hard-coded network password would take days).

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
The terminal on the test device (a NodeMCU) shows the network scan, followed by a list of found APs with mine at the top. Except for "No matching networks found".  So, is the fact that my router is using WEP my problem?

OttoWinter, 2021-04-10:
It will not. That's to prevent downgrade attacks (force the esp to use less secure WEP, then use that to reconstruct password).

SteveRMann, 2021-04-10:
I upgraded my router, so now the question is moot.
(It only took three days to get everything connected again).

closed 2021-04-10 by SteveRMann