esphome/issues #1889: mqtt on_json_message triggers twice on single topic

jds11111, 2021-03-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Running as addon in HassOS

**ESP (ESP32/ESP8266, Board/Sonoff):**

WeMos D1 mini

**ESPHome version (latest production, beta, dev branch)**

Latest production
**Affected component:**

MQTT client component

**Description of problem:**

I have the device listen for a certain topic on MQTT. The topic is published by a sonoff RF bridge, and when it detects a certain RF signal, I want a light to toggle.  That has worked fine for many months. However, with the latest ESPhome version it performs the toggle twice, so the light just flashes very briefly. Toggling by the attached switch works fine. So do toggling by the web socket, and toggling by the HA API. The MQTT topic published by the RF bridge happens only once. Yet, the logs in the ESP device just show a very rapid toggle twice (from on to back off, or vice versa).

**Problem-relevant YAML-configuration entries:**
```yaml
mqtt:
  broker: !secret mqtt_broker_address
  port: 1883
  username: !secret mqtt_broker_user
  password: !secret mqtt_broker_password
  discovery: True
  reboot_timeout: 0s

# Slave the lamp to the wall remote on the wall
  on_json_message:
    topic: tele/sonoffRF1/RESULT
    qos: 0
    then:
      - if:
         condition:
           lambda: 'return x.containsKey("RfReceived");'
         then:
           - if:
              condition:
                lambda: 'return x["RfReceived"]["Data"]=="$hex_code";'
              then:
                light.toggle: lamp_light
```

**Logs (if applicable):**

```
[07:39:17][D][sensor:092]: 'lamp_relay2 Uptime': Sending state 415.91101 s with 0 decimals of accuracy
[07:39:43][D][light:265]: 'Bedroom Lamp Light' Setting:
[07:39:43][D][light:274]:   State: OFF
[07:39:43][D][light:265]: 'Bedroom Lamp Light' Setting:
[07:39:43][D][light:274]:   State: ON

```

**Additional information and things you've tried:**

I could not see anything else to try, since delay is not allowed with this component.

jds11111, 2021-03-06:
I have another light operating the same way, but using a mhetesp32devkit. It exhibits exactly the same bug.

ssieb, 2021-06-06:
You could add a log message to see how many times it's getting called and what the message is.  Have you checked with another mqtt client to see if maybe the message *is* getting duplicated somehow?

jds11111, 2021-06-06:
> You could add a log message to see how many times it's getting called and what the message is. Have you checked with another mqtt client to see if maybe the message _is_ getting duplicated somehow?

Do you mean add it to the ESPhome yaml? Is that just by changing the logging level to debug?

ssieb, 2021-06-06:
Like this:
```
  on_json_message:
    topic: tele/sonoffRF1/RESULT
    qos: 0
    then:
      - lambda: |-
          std::string j;
          x.printTo(j);
          ESP_LOGD("JSON", "mqtt json: %s", j.c_str());
      - if:
```

closed 2021-10-19 by stale[bot]