esphome/issues #1890: TTGO T-Camera v1.6.2 with MIC - boot loops

Bazmundi, 2021-03-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Running esphome on debian 10

**ESP (ESP32/ESP8266, Board/Sonoff):**

LilyGO TTGO T-Camera v1.6.2

**ESPHome version (latest production, beta, dev branch)**

v1.16.1 which pulls in PlatformIO 5.0.4

**Affected component:**

Not sure, seems different combinations of interactions depending on tide?

**Description of problem:**
I have two t-camera v1.6.2 boards.   One would not start and I parred back the configuration file to find it would not run with the camera code included.   This turns out a known problem, locked away in the contributors' issues at: https://github.com/esphome/issues/issues/405

By the description in 405 I could say the problem has not been fixed despite the issue being closed.

I placed the "problem" board aside and worked on the second one board that appeared not to have the same problem.   The code snippet I got working on the second board is at: https://pastebin.com/L538rJpm

That configuration worked a treat and the MQTT messages worked fine, the button and IR detections worked fine, the wifi status reports worked fine, the display worked fine and the camera worked fine.

I powered down the "working" board, left it for a week.  Powered it up today and with no changes to configuration I found 1) I had to use hard reset quite a number of times before the board would start up at all and then 2) mqtt messages and OLED display were not working BUT IR detections, button press, wifi status reports and camera still worked.

So, without changing anything, the configuration stopped working.  Not, at the time I was writing the configuration, I did power on and off a couple of times so, it did originally come up after a power off condition.

**Problem-relevant YAML-configuration entries:**
```yaml
https://pastebin.com/L538rJpm

```

**Logs (if applicable):**

```
Original problem with "faulty" t-camera board at https://pastebin.com/cXrVthZw

Latest problem on second board similar up until the hard reset takes then the board starts.

```

**Additional information and things you've tried:**

Originally, with one board "workging" and one board not-working I did trim back the configuration to find the problem with one board seemed to be the camera code.   The other board was fine, at the time, and I added the mqtt code.  Code before the mqtt code was added is at: https://pastebin.com/n7fRtRhM

Bazmundi, 2021-03-08:
May be fine.  I assumed 405 "closed" meant incorporated into ESPHome as a fix, it turns out if I hack into main.cpp setup {} (avoiding the autogenerated code sections), the "fix" in 405 works, but it appears a generic fix for ESP32 cams and as likely needs to be considered built into ESPHome?

Mikrofarad, 2021-03-20:
I had the same problem, only boot loops if I wanted to use the camera on the board V1.6.2.
My fix is:
```yaml
  - platform: gpio
    name: DummyIO14
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
```

dedalodaelus, 2021-05-25:
It also solved for me:

Thanks!

I added some extra config in order not see the pin as sensor
```yaml
  - platform: gpio
    name: DummyIO14
    filters:
      - delayed_on: 9999ms
      - delayed_off: 9999ms
      - delayed_on_off: 9999ms
      - lambda: |-
         return { "0" };
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
````

> 
> 
> I had the same problem, only boot loops if I wanted to use the camera on the board V1.6.2.
> My fix is:
> 
> ```yaml
>   - platform: gpio
>     name: DummyIO14
>     pin:
>       number: GPIO14
>       mode: INPUT_PULLUP
> ```

BUGBEAR7, 2021-06-07:
Thanks a lot , I also have been banging my head for this
finally got it working after 6 month

Mikrofarad, 2021-06-20:
It took me some time to "fix" this issue. I have been sure there must be a way, because the original firmware had no problem at all.

[Paul Deen](https://community.home-assistant.io/t/working-esphome-config-for-ttgo-esp32-camera-board-with-microphone/126231/18) suggested this:
```yaml
filters:
      - delayed_on: 1000ms
```
Because my ESPHome debug level is ERROR I didn't see these prints.

closed 2021-11-02 by github-actions[bot]