esphome/issues #1891: Captive Portal OTA does not complete

amoedo, 2021-03-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io core-2021.2.3 Home Assistant OS 5.12 oon Raspberry Pi
**ESP (ESP32/ESP8266, Board/Sonoff):**

EDP8266 Wemos D1 Mini 
**ESPHome version (latest production, beta, dev branch)**

v1.16.2
**Affected component:**

https://esphome.io/components/captive_portal.html

**Description of problem:**
I am trying to flash a new firmware from the captive portal, but the firmware upload never completes and there are no errors on the logs. The firmware upload starts fine, but looking at the logs seems like the WiFi scanning kills it. I've tried with Chome and Firefox on Windows 10. I can reproduce this with a minimal configuration as per below yaml.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: testesp
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "secret"
  password: "secret"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Testesp Fallback Hotspot"
    password: "otapassword"

captive_portal:

# Enable logging
debug:
logger:
  level: VERY_VERBOSE
# Enable Home Assistant API
api:
  password: "pass"

ota:

status_led:
  pin:
    number: D4
    inverted: True
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
15:10:59.039 -> [0;33m[W][wifi:394]: No matching network found![0m
15:11:01.236 -> [0;38m[VV][wifi_esp8266:485]: Event: AP receive Probe Request MAC=8E:FE:CE:9D:43:42 RSSI=-92[0m
15:11:01.236 -> wifi evt: 7
15:11:04.033 -> [0;36m[D][wifi:324]: Starting scan...[0m
15:11:04.407 -> [0;38m[VV][wifi_esp8266:485]: Event: AP receive Probe Request MAC=CE:38:BD:57:E6:92 RSSI=-90[0m
15:11:04.407 -> wifi evt: 7
15:11:09.972 -> scandone
15:11:09.972 -> [0;36m[D][wifi:339]: Found networks:[0m
15:11:09.972 -> [0;36m[D][wifi:389]: - [5m'Darpa'[6m [5m(3C:84:6A:AB:AF:CA) [6m[0;32m▂▄▆█[0m[0m
15:11:10.018 -> [0;36m[D][wifi:389]: - [5m'Darpa_Guest'[6m [5m(42:84:6A:AB:AF:CA) [6m[0;32m▂▄▆█[0m[0m
15:11:10.018 -> [0;36m[D][wifi:389]: - [5m''[6m [5m(46:84:6A:AB:AF:CA) [6m[0;32m▂▄▆█[0m[0m
15:11:10.018 -> [0;36m[D][wifi:389]: - [5m'wificlientesR'[6m [5m(72:5A:9E:90:4D:7B) [6m[0;33m▂▄▆[0;37m█[0m[0m
15:11:10.018 -> [0;36m[D][wifi:389]: - [5m'CELIA'[6m [5m(70:5A:9E:90:4D:7A) [6m[0;33m▂▄▆[0;37m█[0m[0m
15:11:10.018 -> [0;36m[D][wifi:389]: - [5m'TP-Link_PEDRO'[6m [5m(0C:80:63:B9:E4:56) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:10.065 -> [0;36m[D][wifi:389]: - [5m'Darpa'[6m [5m(3C:84:6A:AB:B0:22) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:10.065 -> [0;36m[D][wifi:389]: - [5m'Darpa_Guest'[6m [5m(42:84:6A:AB:B0:22) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:10.065 -> [0;36m[D][wifi:389]: - [5m''[6m [5m(46:84:6A:AB:B0:22) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:10.065 -> [0;36m[D][wifi:389]: - [5m'Darpa_Guest'[6m [5m(42:84:6A:AB:B0:46) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:10.065 -> [0;36m[D][wifi:389]: - [5m''[6m [5m(46:84:6A:AB:B0:46) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:10.112 -> [0;36m[D][wifi:389]: - [5m'Darpa'[6m [5m(3C:84:6A:AB:B0:46) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:10.112 -> [0;36m[D][wifi:389]: - [5m'Tech_D0010144'[6m [5m(48:00:33:7A:23:EF) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:10.112 -> [0;36m[D][wifi:389]: - [5m'MOVISTAR_4F34'[6m [5m(CC:D4:A1:01:4F:36) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:10.112 -> [0;36m[D][wifi:389]: - [5m'MOVISTAR_E236'[6m [5m(CC:ED:DC:13:E2:37) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:10.112 -> [0;36m[D][wifi:389]: - [5m'wificlientesR'[6m [5m(72:5A:9E:35:F5:C1) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:10.159 -> [0;36m[D][wifi:389]: - [5m'vodafoneBA5110'[6m [5m(74:9D:79:B4:61:D0) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:10.159 -> [0;36m[D][wifi:389]: - [5m'PEDRO R_WIFI'[6m [5m(70:5A:9E:35:F5:C0) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:10.159 -> [0;36m[D][wifi:389]: - [5m'MOVISTAR_503E'[6m [5m(E8:D1:1B:A4:50:3F) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:10.159 -> [0;36m[D][wifi:389]: - [5m'MIWIFI_2G_03Ht'[6m [5m(4C:6E:6E:D0:85:B3) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:10.159 -> [0;36m[D][wifi:389]: - [5m'MOVISTAR_5E22'[6m [5m(98:97:D1:BD:5E:23) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:10.207 -> [0;36m[D][wifi:389]: - [5m'MiFibra-057B'[6m [5m(44:FE:3B:3E:05:7D) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:10.207 -> [0;36m[D][wifi:389]: - [5m'R-WLAN 3B'[6m [5m(80:29:94:F7:E7:AC) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:10.207 -> [0;36m[D][wifi:389]: - [5m'MOVISTAR_D049'[6m [5m(D4:7B:B0:10:D0:4A) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:10.207 -> [0;33m[W][wifi:394]: No matching network found![0m
15:11:12.773 -> [0;38m[VV][wifi_esp8266:485]: Event: AP receive Probe Request MAC=3C:BD:3E:73:D5:F1 RSSI=-89[0m
15:11:12.773 -> wifi evt: 7
15:11:13.565 -> [0;32m[I][web_server_base:028]: OTA Update Start: testesp.bin[0m
15:11:13.565 -> [begin] already running
15:11:13.565 -> [0;33m[W][web_server_base:021]: OTA Update failed! Error: ERROR[0]: No Error
15:11:13.565 -> [0m
15:11:13.565 -> [0;36m[D][web_server_base:059]: OTA in progress: 0.4%[0m
15:11:14.638 -> [0;36m[D][web_server_base:059]: OTA in progress: 14.4%[0m
15:11:15.247 -> [0;36m[D][wifi:324]: Starting scan...[0m
15:11:15.667 -> [0;36m[D][web_server_base:059]: OTA in progress: 26.7%[0m
15:11:16.739 -> [0;36m[D][web_server_base:059]: OTA in progress: 31.8%[0m
15:11:21.364 -> scandone
15:11:21.364 -> [0;36m[D][wifi:339]: Found networks:[0m
15:11:21.364 -> [0;36m[D][wifi:389]: - [5m''[6m [5m(46:84:6A:AB:AF:CA) [6m[0;32m▂▄▆█[0m[0m
15:11:21.364 -> [0;36m[D][wifi:389]: - [5m'Darpa_Guest'[6m [5m(42:84:6A:AB:AF:CA) [6m[0;32m▂▄▆█[0m[0m
15:11:21.364 -> [0;36m[D][wifi:389]: - [5m'Darpa'[6m [5m(3C:84:6A:AB:AF:CA) [6m[0;32m▂▄▆█[0m[0m
15:11:21.410 -> [0;36m[D][wifi:389]: - [5m'CELIA'[6m [5m(70:5A:9E:90:4D:7A) [6m[0;33m▂▄▆[0;37m█[0m[0m
15:11:21.410 -> [0;36m[D][wifi:389]: - [5m'wificlientesR'[6m [5m(72:5A:9E:90:4D:7B) [6m[0;33m▂▄▆[0;37m█[0m[0m
15:11:21.410 -> [0;36m[D][wifi:389]: - [5m'TP-Link_PEDRO'[6m [5m(0C:80:63:B9:E4:56) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:21.410 -> [0;36m[D][wifi:389]: - [5m'Darpa_Guest'[6m [5m(42:84:6A:AB:B0:22) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:21.410 -> [0;36m[D][wifi:389]: - [5m'Darpa'[6m [5m(3C:84:6A:AB:B0:22) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:21.457 -> [0;36m[D][wifi:389]: - [5m''[6m [5m(46:84:6A:AB:B0:22) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:21.457 -> [0;36m[D][wifi:389]: - [5m''[6m [5m(46:84:6A:AB:B0:46) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:21.457 -> [0;36m[D][wifi:389]: - [5m'MOVISTAR_4F34'[6m [5m(CC:D4:A1:01:4F:36) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:21.457 -> [0;36m[D][wifi:389]: - [5m'Darpa_Guest'[6m [5m(42:84:6A:AB:B0:46) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:21.457 -> [0;36m[D][wifi:389]: - [5m'Darpa'[6m [5m(3C:84:6A:AB:B0:46) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:21.503 -> [0;36m[D][wifi:389]: - [5m'MOVISTAR_E236'[6m [5m(CC:ED:DC:13:E2:37) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:21.503 -> [0;36m[D][wifi:389]: - [5m'wificlientesR'[6m [5m(72:5A:9E:35:F5:C1) [6m[0;33m▂▄[0;37m▆█[0m[0m
15:11:21.503 -> [0;36m[D][wifi:389]: - [5m'vodafoneBA5110'[6m [5m(74:9D:79:B4:61:D0) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:21.503 -> [0;36m[D][wifi:389]: - [5m'PEDRO R_WIFI'[6m [5m(70:5A:9E:35:F5:C0) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:21.503 -> [0;36m[D][wifi:389]: - [5m'MIWIFI_2G_03Ht'[6m [5m(4C:6E:6E:D0:85:B3) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:21.550 -> [0;36m[D][wifi:389]: - [5m'MiFibra-057B'[6m [5m(44:FE:3B:3E:05:7D) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:21.550 -> [0;36m[D][wifi:389]: - [5m'MOVISTAR_503E'[6m [5m(E8:D1:1B:A4:50:3F) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:21.550 -> [0;36m[D][wifi:389]: - [5m'MOVISTAR_5E22'[6m [5m(98:97:D1:BD:5E:23) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:21.550 -> [0;36m[D][wifi:389]: - [5m'MOAÑAMAR2.4'[6m [5m(80:3F:5D:D4:E6:01) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:21.550 -> [0;36m[D][wifi:389]: - [5m'R-WLAN 3B'[6m [5m(80:29:94:F7:E7:AC) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:21.597 -> [0;36m[D][wifi:389]: - [5m'vodafoneBA1911'[6m [5m(74:9D:79:BD:08:C0) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:21.597 -> [0;36m[D][wifi:389]: - [5m'MOVISTAR_D049'[6m [5m(D4:7B:B0:10:D0:4A) [6m[0;31m▂[0;37m▄▆█[0m[0m
15:11:21.597 -> [0;33m[W][wifi:394]: No matching network found![0m
15:11:24.626 -> [0;38m[VV][wifi_esp8266:485]: Event: AP receive Probe Request MAC=9E:65:31:4E:9B:79 RSSI=-92[0m
15:11:24.626 -> wifi evt: 7
15:11:26.628 -> [0;36m[D][wifi:324]: Starting scan...[0m
15:11:32.523 -> scandone
```

**Additional information and things you've tried:**

I've tried the upload multiple times. It never completes. It always seems to be killed off as soon as the device starts wifi scanning.

amoedo, 2021-03-06:
I've tested a similar situation with another board ESP8266 NodeMCuv2. You can see two attempts for OTA from the Captive Portal on the log below. The first one fails. I assume because the upload didn't finish before the wifi scan interval. The second one I've timed it to hit just after the scan finished and it completed uploading before the next one. The OTA was successful but the firmware was corrupted and the device didn't start. Flashing it via USB worked fine.

````
[22:53:02][W][wifi:394]: No matching network found!
[22:53:07][D][wifi:324]: Starting scan...
[22:53:08][I][web_server_base:028]: OTA Update Start: esp1 (1).bin
[22:53:08][D][web_server_base:059]: OTA in progress: 0.3%
[22:53:09][D][dht:048]: Got Temperature=22.3°C Humidity=40.6%
[22:53:09][D][sensor:092]: 'Living Room Temperature': Sending state 22.30000 °C with 1 decimals of accuracy
[22:53:09][D][sensor:092]: 'Living Room Humidity': Sending state 40.60000 % with 0 decimals of accuracy
[22:53:13][D][wifi:339]: Found networks:
[22:53:13][D][wifi:389]: - '' (46:84:6A:AB:AF:CA) ▂▄▆█
[22:53:13][D][wifi:389]: - 'Darpa' (3C:84:6A:AB:AF:CA) ▂▄▆█
[22:53:13][D][wifi:389]: - 'Darpa_Guest' (42:84:6A:AB:AF:CA) ▂▄▆█
[22:53:13][D][wifi:389]: - 'CELIA' (70:5A:9E:90:4D:7A) ▂▄▆█
[22:53:13][D][wifi:389]: - 'wificlientesR' (72:5A:9E:90:4D:7B) ▂▄▆█
[22:53:13][D][wifi:389]: - 'TP-Link_PEDRO' (0C:80:63:B9:E4:56) ▂▄▆█
[22:53:13][D][wifi:389]: - 'Darpa' (3C:84:6A:AB:B0:22) ▂▄▆█
[22:53:13][D][wifi:389]: - 'Darpa_Guest' (42:84:6A:AB:B0:22) ▂▄▆█
[22:53:13][D][wifi:389]: - '' (46:84:6A:AB:B0:22) ▂▄▆█
[22:53:13][D][wifi:389]: - 'Tech_D0010144' (48:00:33:7A:23:EF) ▂▄▆█
[22:53:13][D][wifi:389]: - 'Darpa' (3C:84:6A:AB:B0:46) ▂▄▆█
[22:53:13][D][wifi:389]: - '' (46:84:6A:AB:B0:46) ▂▄▆█
[22:53:13][D][wifi:389]: - 'MOVISTAR_4F34' (CC:D4:A1:01:4F:36) ▂▄▆█
[22:53:13][D][wifi:389]: - 'Darpa_Guest' (42:84:6A:AB:B0:46) ▂▄▆█
[22:53:13][D][wifi:389]: - 'vodafoneBA5110' (74:9D:79:B4:61:D0) ▂▄▆█
[22:53:13][D][wifi:389]: - 'MOVISTAR_E236' (CC:ED:DC:13:E2:37) ▂▄▆█
[22:53:13][D][wifi:389]: - 'MOVISTAR_5E22' (98:97:D1:BD:5E:23) ▂▄▆█
[22:53:13][D][wifi:389]: - 'Cudy-E5A2' (B4:4B:D6:25:E5:A2) ▂▄▆█
[22:53:13][D][wifi:389]: - 'MOVISTAR_503E' (E8:D1:1B:A4:50:3F) ▂▄▆█
[22:53:13][D][wifi:389]: - 'wificlientesR' (72:5A:9E:35:F5:C1) ▂▄▆█
[22:53:13][D][wifi:389]: - 'MOVISTAR_79F8' (84:AA:9C:21:79:F9) ▂▄▆█
[22:53:13][D][wifi:389]: - 'David' (88:F7:C7:2B:0A:92) ▂▄▆█
[22:53:13][D][wifi:389]: - 'MOVISTAR_D049' (D4:7B:B0:10:D0:4A) ▂▄▆█
[22:53:13][D][wifi:389]: - 'vodafoneBA1911' (74:9D:79:BD:08:C0) ▂▄▆█
[22:53:13][D][wifi:389]: - 'wificlientesR' (8A:F7:C7:2B:0A:93) ▂▄▆█
[22:53:13][D][wifi:389]: - 'MOVISTAR_36B5' (E4:AB:89:A5:36:B6) ▂▄▆█
[22:53:13][W][wifi:394]: No matching network found!
[22:53:18][D][wifi:324]: Starting scan...
[22:53:19][D][dht:048]: Got Temperature=22.3°C Humidity=40.5%
[22:53:19][D][sensor:092]: 'Living Room Temperature': Sending state 22.30000 °C with 1 decimals of accuracy
[22:53:19][D][sensor:092]: 'Living Room Humidity': Sending state 40.50000 % with 0 decimals of accuracy
[22:53:24][D][wifi:339]: Found networks:
[22:53:24][D][wifi:389]: - '' (46:84:6A:AB:AF:CA) ▂▄▆█
[22:53:24][D][wifi:389]: - 'Darpa' (3C:84:6A:AB:AF:CA) ▂▄▆█
[22:53:24][D][wifi:389]: - 'Darpa_Guest' (42:84:6A:AB:AF:CA) ▂▄▆█
[22:53:24][D][wifi:389]: - 'CELIA' (70:5A:9E:90:4D:7A) ▂▄▆█
[22:53:24][D][wifi:389]: - 'wificlientesR' (72:5A:9E:90:4D:7B) ▂▄▆█
[22:53:24][D][wifi:389]: - 'TP-Link_PEDRO' (0C:80:63:B9:E4:56) ▂▄▆█
[22:53:24][D][wifi:389]: - 'Darpa_Guest' (42:84:6A:AB:B0:22) ▂▄▆█
[22:53:24][D][wifi:389]: - 'Tech_D0010144' (48:00:33:7A:23:EF) ▂▄▆█
[22:53:24][D][wifi:389]: - '' (46:84:6A:AB:B0:22) ▂▄▆█
[22:53:24][D][wifi:389]: - 'Darpa' (3C:84:6A:AB:B0:22) ▂▄▆█
[22:53:24][D][wifi:389]: - 'Darpa_Guest' (42:84:6A:AB:B0:46) ▂▄▆█
[22:53:24][D][wifi:389]: - '' (46:84:6A:AB:B0:46) ▂▄▆█
[22:53:24][D][wifi:389]: - 'Darpa' (3C:84:6A:AB:B0:46) ▂▄▆█
[22:53:24][D][wifi:389]: - 'MOVISTAR_4F34' (CC:D4:A1:01:4F:36) ▂▄▆█
[22:53:24][D][wifi:389]: - 'MOVISTAR_E236' (CC:ED:DC:13:E2:37) ▂▄▆█
[22:53:24][D][wifi:389]: - 'vodafoneBA5110' (74:9D:79:B4:61:D0) ▂▄▆█
[22:53:24][D][wifi:389]: - 'MIWIFI_2G_03Ht' (4C:6E:6E:D0:85:B3) ▂▄▆█
[22:53:24][D][wifi:389]: - 'MOVISTAR_6E0E' (CC:ED:DC:5A:6E:0F) ▂▄▆█
[22:53:24][D][wifi:389]: - 'MOVISTAR_D049' (D4:7B:B0:10:D0:4A) ▂▄▆█
[22:53:24][D][wifi:389]: - 'PEDRO R_WIFI' (70:5A:9E:35:F5:C0) ▂▄▆█
[22:53:24][D][wifi:389]: - 'wificlientesR' (72:5A:9E:35:F5:C1) ▂▄▆█
[22:53:24][D][wifi:389]: - 'vodafoneBA1911' (74:9D:79:BD:08:C0) ▂▄▆█
[22:53:24][D][wifi:389]: - 'MOVISTAR_5E22' (98:97:D1:BD:5E:23) ▂▄▆█
[22:53:24][D][wifi:389]: - 'MOVISTAR_79F8' (84:AA:9C:21:79:F9) ▂▄▆█
[22:53:24][D][wifi:389]: - 'MiFibra-057B' (44:FE:3B:3E:05:7D) ▂▄▆█
[22:53:24][D][wifi:389]: - 'Cudy-E5A2' (B4:4B:D6:25:E5:A2) ▂▄▆█
[22:53:24][D][wifi:389]: - 'Orange-0839' (A8:D3:F7:14:08:3B) ▂▄▆█
[... 69 lines cut ...]
[22:53:47][I][web_server_base:028]: OTA Update Start: esp1 (1).bin
[22:53:47][W][web_server_base:021]: OTA Update failed! Error: ERROR[0]: No Error
[22:53:47]
[22:53:47][D][web_server_base:059]: OTA in progress: 0.3%
[22:53:48][D][web_server_base:059]: OTA in progress: 48.7%
[22:53:49][D][dht:048]: Got Temperature=22.4°C Humidity=41.1%
[22:53:49][D][sensor:092]: 'Living Room Temperature': Sending state 22.40000 °C with 1 decimals of accuracy
[22:53:49][D][sensor:092]: 'Living Room Humidity': Sending state 41.10000 % with 0 decimals of accuracy
[22:53:49][D][web_server_base:059]: OTA in progress: 92.4%
[22:53:49][I][web_server_base:068]: OTA update successful!
[22:53:49][I][app:142]: Rebooting safely...
[22:53:50]
[22:53:50] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[22:53:50]
[22:53:50]load 0x4010f000, len 3584, room 16 
[22:53:50]tail 0
[22:53:50]chksum 0xb0
[22:53:50]csum 0xb0
[22:53:50]v2843a5ac
[22:53:56]@cp:0
[22:53:56]ld
[22:53:56]e:
[22:53:56] ets Jan  8 2013,rst cause:3, boot mode:(3,6)
[22:53:56]
````

bjornstevens, 2021-03-09:
I might be completely wrong, but will adding this under wifi do anything?
  fast_connect: true

amoedo, 2021-03-10:
Thanks for the suggestion. I gave it a try, but it doesn't help. The behaviour is similar. As soon as it starts trying to connect to the wifi without scan (fast connect), the OTA stops. Nothing relevant that I can see in the logs. It's like the file upload fails.

```
23:33:53.327 -> wifi evt: 7
23:33:53.327 -> [0;32m[I][wifi:194]: WiFi Connecting to 'secret'...[0m
23:33:53.327 -> [0;37m[V][wifi:196]: Connection Params:[0m
23:33:53.327 -> [0;37m[V][wifi:197]:   SSID: 'secret'[0m
23:33:53.373 -> [0;37m[V][wifi:202]:   BSSID: Not Set[0m
23:33:53.373 -> [0;37m[V][wifi:220]:   Password: [5m'secret1234'[6m[0m
23:33:53.373 -> [0;37m[V][wifi:227]:   Channel: Not Set[0m
23:33:53.373 -> [0;37m[V][wifi:235]:   Using DHCP IP[0m
23:33:53.373 -> [0;37m[V][wifi:237]:   Hidden: NO[0m
23:33:56.231 -> scandone
23:33:56.231 -> [0;33m[W][wifi_esp8266:441]: Event: Disconnected ssid='secret' reason='Probe Request Unsuccessful'[0m
23:33:56.231 -> wifi evt: 1
23:33:56.231 -> STA disconnect: 201
23:33:56.231 -> [0;33m[W][wifi:485]: Error while connecting to network.[0m
23:33:57.500 -> [0;32m[I][web_server_base:028]: OTA Update Start: testesp (4).bin[0m
23:33:57.500 -> sleep disable
23:33:57.500 -> [begin] roundedSize:       0x00396000 (3760128)
23:33:57.500 -> [begin] updateEndAddress:  0x003FB000 (4173824)
23:33:57.547 -> [begin] currentSketchSize: 0x00064000 (409600)
23:33:57.547 -> [begin] _startAddress:     0x00065000 (413696)
23:33:57.547 -> [begin] _currentAddress:   0x00065000 (413696)
23:33:57.547 -> [begin] _size:             0x00396000 (3760128)
23:33:57.547 -> [0;36m[D][web_server_base:059]: OTA in progress: 0.3%[0m
23:33:57.593 -> Header: 0xE9 1 3 40
23:33:58.299 -> [0;38m[VV][wifi_esp8266:485]: Event: AP receive Probe Request MAC=04:79:70:B6:1D:77 RSSI=-28[0m
23:33:58.344 -> wifi evt: 7
23:33:58.344 -> [0;38m[VV][wifi_esp8266:485]: Event: AP receive Probe Request MAC=04:79:70:B6:1D:77 RSSI=-35[0m
23:33:58.344 -> wifi evt: 7
23:33:58.344 -> [0;38m[VV][wifi_esp8266:485]: Event: AP receive Probe Request MAC=04:79:70:B6:1D:77 RSSI=-32[0m
23:33:58.344 -> wifi evt: 7
23:33:58.344 -> [0;38m[VV][wifi_esp8266:485]: Event: AP receive Probe Request MAC=04:79:70:B6:1D:77 RSSI=-31[0m
23:33:58.344 -> wifi evt: 7
23:33:58.576 -> [0;36m[D][web_server_base:059]: OTA in progress: 14.3%[0m
23:33:58.668 -> [0;38m[VV][wifi_esp8266:485]: Event: AP receive Probe Request MAC=04:79:70:B6:1D:77 RSSI=-70[0m
23:33:58.668 -> wifi evt: 7
23:33:59.647 -> [0;36m[D][web_server_base:059]: OTA in progress: 23.2%[0m
23:34:00.626 -> [0;36m[D][web_server_base:059]: OTA in progress: 36.4%[0m
23:34:01.515 -> [0;32m[I][wifi:194]: WiFi Connecting to 'secret'...[0m
23:34:01.515 -> [0;37m[V][wifi:196]: Connection Params:[0m
23:34:01.515 -> [0;37m[V][wifi:197]:   SSID: 'secret'[0m
23:34:01.515 -> [0;37m[V][wifi:202]:   BSSID: Not Set[0m
23:34:01.515 -> [0;37m[V][wifi:220]:   Password: [5m'secret1234'[6m[0m
23:34:01.515 -> [0;37m[V][wifi:227]:   Channel: Not Set[0m
23:34:01.515 -> [0;37m[V][wifi:235]:   Using DHCP IP[0m
23:34:01.562 -> [0;37m[V][wifi:237]:   Hidden: NO[0m
23:34:01.656 -> [0;36m[D][web_server_base:059]: OTA in progress: 50.3%[0m
23:34:03.293 -> [0;38m[VV][wifi_esp8266:485]: Event: AP receive Probe Request MAC=04:79:70:B6:1D:77 RSSI=-27[0m
23:34:03.293 -> wifi evt: 7
23:34:03.903 -> [0;38m[VV][wifi_esp8266:485]: Event: AP receive Probe Request MAC=04:79:70:B6:1D:77 RSSI=-71[0m
23:34:03.903 -> wifi evt: 7
23:34:04.469 -> scandone
23:34:04.469 -> [0;33m[W][wifi_esp8266:441]: Event: Disconnected ssid='secret' reason='Probe Request Unsuccessful'[0m
23:34:04.469 -> wifi evt: 1
23:34:04.469 -> STA disconnect: 201
23:34:04.516 -> [0;33m[W][wifi:485]: Error while connecting to network.[0m
```

closed 2021-07-23 by stale[bot]