esphome/issues #1892: Failed config with the ArduinoPortExpander and analog reading

davidusb-geek, 2021-03-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Ubuntu 20.04 LTS

**ESP (ESP32/ESP8266, Board/Sonoff):**
Wemos d1 mini and the ArduinoPortExpander installed on an Arduino Uno

**ESPHome version (latest production, beta, dev branch)**
v1.16.2

**Affected component:**
https://esphome.io/cookbook/arduino_port_extender.html
https://esphome.io/components/sensor/adc.html
https://esphome.io/components/sensor/ct_clamp.html

**Description of problem:**
Hi, I've been using the ArduinoPortExpander with esphome to control mostly relays, digital outputs (to LEDs and buzzers, etc) and for some binary sensors. I'm using a wemos d1 mini and an Arduino Uno as the APE device.
Now I want to read two power loads in my house using the analog input pins and two different current clamps (SCT-013-030). I want to use for this the CT Clamp current sensor from esphome (https://esphome.io/components/sensor/ct_clamp.html). The problem is that analog sensor output from the APE is not a voltage. So either using the ct_clamp or the analog to digital (https://esphome.io/components/sensor/adc.html) sensor integrations gives a failed configuration error.

**Problem-relevant YAML-configuration entries:**
```
esphome:
  name: main_tableau_elec
  platform: ESP8266
  board: d1_mini
  includes:
    - arduino_port_expander.h

wifi:
  ssid: !secret wifi_id
  password: !secret wifi_pwd

logger:
  level: DEBUG

api:
  password: !secret ota_pwd

ota:
  password: !secret ota_pwd

i2c:
  id: i2c_component

custom_component:
  - id: ape
    lambda: |-
      auto ape_component = new ArduinoPortExpander(i2c_component, 0x08, true);
      return {ape_component};

sensor:
  - platform: custom
    lambda: |-
      return {ape_analog_input(ape, 0),  // 0 = A0
              ape_analog_input(ape, 1)};  // 1 = A1
    sensors:
      - name: Analog A1
        id: analog_a1
        filters:
          - throttle: 10s
      - name: Analog A2
        id: analog_a2
        filters:
          - throttle: 10s
  - platform: adc
    pin: analog_a1
    name: ADC1
    update_interval: 60s
  - platform: adc
    pin: analog_a2
    name: ADC2
    update_interval: 60s
```

**Logs (if applicable):**
And the error message obtained is this:
```
Failed config

sensor.adc: [source main_tableau_elec.yaml:61]
  platform: adc
  
  Cannot resolve pin name 'analog_a1' for board d1_mini.
  pin: analog_a1 [source main_tableau_elec.yaml:62]
  name: ADC1
  update_interval: 60s
sensor.adc: [source main_tableau_elec.yaml:65]
  platform: adc
  
  Cannot resolve pin name 'analog_a2' for board d1_mini.
  pin: analog_a2 [source main_tableau_elec.yaml:66]
  name: ADC2
  update_interval: 60s
```

**Additional information and things you've tried:**
The same error arises if I try to use the ct_clamp integration. Do you think that there is a way to achieve some compatibility with the ct_clamp or the adc esphome integrations? Could you help me to solve this issue? What modifications as custom codes needs to be made to achieve this compatibility?
A secondary solution for me is to connect a ADS1115 directly to my wemos d1 mini ESP8266, but then it will be a shame to not use those analog pins available in the Arduino Uno APE.
Thank you for your help

lordzid, 2021-03-26:
tell me it gives me an error
`In file included from src/main.cpp:22:0:
src/arduino_port_expander.h:106:1: error: expected class-name before '{' token
 {
 ^
src/arduino_port_expander.h:108:36: error: expected ')' before '*' token
   ArduinoPortExpander(I2CComponent *parent, uint8_t address, bool vref_default = false) : I2CDevice(parent, address)
                                    ^
src/arduino_port_expander.h: In member function 'virtual void ArduinoPortExpander::loop()':
src/arduino_port_expander.h:134:17: error: 'class ArduinoPortExpander' has no member named 'read_bytes'
       if (this->read_bytes(APE_CMD_DIGITAL_READ, const_cast<uint8_t *>(this->read_buffer_), 3, 1))
                 ^
src/arduino_port_expander.h:142:17: error: 'class ArduinoPortExpander' has no member named 'write_byte'
           this->write_byte(CMD_SETUP_ANALOG_DEFAULT, 0); // 0: unused
                 ^
src/arduino_port_expander.h: In member function 'void ArduinoPortExpander::write_state(uint8_t, bool, bool)':
src/arduino_port_expander.h:273:11: error: 'class ArduinoPortExpander' has no member named 'write_byte'
     this->write_byte(state ? APE_CMD_WRITE_DIGITAL_HIGH : APE_CMD_WRITE_DIGITAL_LOW, pin);
           ^
src/arduino_port_expander.h:281:13: error: 'class ArduinoPortExpander' has no member named 'write_byte'
       this->write_byte(APE_CMD_SETUP_PIN_OUTPUT, pin);
             ^
src/main.cpp: In lambda function:
src/main.cpp:173:47: error: 'i2c_component' was not declared in this scope
       auto expander = new ArduinoPortExpander(i2c_component, 0x08, true);
                                               ^
src/main.cpp:174:23: error: could not convert '{expander}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::Component*>'
       return {expander};
                       ^
src/main.cpp:175:3: warning: control reaches end of non-void function [-Wreturn-type]
   });
   ^
*** [/data/tastgarage/.pioenvs/tastgarage/src/main.cpp.o] Error 1
========================= [FAILED] Took 10.24 seconds =========================

Отвечать`

davidusb-geek, 2021-03-29:
Mmmm, seems like an i2c related problem?
For it compiles and run with no errors. 
As said in the message I'm using the latest v1.16.2

closed 2021-08-10 by stale[bot]