esphome/issues #1893: ili9431 display colors inverted on (older) M5Stack Cores

Josverl, 2021-03-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Python 3.8.5 on Windows 10 
pip 
**ESP (ESP32/ESP8266, Board/Sonoff):**
 ESP32 
 - M5stack Core - hardware rev 2017.11
 - M5stack Fire - hardware rev 2018.2a

**ESPHome version (latest production, beta, dev branch)**

ESPHome Version: 1.16.2
**Affected component:**

https://esphome.io/components/display/ili9341.html

**Description of problem:**
All colors are inverted 
Color used in code | Expected| Color shown on display
---------------------|-----------|--
COLOR_BLACK | Black | White 
COLOR_WHITE | White| Black
Color(1,0,0) | Red | Purple 
Color(0,1,0) | Green | orange 
Color(0,0,1) | Blue | Light Blue

**Problem-relevant YAML-configuration entries:**
```yaml
display:
  - platform: ili9341
    # model: TFT_2.4
    model: M5Stack
    cs_pin: 14
    dc_pin: 27
    led_pin: 32
    reset_pin: 33
    rotation: 0
    update_interval: 10s
    lambda: |-
      // Draw a circle in the middle of the display
      it.fill(id(COLOR_BLACK));
      it.filled_circle(it.get_width() / 2, it.get_height() / 2, 30, id(RED));
      it.rectangle(10,10,200,200,id(BLUE));
      it.print(0, 0, id(my_font), id(GREEN),"Hello World!");

color: #ILI colors inverted :-(
    - id: RED
      red:   1
      green: 0
      blue:  0
    
    - id: BLUE
      red:     0
      green:   0
      blue:    1
    
    - id: GREEN
      red:   0
      green: 1
      blue:  0

```

**Logs (if applicable):**

```
[21:08:29][C][logger:185]: Logger:
[21:08:29][C][logger:186]:   Level: VERBOSE
[21:08:29][C][logger:187]:   Log Baud Rate: 115200
[21:08:29][C][logger:188]:   Hardware UART: UART0
[21:08:29][C][ili9341:035]: ili9341
[21:08:29][C][ili9341:035]:   Rotations: 0 °
[21:08:29][C][ili9341:035]:   Dimensions: 320px x 240px
[21:08:29][C][ili9341:036]:   Width: 320, Height: 240,  Rotation: 0
[21:08:29][C][ili9341:037]:   Reset Pin: GPIO33 (Mode: OUTPUT)
[21:08:29][C][ili9341:038]:   DC Pin: GPIO27 (Mode: OUTPUT)
[21:08:29][C][ili9341:040]:   Backlight Pin: GPIO32 (Mode: OUTPUT)
[21:08:29][C][ili9341:042]:   Update Interval: 10.0s

```

**Additional information and things you've tried:**

the ESPHome driver for the ili9342 display is [hardcoded to invert the display on the M5 Stack](https://github.com/esphome/esphome/blob/2c0fe49b86f5b1ede27e76f3ef68f5f03af22d65/esphome/components/ili9341/ili9341_display.cpp#L227), which makes the driver work incorrectly on hardware that does not need to be inverted.

i have updated the driver to allow 
- the yaml file to specify the display inversion to be configured `invert: true/false`
- i have tested this only on the m5stack

im happy to submit a PR for the code and update the relevant documentation if someone can point me to the appropriate workflows.

TotallyInformation, 2021-04-04:
I also have this problem. Could you please share the change that you made manually?

TotallyInformation, 2021-04-04:
In case anyone else stumbles on this, here is a temporary work around ...

---

Well, that took a while! I finally worked out that you have to edit the code in the place that pip installs it. Probably obvious to most people but I'm a node.js person not really a Python dev. So next problem was finding that location.

As I'm using Windows 10 with the UWP (app store) version of Python 3.8, this location is far from obvious. Eventually worked out that `pip3 show esphome` would tell me where it actually lives.

From there, I could simply replace line 227 of `~\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\esphome\components\ili9341\ili9341_display.cpp` with:

```python
  this->invert_display_(false);
```

Instead of true. Recompile and now the colours are correct! I'll see if I can come up with something more robust by exposing some more flags and then I'll update the Issue, maybe even do a PR.

Josverl, 2021-04-08:
though exploration (trail , error , frustration, insight) i figured out that I can copy the 'somewhere\python\hidden\obfuscated\components\ili9341' folder to my projects './custom_components/ili9341' folder and then could edit and deploy it from there. 
the custom component then takes precedence over the standard component. 
(simple once solved) 

code is located here :
https://github.com/Josverl/ESPHome/tree/m5stack-ili/custom_components/ili9341

TotallyInformation, 2021-04-09:
Yes, that is a frustrating experience isn't it! A very steep learning curve.

Unfortunately, there are an awful lot of assumptions made in the docs. Of course we are free to do a PR on the docs repository and suggest some updates 😁

The code is very complex as well so I couldn't work out how to do a better fix - it would certainly be better to expose some of the other configuration items so that they could be changed in your yaml.

Anyway, I've finally worked out how to do the custom override as well and I've been working through my yaml, splitting it into logical and mostly reusable chunks. I also eventually managed to work out how to add my own custom C++ code for use in lambda functions. Something else not well documented.

Now I'm focused on creating a framework for display layout so I don't have to keep tweaking pixels and recompiling and uploading all the time. It would be really nice to be able to position things logically (taking into account font sizes) as rows and columns.

Still, this is the first framework that has let me use the same base code/configuration on all of my different devices (Wemos D1-mini, mini Pro, TTGO T-Display and M5stack Basic). And is has let me reproduce the MQTT topic layout that I've been using with ESPeasy and my own custom firmware. So some real wins.

closed 2021-08-22 by stale[bot]

geiseri, 2021-10-04:
This still holds true for 2021.9.0.