esphome/issues #1896: 2 pzem's on different pins (and therefore different UART Busses)

bjornstevens, 2021-03-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2021-03-10:
Che k if you can give an ID to modbus then add modbus_id for each sensor. I know it's not documented but it's how usually works

bjornstevens, 2021-03-11:
Thanks, that was the exact piece I was missing!

Config now is as follows:

`modbus:
    - id: mod1
      uart_id: ua1
    - id: mod2
      uart_id: ua2
    

uart:
  - id: ua1
    tx_pin: D1
    rx_pin: D2
    baud_rate: 9600
    #stop_bits: 1
  - id: ua2
    tx_pin: D5
    rx_pin: D6
    baud_rate: 9600
    #stop_bits: 1
  
sensor:
  - platform: pzemac
    current:
      name: "PZEM-004T V3 Current"
    voltage:
      name: "PZEM-004T V3 Voltage"
    energy:
      name: "PZEM-004T V3 Energy"
    power:
      name: "PZEM-004T V3 Power"
    frequency:
      name: "PZEM-004T V3 Frequency"
    power_factor:
      name: "PZEM-004T V3 Power Factor"
    update_interval: 1s
    modbus_id: mod1
  - platform: pzemac
    current:
      name: "PZEM-004T_2 V3 Current"
    voltage:
      name: "PZEM-004T_2 V3 Voltage"
    energy:
      name: "PZEM-004T_2 V3 Energy"
    power:
      name: "PZEM-004T_2 V3 Power"
    frequency:
      name: "PZEM-004T_2 V3 Frequency"
    power_factor:
      name: "PZEM-004T_2 V3 Power Factor"
    update_interval: 1s
    modbus_id: mod2
`

bjornstevens, 2021-03-11:
Maybe an extra: but is there a way to sync the requests? Like sync update_interval?

bjornstevens, 2021-03-12:
Allright, so now comes the new issue: 
[image]

Somtimes when the 2 pzem's are synced up, I receive this message. Is there a way to correct this?

glmnet, 2021-03-12:
Start by setting uptade interval to never, to the second pzem 

Then you'll need to trigger it manually. Use this action: https://esphome.io/guides/automations.html#component-update-action

You can add an on_value trigger to any sensor of first pzem

bjornstevens, 2021-03-12:
Thanks for the reply, it was also exactly what I was looking for! Thanks!

bjornstevens, 2021-03-12:
I will close this case for now due to my problem being fixed. Thanks for the help @glmnet !

closed 2021-03-12 by bjornstevens