esphome/issues #1907: GPIO switch via MCP23017 causes random behavior of binary_sensors

jougs, 2021-03-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant Core on a vanilla Debian 10

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

v1.16.0

**Affected component:**

https://esphome.io/components/mcp230xx.html

**Description of problem:**
I'm using several MCP23017 attached to an ESP32 via a single I<sup>2</sup>C bus. Some of their GPIOs are defined as `switch`, some as `binary_sensor` (using basically the same YAML-code as in [the documentation](https://esphome.io/components/mcp230xx.html#mcp23017), just [much] more of the same).

If I toggle any of the switches, I get random state changes all over the place, i.e., a random selection of `binary_sensors` change state with it, leading to a blinking chain-reaction frenzy of lights and scenes turning on and off in random succession. If I declare the GPIOs in question as `output` instead of `switch`, and then base a `binary light` on each of those, they all work as expected (icon and device class being wrong, obviously).

Is this a known bug? Has anyone else seen this kind of behavior?

**Problem-relevant YAML-configuration entries:**

**Logs (if applicable):**

**Additional information and things you've tried:**

I have not yet checked if all the misbehaving GPIOs are always on the same IC, or if they only happen to be on the same I<sup>2</sup>C bus.

Misiu, 2021-03-17:
Did you try to declare all gpios on a single mcp chip as binary_sensor and all gpios on other mcp as switches? (idea is to use single mcp as binary_sensor and second as switches)

jougs, 2021-03-17:
No, I did not yet try this. However, as I had the same hardware setup working just fine with my prototype using MicroPython, I suspect a bug in esphome rather than a side-effect as you describe it.

kbx81, 2021-03-23:
Have you enabled `mode: INPUT_PULLUP` on the pins you are using as binary_sensors (inputs)? ...or do you have external resistors to pull them up/down? This could very well be some kind of a bug, although I can't recall seeing any similar complaints before. In any event, just a suggestion/thought I'd ask. 😄

closed 2021-08-04 by stale[bot]